raspberrypi/linux #5367: Screen goes blank some seconds with KMS DRM on boot and with vlc

LozramA, 2023-03-02:
### Describe the bug

I am using the newest 32bit Raspberry OS with all updates.
When using KMS my screen goes shortly blank on boot. And also on every start and close of vlc player with "--vout=drm_vout" in python.

If i switch to fkms the screen blanking on boot stops.
if i change vlc to ""--vout mmal_vout"" the screen blanking stops with vlc.

Please tell me if you need more details.

Monitor is a 640x480 5 inch chinese Innolux Display connected over HDMi.

Same behaviour happens with the default chinese HDMI2AV Adapters and connected to any analog Video screen.

### Steps to reproduce the behaviour

boot rasperry os and start python vlc player with DRM KMS

### Device (s)

Raspberry Pi 4 Mod. B

### System

32bit Raspberry OS all newest updates

LozramA, 2023-03-07:
Another thing i found is when i stop my script ( Ctrl + C ) during vlc video playback and start it then again the blanking is gone and it works like it should.
But only on video and music playback with visual plugins not on vlc photo playback.

Perhaps it helps to locate the problem...

popcornmix, 2023-03-07:
Can you post output of raspinfo when booted with default config.txt and cmdline.txt settings.
(so no hdmi_xxx=1 type settings you've added).

LozramA, 2023-03-07:
will do it tomorrow.
have found a workaround an hour ago in your forums regarding other problems.

--drm-vout-no-modeset  fixes the problem on video / music with visual and photo playback.
No blank screens anymore.

Only the one on boot.