raspberrypi/linux #5368: Make KMS drivers ignore bad EDID and force resolution.

b1063n, 2023-03-03:
@6by9 is there a way we can ask the Pi to not look for any EDID/hotplugging and just still connect it, so that we do not have to put any cmdline addition like that ? 

FKMS already works that way, it doesnt rely on EDID as you said. Would it be possible to do the same for KMS?

_Originally posted by @rnayabed in https://github.com/raspberrypi/linux/issues/4719#issuecomment-974660353_

---------------------------------------

I have a display with a wrong EDID file, I confirmed this. I want to run this display at 1920x1080 and it works great when the KMS driver is disabled. When the KMS driver is enabled only lower resolutions are available. I tried editing the cmdline.txt as suggested in the post above but no success.

The splash screens and loading screens are displayed at 1920x1080 all good there, but later on the resolution is limited. I need to run the KMS drivers because I am also using the libcam libraries which rely on the KMS drivers.

Is there a way to load the KMS drivers but ignore the EDID from the monitor? It works very well when KMS is disabled, but I need to enable KMS.

6by9, 2023-03-03:
So KMS processes the EDID incorrectly? It's the same code between FKMS and KMS, so I'd need to see the EDID read in each case to believe that diagnosis.
```xxd /sys/class/drm/card1-HDMI-A-1``` for KMS and FKMS please, and the list of modes (```cat /sys/class/drm/card1-HDMI-A-1/modes```, or the first page from ```modetest -M vc4 | less```). (It may be card0-HDMI-A-1 depending on load order, or cardN-HDMI-A-2 if you're using the second HDMI port).

More likely it is the monitor misbehaving with hotplug. FKMS/firmware will always try reading the EDID even if hotplug isn't asserted.

If you really want to override the EDID, put the raw EDID in /lib/firmware as eg edid.bin, and then edit /boot/cmdline.txt to add ```drm.edid_firmware=HDMI-A-1:edid.bin```.
Forcing hotplug can be done via ```vc4.force_hotplug=0x01``` in cmdline.txt. Amend to 0x02 for the second HDMI port, or 0x03 for both ports.

Forcing hotplug can cause issues if hotplugging in HDMI2.0 modes due to needing to reset scrambling.

b1063n, 2023-03-05:
The display is a cheap 11inch with no brand (native res 1920x1080), super sketchy but it works. It could be misbehaving indeed.

My suspicion of the EDID files comes from the fact that: If I comment the line "#dtoverlay=vc4-kms-v3d" on the boot/config.txt , the display works great. There is nothing to be done, the display just works out of the box full resolution no problems.

However, if uncommented the display will be limited to the resolutions listed by "xrandr --verbose" which if I am not mistaken are read from the EDID? 

I will overrride the EDID as you suggested. Thanks.

IreneJiang123, 2025-02-20:
> The display is a cheap 11inch with no brand (native res 1920x1080), super sketchy but it works. It could be misbehaving indeed.
>
> My suspicion of the EDID files comes from the fact that: If I comment [...]

hi, did it work? i have the similar problem

b1063n, 2025-02-20:
Hi Irene,

Sorry, I actually, gave up, I had more pressing issues.

Perhaps it can work well with the new Raspbian bookworm version? I was using bullseye back then.