raspberrypi/linux #5379: 6.1.16 kernel broke wifi on RPi 3b+

iucoen, 2023-03-12:
### Describe the bug

After upgrading to the latest rpi-6.1.y branch, version 6.1.16 cannot connect to wifi for me anymore. I did a diff between 6.1.15 and 6.1.16 and was able to narrow it down to this commit: 22dfb21bf1cd876616d45cda1bc6daa89eec6747

After reverting that commit wifi connects fine. 

### Steps to reproduce the behaviour

Compile 6.1.16 kernel, install, reboot, and wifi can no longer connect.

### Device (s)

Raspberry Pi 3 Mod. B+

### System

OS is Gentoo Linux.
Kernel is 6.1.16-v8+

DaniPhii, 2023-03-12:
I experienced the same issue using Arch Linux ARM, downgraded kernel to 6.1.14 and Wi-Fi is working again.

iucoen, 2023-03-12:
Upstream confirmed the issue, patch here: https://lore.kernel.org/all/20230311141914.24444-1-marcan@marcan.st/

pelwell, 2023-03-12:
Thanks everyone - we'll cherry-pick the fix ASAP.

popcornmix, 2023-03-14:
We've bumped to 6.1.19 which includes the fix.
Latest rpi-update kernel contains this.

closed 2023-03-15 by iucoen