raspberrypi/linux #5387: Still no network interfaces after upgrading kernel to 6.1.19-2

basaandewiel, 2023-03-18:
### Describe the bug

The bug described in bug 5379 is still present after upgrading the kernel to 6.1.19-2.

### Steps to reproduce the behaviour

Install arch linux on RPI-3B. upgrade kernel to 6.1.19-2, reboot; no wireless and also no fixed network interface found.

If I donwgrade the kernel by
$ pacman -U linux-rpi-6.1.13-1-armv7h.pkg.tar.xz 
followed by a reboot, then the network connections (fixed and wireless) are working again.

### Device (s)

Raspberry Pi 3 Mod. B

### System

Arch linux, fully up-to-date

Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

Linux RPI3191012 6.1.19-2-rpi-ARCH #1 SMP Thu Feb 23 12:34:54 MST 2023 armv7l GNU/Linux

pelwell, 2023-03-18:
Do you have a spare card on which you can install and update an RPiOS image, to see if the issue is specific to Arch?

pelwell, 2023-03-18:
Errm, there's something fishy here:
```
Linux RPI3191012 6.1.19-2-rpi-ARCH #1 SMP Thu Feb 23 12:34:54 MST 2023 armv7l GNU/Linux
```
6.1.19 was only released on March 13th (https://github.com/raspberrypi/linux/commit/6449a0ba6843fe70523eeb7855984054f36f6d24), so I don't understand how your kernel can have the version and date indicated.

basaandewiel, 2023-03-18:
I don't know what has happened; I just downgraded and upgraded the kernel again, rebooted the pi again, and now it is working correctly.
But I discovered a strange behaviour: after a second reboot, the wlan0 did not get an IP-address.

I rebooted 6 times: sometimes wlan0 is working, sometimes not. 
Output of uname -a is:
Linux RPI3191012 6.1.19-2-rpi-ARCH #1 SMP Thu Mar 16 13:14:12 MDT 2023 armv7l GNU/Linux

I attached some dmesg outputs in case it wlan0 is working and when it is not. Hope this helps

[230318_2103_notworking.txt](https://github.com/raspberrypi/linux/files/11009502/230318_2103_notworking.txt)
[230318_2106_working.txt](https://github.com/raspberrypi/linux/files/11009503/230318_2106_working.txt)
[230319_2110_notworking.txt](https://github.com/raspberrypi/linux/files/11009504/230319_2110_notworking.txt)
[230319_2112_working.txt](https://github.com/raspberrypi/linux/files/11009505/230319_2112_working.txt)

NB: Beside the built in wlan0 I have attached an extra WLAN-USB (ralink)

P33M, 2023-03-18:
> ### Steps to reproduce the behaviour
> Install arch linux on RPI-3B. upgrade kernel to 6.1.19-2, reboot; no wireless and also no fixed network interface found.
Arch Linux build their own kernels. Perhaps you should raise the issue with them.

basaandewiel, 2023-03-19:
OK, thanks for your time and effort!

closed 2023-03-19 by basaandewiel