raspberrypi/linux #5390: On a CM3+, the regenerate_ssh_host_keys script in rasapberrypi-sys-mods takes a very long time to run on kernel v6.1

JinShil, 2023-03-21:
### Describe the bug

On a CM3+, the `regenerate_ssh_host_keys` script in `rasapberrypi-sys-mods` is taking a very long time to run after installing 6.1 via apt using our pi-gen-based OS creation script. The offending line appears to be `dd if=/dev/hwrng of=/dev/urandom count=1 bs=4096 status=none`

```
$ time sudo dd if=/dev/hwrng of=/dev/urandom count=1 bs=4096 status=none
real    3m28.951s
user    0m0.024s
sys     0m0.041s
```

The same command runs in under a second on a CM4 or CM4S.

### Steps to reproduce the behaviour

Using a CM3+, install a 64-bit Bullseye OS with kernel v6.1 and run the `regenerate_ssh_host_keys` script.

### Device (s)

Raspberry Pi CM3+, Raspberry Pi CM3+ Lite

### System

OS was generated using a script based on pi-gen.

Kernel version 6.1.19-v8+

pelwell, 2023-03-21:
I can confirm the regression - a 3B+ shows the same problem (I've not checked other models yet).

popcornmix, 2023-03-21:
Pi 4 seemed quick for me.

popcornmix, 2023-03-21:
On Pi4:
```
sudo cat /dev/hwrng
+K���(��u<��pP�3�<0�\R�⾬}���F����
```
(spurts lots of random stuff).

On Pi3 nothing comes out.

pelwell, 2023-03-21:
It's interesting that there should be a difference (which I'll investigate), but on the surface the problem appears to come from this upstream commit: 96cb9d0554457086664d3bd10630b11193d863f1, the payload of which is:
```diff
diff --git a/drivers/char/hw_random/bcm2835-rng.c b/drivers/char/hw_random/bcm2835-rng.c
index e7dd457e9b22b..e98fcac578d66 100644
--- a/drivers/char/hw_random/bcm2835-rng.c
+++ b/drivers/char/hw_random/bcm2835-rng.c
@@ -71,7 +71,7 @@ static int bcm2835_rng_read(struct hwrng *rng, void *buf, size_t max,
        while ((rng_readl(priv, RNG_STATUS) >> 24) == 0) {
                if (!wait)
                        return 0;
-               cpu_relax();
+               hwrng_msleep(rng, 1000);
        }
 
        num_words = rng_readl(priv, RNG_STATUS) >> 24;
```
i.e. wait for _a whole second_ between reads. Dropping this to 1ms gives a big improvement:
```
$ time sudo dd if=/dev/hwrng of=/dev/urandom count=1 bs=10000 status=none
real    0m15.025s
user    0m0.011s
sys     0m0.041s
```
But the reversion to `cpu_relax()` is equally dramatic:
```
$ time sudo dd if=/dev/hwrng of=/dev/urandom count=1 bs=100000 status=none
real    0m0.855s
user    0m0.000s
sys     0m0.851s
```
(note this is for 0.855s for 100,000 bytes as opposed to 15.025s for 10,000 bytes).

I'm going to revert the patch for now, but we should investigate whether this is even needed - the purpose of the kthread the commit refers to seems to be to shovel data into the random number generator.

pelwell, 2023-03-21:
Florian can't have been awake that day.

lategoodbye, 2023-03-22:
Hi,
reverting the patch seems a short term solution to me. I think we should keep the cpu_relax(), but introduce another exit criteria for this busy loop like max cycles or a timeout.

@ffainelli What's your opinion?

XECDesign, 2023-03-22:
> @XECDesign Do we still need the explicit random number seeding in `regenerate_ssh_host_keys`?

Yeah, it runs on the very first boot, before systemd runs or has a chance to start rngd.

pelwell, 2023-03-22:
That would be a very invasive change, with patches to the driver and DTS files. The iproc_rng200 driver that's used on 2711 (hence the absence of a problem on Pi 4) waits for up to 500us, with the actual length government by the amount of data that's required:
```
			/* Can wait, give others chance to run */
			usleep_range(min(num_remaining * 10, 500U), 500);
```

JinShil, 2023-03-22:
Thank you, I'll test tomorrow.  rpi-update will provide us with a temporary workaround, but our OS creation scripts are based on pi-gen, so can you estimate when this fix might be available through apt?

popcornmix, 2023-03-22:
We don't have a scheduled date for the next apt update.
We're keeping an eye on issues reported due to the recent kernel bump,
and results of that may trigger an apt update.

pelwell, 2023-03-22:
See https://github.com/raspberrypi/linux/pull/5396 for a better fix.

pelwell, 2023-03-23:
To install a trial build with the more intelligent sleeping, run `sudo rpi-update pulls/5396`.

JinShil, 2023-03-24:
> To install a trial build with the more intelligent sleeping, run `sudo rpi-update pulls/5396`.

I tested that today, and it seems to work fine.

popcornmix, 2023-03-28:
> A timeline for the fixed version on apt here would indeed be nice, this is currently a blocker for providing up-to-date OctoPi images for OctoPrint.

It's hard to predict the future, but we are planning an apt bump for kernel soon (perhaps 1 week).

foosel, 2023-03-28:
Thank you, that gives some rough idea :)

ffainelli, 2023-03-28:
@lategoodbye do you have cycles to update bcm2835-rng.c to work more closely to the iproc-rng200.c driver?

pelwell, 2023-03-28:
@ffainelli See https://github.com/raspberrypi/linux/issues/5390 for a candidate.

lategoodbye, 2023-03-29:
> @lategoodbye do you have cycles to update bcm2835-rng.c to work more closely to the iproc-rng200.c driver?

Hard to say when this will be.

JinShil, 2023-04-06:
It appears this fix was released on apt today.  I ran our pi-gen script and it generated an OS image that no longer exhibited this issue.  This issue appears fixed to me, so I'm closing this issue.  Thank you.

closed 2023-04-06 by JinShil

lategoodbye, 2023-09-15:
FWIW the following patch has been applied in mainline:
https://lore.kernel.org/linux-crypto/ZQQ1tgDEGGXwfu%2F4@gondor.apana.org.au/T/#t

popcornmix, 2023-09-15:
On a Pi3+, our current downstream kernel (which uses usleep_range):
```
pi@b3plus:~ $ sudo dd if=/dev/hwrng of=/dev/null count=1 bs=10000
10000 bytes (10 kB, 9.8 KiB) copied, 0.0783782 s, 128 kB/s
```
Switching to previous upstream version:
```
pi@b3plus:~ $ sudo dd if=/dev/hwrng of=/dev/null count=1 bs=1000
1000 bytes (1.0 kB) copied, 67.2426 s, 0.0 kB/s  [15 bytes/s]
```
Switching to new upstream upstream version:
```
pi@b3plus:~ $ sudo dd if=/dev/hwrng of=/dev/null count=1 bs=10000
10000 bytes (10 kB, 9.8 KiB) copied, 3.99268 s, 2.5 kB/s
pi@b3plus:~ $ 
```

So while it's definitely an improvement for upstream, it's not tempting to switch from our patch that is still 50 times better than improved upstream.

lategoodbye, 2023-09-15:
I'm pretty aware of that and had discussions about the performance ( https://lore.kernel.org/linux-arm-kernel/ZOoe0lOR9zpcAw5I@zx2c4.com/T/ ) so i'm happy that at least there is a performance improvement. My intention wasn't that you drop the downstream solution. Sometimes you need make compromises in mainline kernel development. I just want to create a reference for interested users.

But to be honest the use case of copying of 10000 random bytes direct from the hardware random generator isn't realistic.

popcornmix, 2023-09-15:
I did need to when collecting data for rngtest/dieharder.
But I agree the average user won't.