raspberrypi/linux #5395: OOM killer triggers very early/aggressively on armv7 6.1.x

gavinbeatty, 2023-03-22:
### Describe the bug

On armv7 builds (but not aarch64 ones), OOM killer triggers and kills numerous processes when memory usage goes above a certain threshold, but that threshold is far below the volatile memory available (the threshold is not consistent, I have experienced it anywhere from 10-50% of volatile memory). This seems to coincide with my distribution changing to a 6.1.x kernel in the last 2 months, and definitely did not happen for any 5.15.x kernel. (And to repeat, this does not occur for 6.1.x aarch64)

The threshold is really very low: just running a zsh shell with zimfw might trigger it. Just running a low usage gitea service occasionally triggers it. It is impossible to run "heavier" software such as media servers as they always trigger a cascade of OOM kills. While creating this issue just now, I experienced it when running `inxi -F`, and was locked out of the system (it's common for both systemd and ssh to be killed when it triggers), and had to hard reset power.

### Steps to reproduce the behaviour

- I am using archlinuxarm, installed according to the armv7 instructions https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-4)
- Install a 6.1.x build of this kernel (`pacman -S linux-rpi linux-rpi-headers`)
- Install some software than can use a decent amount of memory (e.g., `pacman -S ffmpeg`)
- Start process(es) that use a few hundred MBs of memory. I made a loop to spawn 30 concurrent processes re-encoding video, with `ffmpeg -i ./example.mkv ./example1.mp4 &`, and a `sleep 1` in between
- Watch memory usage gradually increase as additional ffmpeg processes start, using htop or btm or free in a loop
- If logged in via ssh/mosh, prepare for disconnect or timeout as a number of important processes are OOM killed, including sshd/mosh-server
- Note that memory usage never approached full when OOM kills were triggered

### Device (s)

Raspberry Pi 4 Mod. B

### System

Archlinuxarm armv7h

```
uname -a
Linux dill 6.1.20-1-rpi-ARCH #1 SMP Tue Mar 21 12:49:06 MDT 2023 armv7l GNU/Linux
```

```
grep Model /proc/cpuinfo
Model           : Raspberry Pi 4 Model B Rev 1.1
```

```
vcgencmd version
Mar 17 2023 10:50:39
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
```

```
rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Wed Jan 11 05:40:52 PM UTC 2023 (1673458852)
    LATEST: Wed Jan 11 05:40:52 PM UTC 2023 (1673458852)
   RELEASE: critical (/lib/firmware/raspberrypi/bootloader/critical)

  VL805_FW: Dedicated VL805 EEPROM
     VL805: version unknown. Try sudo rpi-eeprom-update
   CURRENT:
    LATEST: 000138c0
```

### Logs

From journalctl (wrapped)
```
Mar 22 09:34:30 dill kernel: Out of memory: Killed process 2169 (gitea) total-vm:843364kB, anon-rss:80128kB, file-rss:65252kB, shmem-rss:0kB, UID:972 pgtables:412kB oom_score_adj:0
Mar 22 09:34:30 dill kernel: bash invoked oom-killer: gfp_mask=0x400dc0(GFP_KERNEL_ACCOUNT|__GFP_ZERO), order=0, oom_score_adj=0
Mar 22 09:34:30 dill kernel: CPU: 2 PID: 2241 Comm: bash Tainted: P         C O       6.1.20-1-rpi-ARCH #1
Mar 22 09:34:30 dill kernel: Hardware name: BCM2711
Mar 22 09:34:30 dill kernel:  unwind_backtrace from show_stack+0x18/0x1c
Mar 22 09:34:30 dill kernel:  show_stack from dump_stack_lvl+0x90/0xac
Mar 22 09:34:30 dill kernel:  dump_stack_lvl from dump_header+0x54/0x1fc
Mar 22 09:34:30 dill kernel:  dump_header from oom_kill_process+0x23c/0x248
Mar 22 09:34:30 dill kernel:  oom_kill_process from out_of_memory+0x218/0x34c
Mar 22 09:34:30 dill kernel:  out_of_memory from __alloc_pages+0xa98/0x1044
Mar 22 09:34:30 dill kernel:  __alloc_pages from __pmd_alloc+0x3c/0x1d8
Mar 22 09:34:30 dill kernel:  __pmd_alloc from pgd_alloc+0x2e0/0x32c
[... 73 lines cut ...]
Mar 22 09:34:30 dill kernel: [   1794]  1000  1794     2309     1061    45056        0             0 bash
Mar 22 09:34:30 dill kernel: [   1805]  1000  1805   106637    31616   561152        0             0 ffmpeg
Mar 22 09:34:30 dill kernel: [   1822]  1000  1822   106603    31717   557056        0             0 ffmpeg
Mar 22 09:34:30 dill kernel: [   1868]  1000  1868   106564    31560   557056        0             0 ffmpeg
Mar 22 09:34:30 dill kernel: [   1883]  1000  1883   106530    31550   552960        0             0 ffmpeg
Mar 22 09:34:30 dill kernel: [   2000]  1000  2000   106530    31352   565248        0             0 ffmpeg
Mar 22 09:34:30 dill kernel: [   2018]  1000  2018   106456    31021   561152        0             0 ffmpeg
Mar 22 09:34:30 dill kernel: [   2033]  1000  2033     2309     1061    45056        0             0 bash
Mar 22 09:34:30 dill kernel: [   2098]  1000  2098   106456    30924   561152        0             0 ffmpeg
Mar 22 09:34:30 dill kernel: [   2102]  1000  2102     2309     1061    49152        0             0 bash
Mar 22 09:34:30 dill kernel: [   2129]  1000  2129     2309     1055    40960        0             0 bash
Mar 22 09:34:30 dill kernel: [   2136]  1000  2136     2309     1061    40960        0             0 bash
Mar 22 09:34:30 dill kernel: [   2145]  1000  2145     2309     1061    36864        0             0 bash
Mar 22 09:34:30 dill kernel: [   2150]  1000  2150     2309     1061    45056        0             0 bash
Mar 22 09:34:30 dill kernel: [   2159]  1000  2159     2309     1055    45056        0             0 bash
Mar 22 09:34:30 dill kernel: [   2178]  1000  2178     2309     1061    40960        0             0 bash
Mar 22 09:34:30 dill kernel: [   2208]  1000  2208    90682    12887   430080        0             0 ffmpeg
Mar 22 09:34:30 dill kernel: [   2214]  1000  2214    33643     3968   200704        0             0 ffmpeg
Mar 22 09:34:30 dill kernel: [   2241]  1000  2241     2309      247    28672        0             0 bash
Mar 22 09:34:30 dill kernel: oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=ffmpeg,     pid=1360,uid=1000
```

### Additional context

There are several other reports of this issue in this archlinuxarm forum post, along with lots more logs and `/boot/*.txt`, etc., and some info from one of the maintainers, who cannot find a distro specific cause: https://archlinuxarm.org/forum/viewtopic.php?f=23&t=16377

gavinbeatty, 2023-03-22:
For some reason, a very reliable way I've been able to trigger it is to reinstall mkinitcpio (`pacman -S mkinitcpio`)

popcornmix, 2023-03-22:
RPiOS is our primary focus here. If you can reproduce your issue using a clean RPiOS image, then it will be a lot easier for us to reproduce.

My initial thought is MGLRU is one of the big new features of 6.1 kernel.
Firstly is it enabled on archlinux? On RPiOS it is:
```
$ cat /sys/kernel/mm/lru_gen/enabled
0x0001
```
if it is you could try disabling it and see if behaviour changes.
```
$ echo 0 | sudo tee /sys/kernel/mm/lru_gen/enabled
0
```

gavinbeatty, 2023-03-22:
I'll try disabling that now, just going to post the following info that I wrote before I saw your post:

Looking in more detail at each stacktrace, including lots of new ones I've been able to generate thanks to being able to reliably cause it with pacman reinstalls, it seems to be a kernel issue where the following happens:

- userspace makes a filesystem syscall like statx, openat, faccessat, etc.
- after context switch to kernel, it ends up calling __alloc_pages which spuriously fails and calls either out_of_memory or warn_alloc
- the userspace process is killed

So systemd is not involved at all, and it can be triggered with non-destructive and non-memory intensive programs. Filesystem use is actually the issue.

I can now reliably reproduce it with a simple find command that issues stat-type syscalls, like `find / -amin -1`.

pelwell, 2023-03-22:
If you are using one of the `vc4-*-v3d` overlays, does appending `,cma-320` make any difference?

gavinbeatty, 2023-03-22:
@popcornmix Thanks, it was enabled, and after disabling, the OOMs are far less frequent, but not entirely gone. I re-enabled the media server processes, etc., and was testing it out. For a while, everything worked fine, but I did eventually see the same "syscall leads to __alloc_pages, out_of_memory" chain, this time with sendto (so not the VFS layer). Previously this would occur within 1 second of running the media server.

```
Mar 22 13:49:47 dill kernel: Hardware name: BCM2711
Mar 22 13:49:47 dill kernel:  unwind_backtrace from show_stack+0x18/0x1c
Mar 22 13:49:47 dill kernel:  show_stack from dump_stack_lvl+0x90/0xac
Mar 22 13:49:47 dill kernel:  dump_stack_lvl from dump_header+0x54/0x1fc
Mar 22 13:49:47 dill kernel:  dump_header from oom_kill_process+0x23c/0x248
Mar 22 13:49:47 dill kernel:  oom_kill_process from out_of_memory+0x218/0x34c
Mar 22 13:49:47 dill kernel:  out_of_memory from __alloc_pages+0xa98/0x1044
Mar 22 13:49:47 dill kernel:  __alloc_pages from skb_page_frag_refill+0xd4/0xf8
Mar 22 13:49:47 dill kernel:  skb_page_frag_refill from sk_page_frag_refill+0x1c/0x94
Mar 22 13:49:47 dill kernel:  sk_page_frag_refill from tcp_sendmsg_locked+0x1fc/0xa9c
Mar 22 13:49:47 dill kernel:  tcp_sendmsg_locked from tcp_sendmsg+0x30/0x44
Mar 22 13:49:47 dill kernel:  tcp_sendmsg from sock_sendmsg+0x3c/0x4c
Mar 22 13:49:47 dill kernel:  sock_sendmsg from __sys_sendto+0x110/0x154
Mar 22 13:49:47 dill kernel:  __sys_sendto from __sys_trace_return+0x0/0x10
Mar 22 13:49:47 dill kernel: Exception stack(0xf4759fa8 to 0xf4759ff0)
Mar 22 13:49:47 dill kernel: 9fa0:                   00000000 00000000 00000028 b26e08e0 00010000 00004000
Mar 22 13:49:47 dill kernel: 9fc0: 00000000 00000000 00000000 00000122 b24c5bcc b26e08e0 00000000 b26e08e0
Mar 22 13:49:47 dill kernel: 9fe0: b24c5b38 b24c5b28 b6d17f37 b6d176f8
Mar 22 13:49:47 dill kernel: Mem-Info:
Mar 22 13:49:47 dill kernel: active_anon:621 inactive_anon:84478 isolated_anon:0
                              active_file:36887 inactive_file:53736 isolated_file:27
                              unevictable:395 dirty:7615 writeback:69
                              slab_reclaimable:4231 slab_unreclaimable:23640
                              mapped:35696 shmem:2253 pagetables:834
                              sec_pagetables:0 bounce:0
                              kernel_misc_reclaimable:0
                              free:755526 free_pcp:70 free_cma:129347
Mar 22 13:49:47 dill kernel: Node 0 active_anon:2484kB inactive_anon:337912kB active_file:147548kB inactive_file:214944kB unevictable:1580kB isola
ted(anon):0kB isolated(file):108kB mapped:142784kB dirty:30460kB writeback:276kB shmem:9012kB writeback_tmp:0kB kernel_stack:3344kB pagetables:333
6kB sec_pagetables:0kB all_unreclaimable? no
Mar 22 13:49:47 dill kernel: DMA free:531136kB boost:0kB min:3232kB low:4040kB high:4848kB reserved_highatomic:4096KB active_anon:0kB inactive_ano
n:0kB active_file:528kB inactive_file:0kB unevictable:0kB writepending:44kB present:786432kB managed:663844kB mlocked:0kB bounce:0kB free_pcp:280k
B local_pcp:0kB free_cma:517388kB
Mar 22 13:49:47 dill kernel: lowmem_reserve[]: 0 0 3188 3188
Mar 22 13:49:47 dill kernel: DMA: 583*4kB (UEHC) 856*8kB (UEHC) 195*16kB (UEHC) 65*32kB (UEHC) 30*64kB (UHC) 10*128kB (HC) 4*256kB (C) 7*512kB (C)
 3*1024kB (C) 3*2048kB (C) 122*4096kB (C) = 531116kB
Mar 22 13:49:47 dill kernel: 92358 total pagecache pages
Mar 22 13:49:47 dill kernel: 0 pages in swap cache
Mar 22 13:49:47 dill kernel: Free swap  = 16777212kB
Mar 22 13:49:47 dill kernel: Total swap = 16777212kB
Mar 22 13:49:47 dill kernel: 1012736 pages RAM
Mar 22 13:49:47 dill kernel: 816128 pages HighMem/MovableOnly
Mar 22 13:49:47 dill kernel: 30647 pages reserved
Mar 22 13:49:47 dill kernel: 131072 pages cma reserved
```

For now, I've made that config permanent by running `systemd-tmpfiles --create sysfs-memory-fix.conf` with the following config:

```
# /etc/tmpfiles.d/sysfs-memory-fix.conf
w- /sys/kernel/mm/lru_gen/enabled - - - - 0
```

@pelwell Thanks for the suggestion, I will try that shortly.

gavinbeatty, 2023-03-22:
@pelwell Hard to say conclusively, but I think that is a further improvement. With the CMA change, I have experienced one crash, originating in sendto again -- but just one. I can't say for certain it had an effect given I've only let it run for an hour each with "mm/lru_gen disabled" and "mm/lru_gen disabled + cma-320" and might have just been unlucky with the former.

And just to be clear, I originally had `dtoverlay=vc4-kms-v3d`; now `dtoverlay=vc4-kms-v3d,cma-320`.

I'll ask the archlinuxarm maintainers if all this information points to any distro-specific issues. I'm not sure I'll be able to try a fresh repro in RPiOS soon, but I'm happy to try more suggestions with my existing install.

gavinbeatty, 2023-03-22:
By the way, what is the default linux kernel version for RPiOS right now? Is it < 6.1.x?

graysky2, 2023-03-22:
@popcornmix - I maintain the linux-rpi kernel for Arch ARM.  The config is RPiOS's `bcm2711_defconfig` target, see [here](https://github.com/archlinuxarm/PKGBUILDs/blob/master/core/linux-rpi/PKGBUILD#L63-L71).  I then append some options to that before running `make oldconfig` to harmonize/setup the distro config.  This was a suggestion by @pelwell a while ago.

Those are the only changes vs. RPiOS's  `bcm2711_defconfig` with the exception of enabling `CONFIG_BCM2835_THERMAL=y` which cannot be set by the method.  It is set after by call to scripts/config as you see [here](https://github.com/archlinuxarm/PKGBUILDs/blob/master/core/linux-rpi/PKGBUILD#L70).  In any case, I do not believe anything in [archarm.diffconfig]([modules/options](https://github.com/archlinuxarm/PKGBUILDs/blob/master/core/linux-rpi/archarm.diffconfig)) would be driving OOM triggers.

MGLRU is enabled on Arch ARM/armv7h and aarch64:
```
% cat /sys/kernel/mm/lru_gen/enabled 
0x0001

% zgrep -i lru /proc/config.gz 
CONFIG_LRU_GEN=y
CONFIG_LRU_GEN_ENABLED=y
# CONFIG_LRU_GEN_STATS is not set
CONFIG_LRU_CACHE=m
```

gavinbeatty, 2023-03-23:
In the new "mm/lru_gen disabled + cma-320" configuration I have a novel __alloc_pages failure:

```
Mar 22 21:10:35 dill earlyoom[361]: mem avail:  3056 of  3836 MiB (79.66%), swap free: 16373 of 16383 MiB (99.94%)
Mar 22 21:11:57 dill kernel: ksoftirqd/0: page allocation failure: order:0, mode:0xa20(GFP_ATOMIC), nodemask=(null),cpuset=/,mems_allowed=0
Mar 22 21:11:57 dill kernel: CPU: 0 PID: 13 Comm: ksoftirqd/0 Tainted: P         C O       6.1.20-1-rpi-ARCH #1
Mar 22 21:11:57 dill kernel: Hardware name: BCM2711
Mar 22 21:11:57 dill kernel:  unwind_backtrace from show_stack+0x18/0x1c
Mar 22 21:11:57 dill kernel:  show_stack from dump_stack_lvl+0x90/0xac
Mar 22 21:11:57 dill kernel:  dump_stack_lvl from warn_alloc+0x110/0x1b0
Mar 22 21:11:57 dill kernel:  warn_alloc from __alloc_pages+0xef8/0x1044
Mar 22 21:11:57 dill kernel:  __alloc_pages from new_slab+0x384/0x43c
Mar 22 21:11:57 dill kernel:  new_slab from ___slab_alloc+0x3e8/0xa0c
Mar 22 21:11:57 dill kernel:  ___slab_alloc from kmem_cache_alloc+0x470/0x4d4
Mar 22 21:11:57 dill kernel:  kmem_cache_alloc from __netdev_alloc_skb+0xe4/0x198
Mar 22 21:11:57 dill kernel:  __netdev_alloc_skb from bcmgenet_rx_refill+0x38/0x1ec
Mar 22 21:11:57 dill kernel:  bcmgenet_rx_refill from bcmgenet_rx_poll+0x1d4/0x468
Mar 22 21:11:57 dill kernel:  bcmgenet_rx_poll from __napi_poll.constprop.0+0x34/0x1d0
Mar 22 21:11:57 dill kernel:  __napi_poll.constprop.0 from net_rx_action+0x344/0x3b0
Mar 22 21:11:57 dill kernel:  net_rx_action from __do_softirq+0x15c/0x3dc
Mar 22 21:11:57 dill kernel:  __do_softirq from run_ksoftirqd+0x44/0x5c
Mar 22 21:11:57 dill kernel:  run_ksoftirqd from smpboot_thread_fn+0xc0/0x15c
Mar 22 21:11:57 dill kernel:  smpboot_thread_fn from kthread+0xd8/0xf4
Mar 22 21:11:57 dill kernel:  kthread from ret_from_fork+0x14/0x34
Mar 22 21:11:57 dill kernel: Exception stack(0xf085dfb0 to 0xf085dff8)
Mar 22 21:11:57 dill kernel: dfa0:                                     00000000 00000000 00000000 00000000
Mar 22 21:11:57 dill kernel: dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar 22 21:11:57 dill kernel: dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
Mar 22 21:11:57 dill kernel: Mem-Info:
Mar 22 21:11:57 dill kernel: active_anon:1427 inactive_anon:93802 isolated_anon:0
                              active_file:84493 inactive_file:606758 isolated_file:1
                              unevictable:395 dirty:0 writeback:10
                              slab_reclaimable:14479 slab_unreclaimable:39039
                              mapped:34487 shmem:467 pagetables:937
                              sec_pagetables:0 bounce:0
                              kernel_misc_reclaimable:0
                              free:85635 free_pcp:209 free_cma:80196
Mar 22 21:11:57 dill kernel: Node 0 active_anon:5708kB inactive_anon:375208kB active_file:337972kB inactive_file:2427032kB unevictable:1580kB isolated(anon):0kB isolated(file):4kB mapped:137948kB dirty:0kB writeback:40kB shmem:1868kB writeback_tmp:0kB kernel_stack:2992kB pagetables:3748kB sec_pagetables:0kB all_unreclaimable? no
Mar 22 21:11:57 dill kernel: DMA free:321404kB boost:7272kB min:10504kB low:11312kB high:12120kB reserved_highatomic:0KB active_anon:0kB inactive_anon:40kB active_file:216kB inactive_file:0kB unevictable:0kB writepending:28kB present:786432kB managed:664228kB mlocked:0kB bounce:0kB free_pcp:836kB local_pcp:68kB free_cma:320784kB
Mar 22 21:11:57 dill kernel: lowmem_reserve[]: 0 0 3188 3188
Mar 22 21:11:57 dill kernel: DMA: 117*4kB (UEC) 21*8kB (EC) 26*16kB (UEC) 21*32kB (UEC) 13*64kB (C) 15*128kB (C) 12*256kB (C) 5*512kB (C) 4*1024kB (C) 4*2048kB (C) 73*4096kB (C) = 321404kB
Mar 22 21:11:57 dill kernel: 691807 total pagecache pages
Mar 22 21:11:57 dill kernel: 148 pages in swap cache
Mar 22 21:11:57 dill kernel: Free swap  = 16766460kB
Mar 22 21:11:57 dill kernel: Total swap = 16777212kB
Mar 22 21:11:57 dill kernel: 1012736 pages RAM
Mar 22 21:11:57 dill kernel: 816128 pages HighMem/MovableOnly
Mar 22 21:11:57 dill kernel: 30551 pages reserved
Mar 22 21:11:57 dill kernel: 81920 pages cma reserved
Mar 22 21:11:57 dill kernel: SLUB: Unable to allocate memory on node -1, gfp=0xa20(GFP_ATOMIC)
Mar 22 21:11:57 dill kernel:   cache: skbuff_head_cache, object size: 184, buffer size: 192, default order: 0, min order: 0
Mar 22 21:11:57 dill kernel:   node 0: slabs: 48, objs: 1008, free: 0
Mar 22 21:11:57 dill kernel: SLUB: Unable to allocate memory on node -1, gfp=0xa20(GFP_ATOMIC)
Mar 22 21:11:57 dill kernel:   cache: skbuff_head_cache, object size: 184, buffer size: 192, default order: 0, min order: 0
Mar 22 21:11:57 dill kernel:   node 0: slabs: 48, objs: 1008, free: 0
Mar 22 21:11:57 dill kernel: SLUB: Unable to allocate memory on node -1, gfp=0xa20(GFP_ATOMIC)
Mar 22 21:11:57 dill kernel:   cache: skbuff_head_cache, object size: 184, buffer size: 192, default order: 0, min order: 0
Mar 22 21:11:57 dill kernel:   node 0: slabs: 48, objs: 1008, free: 0
Mar 22 21:11:57 dill kernel: SLUB: Unable to allocate memory on node -1, gfp=0xa20(GFP_ATOMIC)
Mar 22 21:11:57 dill kernel:   cache: skbuff_head_cache, object size: 184, buffer size: 192, default order: 0, min order: 0
Mar 22 21:11:57 dill kernel:   node 0: slabs: 48, objs: 1008, free: 0
Mar 22 21:11:57 dill kernel: SLUB: Unable to allocate memory on node -1, gfp=0xa20(GFP_ATOMIC)
[... cut ...]
Mar 22 21:11:57 dill kernel:   cache: skbuff_head_cache, object size: 184, buffer size: 192, default order: 0, min order: 0
Mar 22 21:11:57 dill kernel:   node 0: slabs: 48, objs: 1008, free: 0
Mar 22 21:11:57 dill kernel: SLUB: Unable to allocate memory on node -1, gfp=0xa20(GFP_ATOMIC)
Mar 22 21:11:57 dill kernel:   cache: skbuff_head_cache, object size: 184, buffer size: 192, default order: 0, min order: 0
Mar 22 21:11:57 dill kernel:   node 0: slabs: 48, objs: 1008, free: 0
```

I didn't notice any system degradation, and `[ksoftirqd/0]` is still present in `ps aux`, but seems related.

popcornmix, 2023-03-23:
>By the way, what is the default linux kernel version for RPiOS right now? Is it < 6.1.x?

An updated RPiOS bullseye is using the 6.1 kernel.
And we are not seeing large numbers of OOM reports on RPiOS.

flipreverse, 2023-03-23:
@popcornmix I'm running Raspbian 11 on armv7l, and seeing the same issue: The oom killer runs way to earlier. I can hardly clone the linux kernel soruces.
I'm switching back to aarch64, and try that one.

popcornmix, 2023-03-24:
does running
```echo 0 | sudo tee /sys/kernel/mm/lru_gen/enabled```
before doing the failing operation help?

terminet85, 2023-03-24:
> does running
> `echo 0 | sudo tee /sys/kernel/mm/lru_gen/enabled`
> before doing the failing operation help?

Yeah that solved my OOM killer with btrbk.

graysky2, 2023-03-24:
> does running
echo 0 | sudo tee /sys/kernel/mm/lru_gen/enabled

Does this translate into `# CONFIG_LRU_GEN_ENABLED is not set` in the config?  Is that a good way to fix this bug at least on armv7h?

terminet85, 2023-03-25:
@graysky2 I don't building kernel I'm using raspbian 11

popcornmix, 2023-03-25:
> @popcornmix Does this translate into # CONFIG_LRU_GEN_ENABLED is not set in the config?

Yes

> Is that a good way to fix this bug at least on armv7h?

I don't know if it's clear it's a bug. MGLRU changes the behaviour of a system in low memory conditions, and there are likely to be use cases that have preferable behaviour with this enabled or disabled.

There are definitely [performance improvements available](https://www.phoronix.com/news/MGLRU-July-2022-Performance) when this is enabled.

But, yes, changing that config setting should make the kernel go back to the previous behaviour (no MGLRU) and is something worth considering.

graysky2, 2023-03-25:
> I don't building kernel I'm using raspbian 11

For those using Arch ARM, please update and report back.  linux-rpi-6.1.21-2 for armv7h [has MGLRU compiled in but disabled by default](https://github.com/archlinuxarm/PKGBUILDs/commit/6cf36ea455ebecdb942843ab7118c66d7123aae6).

terminet85, 2023-03-25:
> > I don't building kernel I'm using raspbian 11
> 
> For those using Arch ARM, please update and report back. linux-rpi-6.1.21-2 for armv7h [has MGLRU compiled in but disabled by default](https://github.com/archlinuxarm/PKGBUILDs/commit/6cf36ea455ebecdb942843ab7118c66d7123aae6).

That is the same keeping this:
```
# cat  /sys/kernel/mm/lru_gen/enabled
0x0000
```

right?

toofishes, 2023-03-25:
I'm seeing this as well. Is everyone that is affected by this on a **4 GB or greater** RPi 4? I'm wondering if this is only triggered on higher memory setups. I saw this myself on a 4 GB Pi:

```
Mar 14 22:45:02 cork munin-node[17468]: 2023/03/14-22:45:02 CONNECT TCP Peer: "[::ffff:10.9.0.1]:54226" Local: "[::ffff:10.9.0.2]:4949"
Mar 14 22:45:04 cork kernel: find invoked oom-killer: gfp_mask=0xcd0(GFP_KERNEL|__GFP_RECLAIMABLE), order=0, oom_score_adj=0
Mar 14 22:45:04 cork kernel: CPU: 0 PID: 17513 Comm: find Tainted: G         C         6.1.16-3-rpi-ARCH #1
Mar 14 22:45:04 cork kernel: Hardware name: BCM2711
Mar 14 22:45:04 cork kernel:  unwind_backtrace from show_stack+0x18/0x1c
Mar 14 22:45:04 cork kernel:  show_stack from dump_stack_lvl+0x90/0xac
Mar 14 22:45:04 cork kernel:  dump_stack_lvl from dump_header+0x54/0x1fc
Mar 14 22:45:04 cork kernel:  dump_header from oom_kill_process+0x23c/0x248
Mar 14 22:45:04 cork kernel:  oom_kill_process from out_of_memory+0x218/0x34c
Mar 14 22:45:04 cork kernel:  out_of_memory from __alloc_pages+0xa98/0x1044
Mar 14 22:45:04 cork kernel:  __alloc_pages from new_slab+0x384/0x43c
...
Mar 14 22:45:04 cork kernel: Mem-Info:
Mar 14 22:45:04 cork kernel: active_anon:99244 inactive_anon:243 isolated_anon:0
                              active_file:39172 inactive_file:68153 isolated_file:0
                              unevictable:0 dirty:4 writeback:0
                              slab_reclaimable:5594 slab_unreclaimable:7341
                              mapped:24772 shmem:219 pagetables:1861
                              sec_pagetables:0 bounce:0
                              kernel_misc_reclaimable:0
                              free:741220 free_pcp:144 free_cma:129007
Mar 14 22:45:04 cork kernel: Node 0 active_anon:396976kB inactive_anon:972kB active_file:156688kB inactive_file:272612kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:99088kB dirty:16kB writeback:0kB shmem:876kB writeback_tmp:0kB kernel_stack:2104kB pagetables:7444kB sec_pagetables:0kB all_unreclaimable? yes
Mar 14 22:45:04 cork kernel: DMA free:520420kB boost:4096kB min:7328kB low:8136kB high:8944kB reserved_highatomic:0KB active_anon:0kB inactive_anon:0kB active_file:612kB inactive_file:76868kB unevictable:0kB writepending:16kB present:786432kB managed:664308kB mlocked:0kB bounce:0kB free_pcp:576kB local_pcp:380kB free_cma:516028kB
Mar 14 22:45:04 cork kernel: lowmem_reserve[]: 0 0 3136 3136
Mar 14 22:45:04 cork kernel: DMA: 423*4kB (UEC) 153*8kB (UEC) 74*16kB (UEC) 21*32kB (UEC) 5*64kB (C) 2*128kB (C) 2*256kB (C) 1*512kB (C) 0*1024kB 1*2048kB (C) 125*4096kB (C) = 520420kB
Mar 14 22:45:04 cork kernel: 107545 total pagecache pages
Mar 14 22:45:04 cork kernel: 0 pages in swap cache
Mar 14 22:45:04 cork kernel: Free swap  = 262140kB
Mar 14 22:45:04 cork kernel: Total swap = 262140kB
Mar 14 22:45:04 cork kernel: 999424 pages RAM
Mar 14 22:45:04 cork kernel: 802816 pages HighMem/MovableOnly
Mar 14 22:45:04 cork kernel: 30531 pages reserved
Mar 14 22:45:04 cork kernel: 131072 pages cma reserved
...
Mar 14 22:45:04 cork kernel: oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=homebridge,pid=389,uid=1001
Mar 14 22:45:04 cork kernel: Out of memory: Killed process 389 (homebridge) total-vm:249024kB, anon-rss:77820kB, file-rss:43336kB, shmem-rss:0kB, UID:1001 pgtables:1080kB oom_score_adj:200
Mar 14 22:45:04 cork kernel: systemd-journal invoked oom-killer: gfp_mask=0xcd0(GFP_KERNEL|__GFP_RECLAIMABLE), order=0, oom_score_adj=-250
Mar 14 22:45:04 cork kernel: CPU: 3 PID: 200 Comm: systemd-journal Tainted: G         C         6.1.16-3-rpi-ARCH #1
Mar 14 22:45:04 cork kernel: Hardware name: BCM2711
Mar 14 22:45:04 cork kernel:  unwind_backtrace from show_stack+0x18/0x1c
Mar 14 22:45:04 cork kernel:  show_stack from dump_stack_lvl+0x90/0xac
Mar 14 22:45:04 cork kernel:  dump_stack_lvl from dump_header+0x54/0x1fc
Mar 14 22:45:04 cork kernel:  dump_header from oom_kill_process+0x23c/0x248
Mar 14 22:45:04 cork kernel:  oom_kill_process from out_of_memory+0x218/0x34c
Mar 14 22:45:04 cork kernel:  out_of_memory from __alloc_pages+0xa98/0x1044
Mar 14 22:45:04 cork kernel:  __alloc_pages from new_slab+0x384/0x43c
...
```

rvk01, 2023-03-25:
> Is everyone that is affected by this on a **4 GB or greater** RPi 4?

Yes, for me on a 8 GB RPI 4. I went back to the 64 bit kernel where this is not a problem. But the 64 bit kernel in combination with 32 bit userland gives problems with PHP/PDO_Firebird for which a fix is pending. But now I also have problems with docker (memory crashes during startup of a blanc nginx container). I want to go back to the 32 bit kernel but the OOM-killer was so aggressive that of my 8GB it killed ALL processes even upto where there was only 100MB used. It just kept on killing everything.

Maybe I should just bite the bullet and go full 64 bit.

gavinbeatty, 2023-03-25:
@graysky2 I deleted my armv7h instance's tmpfiles.d config, and upgraded my aarch64 and armv7h instances:

- armv7h: `cat /sys/kernel/mm/lru_gen/enabled` shows `0x0000` as desired/expected, no OOM issues so far ✔️ 
- aarch64: `cat /sys/kernel/mm/lru_gen/enabled` shows `0x0001` as desired/expected, never had OOM issues there, so that seems best ✔️ 

@toofishes 4GB model, yes. It would be interesting if anyone with a <4GB model could run `sudo sh -c "while sleep 1 ; do find / -amin -1 ; done"` for a while to see if it's possible to trigger it there. But I would guess the <4GB models are far less popular, so absence of reports might not mean much.

@popcornmix There are at least two reports of OOM issues on RPiOS/Raspbian here now. I think it's likely that there is an unsigned overflow issue in the MGLRU.

terminet85, 2023-03-26:
> that is affected by this on a **4 GB or greater** RPi 4

Mine is 8Gb

Keeping /sys/kernel/mm/lru_gen/enabled to 0 seems like OOM Killer doesn't come up.
I'll go ahead and report.

rajo, 2023-03-28:
RPi4 user with 4GB here, running bullseye. After the latest update, 6.1.19-v7l+ was installed and I am victim of the oom-killer as well. As I use the RPi as a NAS and smart home central and many other things, oom basically behaves like a shotgun and randomly kills services and renders it pretty unusable. Switching to aarch64 isn't an option either, as - for whatever reason - the kernel-headers aren't available for it, so I cannot compile neccessary addtional modules for my smart home stuff. 

For now, I'm trying with disabling the lru__gen stuff, fingers crossed that this will help. (But I would appreciate a more proper fix here)

terminet85, 2023-03-28:
> For now, I'm trying with disabling the lru__gen stuff, fingers crossed that this will help. (But I would appreciate a more proper fix here)

Rock solid from almost few days with lru_gen disabled. MGLRU is pretty involved...

flipreverse, 2023-03-28:
> RPi4 user with 4GB here, running bullseye. After the latest update, 6.1.19-v7l+ was installed and I am victim of the oom-killer as well. As I use the RPi as a NAS and smart home central and many [...]
Providing those headers isn't a hard task. But annoying for sure.

@popcornmix I have a 4GB Pi4 as well. I'll give the workaround a try, and report back.

flipreverse, 2023-03-28:
> I note that some users have reported running 6.1.19-v7l+ (i.e. 32-bit kernel) and gettin OOM.
> 
> Can users with this issue confirm if you are running: 32-bit distro + 32-bit kernel 32-bit distro + 64-bit kernel 64-bit distro + 64-bit kernel
> 
> If the OOM issue is occurring on 32-bit kernel, does it still occur with 64-bit kernel?

I'm running a 32-bit distro. Using a 32-bit kernel, the OOM issue occurs. The issue does not occur when running the 64-bit kernel.

rvk01, 2023-03-28:
On raspberry pi 4 8GB.
32-bit distro + 32-bit kernel = OOM-killer kills everything to 100MB
32-bit distro + 64-bit kernel = no issues
64-bit distro + 64-bit kernel = don't know

I did have an issue earlier with "32-bit distro + 64-bit kernel" but that's resolved (PDO Firebird driver issues). The Docker issues came from earlier attempts to disable the OOM-killer but it's working fine now.

graysky2, 2023-03-28:
@popcornmix - I haven't tried that before.  @kmihelich - is this possible?

kmihelich, 2023-03-29:
> @popcornmix - I haven't tried that before. @kmihelich - is this possible?

You can try with: pacman -U --arch aarch64 linux-rpi-6.1.21-2-aarch64.pkg.tar.xz

josepmaria79, 2023-03-29:
I upgraded to latest linux-rpi-6.1.21-2-armv7h.pkg.tar.xz and oom kills triggered aggressively the same way as previous 6.1 kernels.
Instead, I tried with linux-rpi-5.15.84-1-armv7h.pkg.tar.xz from 2022 and oom kills triggered anyway, but not so often.
Finally, I tried with linux-rpi-5.10.83-1-armv7h.pkg.tar.xz from 2021 and oom kills disappeared completely!

terminet85, 2023-03-29:
> If the OOM issue is occurring on 32-bit kernel, does it still occur with 64-bit kernel?

Switched to 64bit kernel with a 32bit userland and OOM Killer issue doesn't come up. Instead on kernel/userland 32 bit, without lru_gen occurs. 

I switched to 64bit kernel 'cause would be the default setup to DietPI distro.

josepmaria79, 2023-03-29:
> I upgraded to latest linux-rpi-6.1.21-2-armv7h.pkg.tar.xz and oom kills triggered aggressively the same way as previous 6.1 kernels. Instead, I tried with linux-rpi-5.15.84-1-armv7h.pkg.tar.xz from [...]

Interesting that issue happens with old 5.15 but not with 5.10
Why?

rajo, 2023-03-29:
32bit/32bit here - after 24 hours I can confirm that disabling lru is helping as OOM didn't occur any more.

popcornmix, 2023-03-30:
Based on feedback here, we've disabled `LRU_GEN_ENABLED` for 32-bit kernel, but left it enabled for 64-bit,
as it seemed the reports of OOM were predominantly using 32-bit kernels.

We'll continue to keep an eye for OOM reports, and if there are any changes that sound like MGLRU on 32-bit behaviour may be improved and can be reconsidered.

To clarify, once the source changes reach built kernels (rpi-update build is in progress):
On 32-bit kernels MGLRU is disabled by default and can be enabled manually with
```
echo 1 | sudo tee /sys/kernel/mm/lru_gen/enabled
```
On 64-bit kernels MGLRU is enabled by default and can be disabled manually with
```
echo 0 | sudo tee /sys/kernel/mm/lru_gen/enabled
```

lategoodbye, 2023-03-30:
Anyone able to reproduce the OOM issue on a Raspberry Pi 3 (Plus)?

Since this issue seems specific for Raspberry Pi 4 32 bit, maybe the issue is related to `CONFIG_ARM_LPAE`.

graysky2, 2023-03-30:
Thanks for the support, all.  For the record, Arch ARM has it compiled in but not-enabled by default for armv7h.  For aarch64, it is both compiled in and enabled.

pelwell, 2023-03-30:
Any more datapoints would be useful. Please state model, memory, 32- or 64-bit, OOM yes or no.

graysky2, 2023-03-30:
RPi4B/8G, 64-bit, no OOM
RPi4B/4G, 64-bit, no OOM
RPi4B/2G, 64-bit, no OOM

yuzhaogoogle, 2023-04-17:
Hi all,

Based on the logs in this thread, e.g.,

```
Mar 22 09:34:30 dill kernel: 1012736 pages RAM
Mar 22 09:34:30 dill kernel: 816128 pages HighMem/MovableOnly
```

It seems the 32-bit kernel was built with the default `CONFIG_ VMSPLIT_3G`. Long story short: the OOM kills were caused by the depletion of the cold memory in the DMA zone because its size was too small. On the other hand, the highmem zone was too large.

This can cause unnecessary OOM kills for both MGLRU and non-MGLRU configurations on 32-bit platforms. But MGLRU is more prone to it (OpenWrt reported [this](https://github.com/openwrt/openwrt/pull/12227#issuecomment-1479074065) too).

Is it possible for Pi to switch to `CONFIG_ VMSPLIT_2G`? @popcornmix 

And please feel free to tag me. Thanks.

pelwell, 2023-04-17:
@yuzhaogoogle As you may know, we use `CONFIG_ VMSPLIT_2G` on Pi 2 and Pi 3, where limiting the application address space to 2GB (as it does) has less effect because they can only address 1GB of RAM in total. Pi 4s exist with 4 and 8GB of RAM, so a 2GB address space limit will prevent some large applications from running that would otherwise have worked. I welcome @popcornmix's input here, but the preferred route forward has to be moving to a 64-bit kernel with none of these addressing limitation (but a few compatibility issues for some ioctl interfaces when running with a 32-bit userspace, as we're discovering).

popcornmix, 2023-04-17:
Yes, 64-bit for Pi4 is the preferred option

If someone manually switches to 32-bit kernel on Pi4 (for compatibility with code or drivers that handle mixed userland/kernel badly) then it seems there are two options (that probably avoid OOM):
CONFIG_ VMSPLIT_3G with no MGLRU
CONFIG_ VMSPLIT_2G with MGLRU

As this setup is likely to affect a very small percentage of users it may not be too critical, and the former option has the advantage of being well tested.

This sounds like further evidence that the OOM issue here doesn't affect Pi0-3, so I'm still keen to enable MGLRU again on those platforms (in rpi-update and we'll keep an eye out for issues reported).

yuzhaogoogle, 2023-04-17:
> Pi 4s exist with 4 and 8GB of RAM, so a 2GB address space limit will prevent some large applications from running that would otherwise have worked.

Yes, this makes senses.

> I welcome @popcornmix's input here, but the preferred route forward has to be moving to a 64-bit kernel with none of these addressing limitation (but a few compatibility issues for some ioctl interfaces when running with a 32-bit userspace, as we're discovering).

Agreed.

rsekman, 2023-04-25:
> Any more datapoints would be useful. Please state model, memory, 32- or 64-bit, OOM yes or no.

```
[karl@schwarzschild] ~% uname -a
Linux schwarzschild 6.1.24-2-rpi-ARCH #1 SMP Fri Apr 21 07:31:33 MDT 2023 armv7l GNU/Linux
[karl@schwarzschild] ~% grep Model /proc/cpuinfo
Model           : Raspberry Pi 4 Model B Rev 1.1
[karl@schwarzschild] ~% sudo vcgencmd version
Mar 21 2023 17:18:16
Copyright (c) 2012 Broadcom
version 3cc1c2dfc5460da9e1a0a4f48b48ab508c48bfe5 (clean) (release) (start)
[karl@schwarzschild] ~% sudo rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: critical (/lib/firmware/raspberrypi/bootloader/critical)

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

Getting very aggressive OOM-kills. Even just adding a torrent to transmission can bring down the entire system as the transmission daemon restarting after being killed uses a fair amount of memory, triggering more OOM-kills including itself, and so on. Never seen memory usage >20% with `top`.

yuzhaogoogle, 2023-04-25:
Yes, this one has the same root cause (Pi 4 + 32-bit kernel + MGLRU).

Now MGLRU is disabled on 32-bit kernel for Pi 4 (but not other models since they don't have this problem).

gavinbeatty, 2023-04-25:
They seem to be running Arch ARM which should already have MGLRU disabled on armv7l since 6.1.21-2. @graysky2 I haven't rebooted into any new kernel updates since your fix worked for me, but maybe the MGLRU default=disable was lost or became ineffective in 6.1.24-2

> For those using Arch ARM, please update and report back. linux-rpi-6.1.21-2 for armv7h [has MGLRU compiled in but disabled by default](https://github.com/archlinuxarm/PKGBUILDs/commit/6cf36ea455ebecdb942843ab7118c66d7123aae6).

gavinbeatty, 2023-04-25:
@rsekman Try the procedure in https://github.com/raspberrypi/linux/issues/5395#issuecomment-1479948663

rsekman, 2023-04-25:
> @rsekman Try the procedure in [#5395 (comment)](https://github.com/raspberrypi/linux/issues/5395#issuecomment-1479948663)

```
[karl@schwarzschild] ~% cat /sys/kernel/mm/lru_gen/enabled
0x0000
```

The 6.1.21-2 update made my system more stable. Previously it was entering OOM-kill loops just a few minutes after booting. They could also be triggered by very lightweight programs. But I am still getting OOM-kills during peak loads (which aren't that high).

popcornmix, 2023-04-26:
@rsekman it's not completely clear what you are reporting.
I think you are saying that with a previous kernel mglru was enabled an OOM was very aggressive.
With current kernel OOM is less common but still occurs.

If you think this is a kernel issue, then are you able to revert to an older kernel that has no OOMs when running a similar load?

rsekman, 2023-04-27:
Indeed. Before kernel version 6, never any OOM problems. After kernel version 6 but before 6.1.21-2 (MGLRU enabled), exceedingly aggressive OOM killing to where the system struggled to stay stable for minutes after booting. After 6.1.21-2 (MGLRU disabled), OOM killer less aggressive than before but still occassionally kneecaps or brings the entire system down during load spikes (which may be no more stressing than adding a torrent to transmission). Load profile has stayed unchanged throughout this entire history. Since the OOM kills happen way before actually running OOM this seems consistent with this issue and no other explanation. Someone else above was also reporting better but still not fixed behaviour since disabling MGLRU.

pelwell, 2023-04-27:
Are you able to say what the CMA settings were on the 5.15 and 6.1 kernels? The default in both cases is 64MB, but using the `vc4-kms-v3d`, `vc4-fkms-v3d` or `cma` overlays changes that.

toofishes, 2023-05-06:
Some more data points, using the 32 bit Arch Linux ARM kernel with a 32 bit userspace. It is the same kernel build across both Pi 3 and Pi 4, built with `CONFIG_VMSPLIT_3G=y`.

Pi 4 Model B Rev 1.2, 4 GB RAM:
* `5.15.84-1-rpi-ARCH` was up 52 days with no OOM issues
* `6.1.21-2` with MGLRU enabled had almost immediate OOM trouble
* `6.1.27-1-rpi-ARCH` with MGLRU disabled has been stable
* I am using `dtoverlay=vc4-kms-v3d` and I see `CmaTotal: 524288 kB` in `/proc/meminfo`.

Pi 3 Model B Rev 1.2, 1 GB RAM:
* `6.1.16-3-rpi-ARCH` with MGRLU enabled has been fully stable for 52 days
* I am using `dtoverlay=vc4-kms-v3d` and I see `CmaTotal: 262144 kB` in `/proc/meminfo`.

Pi 3 Model B Rev 1.2, 1 GB RAM (different one):
* `6.1.19-2-rpi-ARCH` with MGRLU enabled has been fully stable for 48 days
* I am using `dtoverlay=vc4-kms-v3d` and I see `CmaTotal: 262144 kB` in `/proc/meminfo`.

rsekman, 2023-05-08:
> Are you able to say what the CMA settings were on the 5.15 and 6.1 kernels?

They would have been whatever they are in the 32 bit Arch Linux ARM kernel. I could grab the packages and check if you let me know where to look.

yuzhaogoogle, 2023-05-08:
> > Is everyone that is affected by this on a 4 GB or greater RPi 4?
> 
> Just bumped into it on a 2GB one. Was zeroing out my 10T USB HDD with `dd` and it happened after having written 4~5T, I think. Interestingly when I tried to re-run the `dd` command the OOM happen right away, even when `free -h` looks fine. (I re-tried twice, and the second time I lost ssh.)
> 
> Heading to reboot and disable MGLRU and see if I can complete the job.

The latest 32-bit kernel for Pi 4 should already have MGLRU disable by default. Are you using an old Pi 4 32-bit kernel or other kernels?

tomty89, 2023-05-08:
Yeah I am on `6.1.19-2-rpi-ARCH`, since the latest Arch ARM linux-rpi has some build script problem and I decided to revert and stick to this older cached version I have for now, which has led me to confirm that it happens on a 2G model as well, heh. I am on armv7 build btw.

resetko, 2023-05-15:
```
alarmpi /boot # uname -a
Linux alarmpi 6.1.27-1-rpi-ARCH #1 SMP Tue May  2 18:51:06 MDT 2023 armv7l GNU/Linux
alarmpi /boot # free -h
               total        used        free      shared  buff/cache   available
Mem:           7.7Gi       334Mi       6.6Gi       0.0Ki       851Mi       7.3Gi
Swap:             0B          0B          0B
alarmpi /boot # cat /sys/kernel/mm/lru_gen/enabled
0x0000
alarmpi /boot # zgrep -i lru /proc/config.gz
CONFIG_LRU_GEN=y
# CONFIG_LRU_GEN_ENABLED is not set
# CONFIG_LRU_GEN_STATS is not set
CONFIG_LRU_CACHE=m
```

Have quite often OOMs during io (downloading stuff, converting video, watching over DLNA)

yuzhaogoogle, 2023-05-15:
> ```
> alarmpi /boot # uname -a
> Linux alarmpi 6.1.27-1-rpi-ARCH #1 SMP Tue May  2 18:51:06 MDT 2023 armv7l GNU/Linux
> alarmpi /boot # free -h
>                total        used        free      shared  buff/cache   available
> Mem:           7.7Gi       334Mi       6.6Gi       0.0Ki       851Mi       7.3Gi
> Swap:             0B          0B          0B
> alarmpi /boot # cat /sys/kernel/mm/lru_gen/enabled
> 0x0000
> alarmpi /boot # zgrep -i lru /proc/config.gz
> CONFIG_LRU_GEN=y
> # CONFIG_LRU_GEN_ENABLED is not set
> # CONFIG_LRU_GEN_STATS is not set
> CONFIG_LRU_CACHE=m
> ```
> 
> My 5 cents with 8G pi
> 
> Have quite often OOMs during io (downloading stuff, converting video, watching over DLNA)

This can happen with or without MGLRU (MGLRU only makes it worse).

The only solution is to switch to the 64-bit kernel.

graysky2, 2023-12-09:
@popcornmix - on Arch ARM armv7h this is still happening but only in RPi4B with 8G.  The very same uSD card in RPi4B with 2G runs just fine.  The kernel is the latest in your rpi-6.1.y branch.  Again, only 8G running armv7h.

yuzhaogoogle, 2023-12-09:
> @popcornmix - on Arch ARM armv7h this is still happening but only in RPi4B with 8G. The very same uSD card in RPi4B with 2G runs just fine. The kernel is the latest in your rpi-6.1.y branch. Again, only 8G running armv7h.

I can take a close look if you have the kernel from the OOM killer. Thanks.

graysky2, 2023-12-10:
Here is the dmesg output when the OOM panic is triggered:
```
[  +0.002367] CPU: 0 PID: 34 Comm: kworker/0:1 Tainted: G         C         6.1.66-2-rpi-ARCH #1
[  +0.002349] Hardware name: BCM2711
[  +0.001976] Workqueue: events request_firmware_work_func
[  +0.001780]  unwind_backtrace from show_stack+0x18/0x1c
[  +0.001831]  show_stack from dump_stack_lvl+0x90/0xac
[  +0.001839]  dump_stack_lvl from warn_alloc+0x110/0x19c
[  +0.001805]  warn_alloc from __alloc_pages+0xe9c/0xfe8
[  +0.001686]  __alloc_pages from __kmalloc_large_node+0x70/0x14c
[  +0.002096]  __kmalloc_large_node from kmalloc_large+0x24/0xcc
[  +0.001741]  kmalloc_large from brcmf_fws_attach+0x38/0x39c [brcmfmac]
[  +0.001751]  brcmf_fws_attach [brcmfmac] from brcmf_proto_bcdc_init_done+0x18/0x30 [brcmfmac]
[  +0.001722]  brcmf_proto_bcdc_init_done [brcmfmac] from brcmf_attach+0x15c/0x4e8 [brcmfmac]
[  +0.001923]  brcmf_attach [brcmfmac] from brcmf_sdio_firmware_callback+0x84c/0x974 [brcmfmac]
[  +0.002157]  brcmf_sdio_firmware_callback [brcmfmac] from brcmf_fw_request_done+0x160/0x18c [brcmfmac]
[  +0.002742]  brcmf_fw_request_done [brcmfmac] from request_firmware_work_func+0x58/0x9c
[  +0.002201]  request_firmware_work_func from process_one_work+0x21c/0x4d0
[  +0.001950]  process_one_work from worker_thread+0x58/0x560
[  +0.001826]  worker_thread from kthread+0xd8/0xf4
[  +0.001835]  kthread from ret_from_fork+0x14/0x30
[  +0.001974] Exception stack(0xf08c5fb0 to 0xf08c5ff8)
[  +0.002038] 5fa0:                                     00000000 00000000 00000000 00000000
[  +0.001936] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  +0.001972] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  +0.001907] Mem-Info:
[  +0.002220] active_anon:59 inactive_anon:4414 isolated_anon:0
               active_file:2336 inactive_file:3449 isolated_file:0
               unevictable:0 dirty:62 writeback:0
               slab_reclaimable:2081 slab_unreclaimable:4542
               mapped:4063 shmem:125 pagetables:592
               sec_pagetables:0 bounce:0
               kernel_misc_reclaimable:0
               free:1999138 free_pcp:277 free_cma:129541
[  +0.015478] Node 0 active_anon:236kB inactive_anon:17828kB active_file:9344kB inactive_file:13796kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:16252kB dirty:248kB writeback:0kB shmem:500kB writeback_tmp:0kB kernel_stack:1276kB pagetables:2540kB sec_pagetables:0kB all_unreclaimable? yes
[  +0.004466] DMA free:588112kB boost:0kB min:65536kB low:81920kB high:98304kB reserved_highatomic:4096KB active_anon:0kB inactive_anon:0kB active_file:472kB inactive_file:148kB unevictable:0kB writepending:0kB present:786432kB managed:626972kB mlocked:0kB bounce:0kB free_pcp:1052kB local_pcp:0kB free_cma:518164kB
[  +0.004757] lowmem_reserve[]: 0 0 7284 7284
[  +0.002203] DMA: 184*4kB (UMHC) 149*8kB (UMEHC) 67*16kB (UMEHC) 30*32kB (UMEHC) 18*64kB (UMEC) 10*128kB (UMEC) 7*256kB (UME) 8*512kB (MEC) 6*1024kB (MC) 2*2048kB (ME) 138*4096kB (MC) = 587768kB
[  +0.002268] 5943 total pagecache pages
[  +0.002379] 0 pages in swap cache
[  +0.002765] Free swap  = 0kB
[  +0.002277] Total swap = 0kB
[  +0.002319] 2061312 pages RAM
[  +0.001997] 1864704 pages HighMem/MovableOnly
[  +0.001743] 39865 pages reserved
[  +0.001885] 131072 pages cma reserved
[  +0.001789] ieee80211 phy0: brcmf_bus_started: failed: -12
[  +0.001885] ieee80211 phy0: brcmf_attach: dongle is not responding: err=-12
[  +0.054322] FAT-fs (mmcblk0p6): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[  +0.015741] brcmfmac: brcmf_sdio_firmware_callback: brcmf_attach failed
[  +0.042698] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[  +0.005783] bcmgenet fd580000.ethernet end0: Link is Down
[  +0.009034] systemd invoked oom-killer: gfp_mask=0x400dc0(GFP_KERNEL_ACCOUNT|__GFP_ZERO), order=0, oom_score_adj=0
```

yuzhaogoogle, 2023-12-10:
> Here is the dmesg output when the OOM panic is triggered:
> 
> ```
> [  +0.015478] Node 0 active_anon:236kB inactive_anon:17828kB active_file:9344kB inactive_file:13796kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:16252kB dirty:248kB writeback:0kB shmem:500kB writeback_tmp:0kB kernel_stack:1276kB pagetables:2540kB sec_pagetables:0kB all_unreclaimable? yes
> [  +0.004466] DMA free:588112kB boost:0kB min:65536kB low:81920kB high:98304kB reserved_highatomic:4096KB active_anon:0kB inactive_anon:0kB active_file:472kB inactive_file:148kB unevictable:0kB writepending:0kB present:786432kB managed:626972kB mlocked:0kB bounce:0kB free_pcp:1052kB local_pcp:0kB free_cma:518164kB
> [  +0.004757] lowmem_reserve[]: 0 0 7284 7284
> [  +0.002203] DMA: 184*4kB (UMHC) 149*8kB (UMEHC) 67*16kB (UMEHC) 30*32kB (UMEHC) 18*64kB (UMEC) 10*128kB (UMEC) 7*256kB (UME) 8*512kB (MEC) 6*1024kB (MC) 2*2048kB (ME) 138*4096kB (MC) = 587768kB
> [  +0.002268] 5943 total pagecache pages
> [  +0.002379] 0 pages in swap cache
> [  +0.002765] Free swap  = 0kB
> [  +0.002277] Total swap = 0kB
> [  +0.002319] 2061312 pages RAM
> [  +0.001997] 1864704 pages HighMem/MovableOnly
> [  +0.001743] 39865 pages reserved
> [  +0.001885] 131072 pages cma reserved
> [  +0.009034] systemd invoked oom-killer: gfp_mask=0x400dc0(GFP_KERNEL_ACCOUNT|__GFP_ZERO), order=0, oom_score_adj=0
> ```

This one seems to be a legitimate OOM kill -- there was only about 600KB reclaimable memory left (active_file:472kB + inactive_file:148kB) from the only zone (DMA) where the allocation (GFP_KERNEL) can be from, and the 600KB reclaimable memory probably was all hot.

Does OOM kills happen often? If so, can you still reproduce it after `echo 0 >/sys/kernel/mm/lru_gen/enabled`? Thanks.

[20 more replies]