raspberrypi/linux #5398: Sense Hat matrix not working on 6.1 kernel.

raspberrydeveloper, 2023-03-24:
### Describe the bug

After upgrading the raspberry pi kernel to 6.1 the sense hat matrix does no longer work. Thonny and python report that they can communicate with it and seem like they are trying to use the matrix but without dropping any error. I can confirm that the gyro, accelerometer, temperature, humidity and joystick work fine.

### Steps to reproduce the behaviour

You can check it by yourself by just upgrading the pi to the newest 6.1 kernel.

### Device (s)

Raspberry Pi 4 Mod. B

### System

The `rpi-issue` file:

```
Raspberry Pi reference 2023-02-21
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 25e2319effa91eb95edd9d9209eb9f8a584d67be, stage4
```

Firmware version:

```
Mar 17 2023 10:50:39
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
```

Kernel version:

```
Linux raspberrypi4 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
```

### Logs

Here are the dmesg logs, if needed:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.19-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s78504 r8192 d32088 u118784
[    0.000000] pcpu-alloc: s78504 r8192 d32088 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[... 507 lines cut ...]
GPIO 33: level=1 fsel=7 alt=3 func=RXD0 pull=UP
GPIO 34: level=1 fsel=7 alt=3 func=SD1_CLK pull=NONE
GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD pull=UP
GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0 pull=UP
GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1 pull=UP
GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2 pull=UP
GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3 pull=UP
GPIO 40: level=1 fsel=4 alt=0 func=PWM1_0 pull=NONE
GPIO 41: level=1 fsel=4 alt=0 func=PWM1_1 pull=NONE
GPIO 42: level=0 fsel=1 func=OUTPUT pull=UP
GPIO 43: level=1 fsel=0 func=INPUT pull=UP
GPIO 44: level=1 fsel=5 alt=1 func=SDA0 pull=UP
GPIO 45: level=1 fsel=5 alt=1 func=SCL0 pull=UP
BANK2 (GPIO 46 to 53):
GPIO 46: level=0 fsel=0 func=INPUT pull=UP
GPIO 47: level=0 fsel=0 func=INPUT pull=UP
GPIO 48: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 49: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 50: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 51: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 52: level=0 fsel=0 func=INPUT pull=DOWN
GPIO 53: level=0 fsel=0 func=INPUT pull=DOWN
```

**Info:** I also have a pimoroni fanshim attached so you may see it in the i2c and gpio command outputs.

raspberrydeveloper, 2023-03-24:
Thank you very much for your help! For now I am cross compiling the 6.0 kernel, is this kernel working with the sense hat?

pelwell, 2023-03-24:
Yes - Sense HAT works with rpi-6.0.y.

raspberrydeveloper, 2023-03-24:
Cool! Suddenly the sense hat worked for a second with 6.1 but only for a second and it does not work again...

pelwell, 2023-03-24:
Yes - the sypmtom is that the very first update, the one that clears the rainbow pattern, succeeds, but then nothing else.

raspberrydeveloper, 2023-03-24:
Can I download a prebuild 6.0 kernel from somewhere because cross compiling it on my i7-7500U inside windows subsystem for linux version 2 and with 8gb ram takes a long time.

pelwell, 2023-03-24:
You may want to hold off a while - I'm close to a solution.

raspberrydeveloper, 2023-03-24:
Ok I will wait! Thanks for your time!!!

pelwell, 2023-03-24:
The problem is cause by a bug-fix patch that discards pending changes when the file handle is closed. If you keep the file handle open a bit longer you'll see the display does update - try running `cat > /dev/fb1` (or whatever the correct number is) and hitting Return a few times.

#5399 has a fix that flushes the updates synchronously on close. The auto-build will complete in about an hour, at which point you'll be able to install a test kernel with `sudo rpi-update pulls/5399`.

raspberrydeveloper, 2023-03-24:
I tried again after a reboot and it worked successfully. I can confirm that the sense hat works perfectly fine now! Again thanks a lot for your help and your time!!

closed 2023-03-24 by raspberrydeveloper

raspberrydeveloper, 2023-03-29:
So... the matrix is going to be faster with this kernel?

popcornmix, 2023-03-29:
> So... the matrix is going to be faster with this kernel?

The linked PR says:

> This alternative patch avoids the horrendous slowdown seen with the sense_hat.py library, which opens and closes the framebuffer, incurring an 18ms delay while the data is pushed over I2C, for every get_pixel and set_pixel call.

so yes - should be faster than without the PR.

pelwell, 2023-03-29:
Whether it's faster with this PR depends on what you are comparing it to. Using the Python library, 6.1 with this PR is much faster than 6.1 without it, but 6.1 with this PR is about the same as 5.15.

raspberrydeveloper, 2023-03-29:
Ok, I use python in a small program so I will use the updated kernel because why not?

steveiliop56, 2023-03-29:
I added the new kernel! The sense hat still works ok... Do you want me to run any tests?

pelwell, 2023-03-29:
Realistically it should only affect the display, so I'm happy enough with your feedback to merge it.