raspberrypi/linux #5400: Boots into a black screen after todays raspian update

Vertux, 2023-03-25:
### Describe the bug

My Raspberry Pi 4 Model B Rev 1.1 4GB was working fine, but after todays update, first it seems to boot fine but after the raspbian logo my monitor Eizo FlexScan L767 ends up with a black screen, the desktop environment does not show up as normal. It seems the update broke the monitor/resolution detection. My boot parameters  are all default, since my first installation I did not change anything.

### Steps to reproduce the behaviour

Just boot the Raspberry Pi

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi CM4

### System

`raspinfo`
https://paste.togart.de/?02c6fcd05c9c2b10#HYfnhCujmQXF7rvH1pw3Faj6ZVM6Qh8gHZc4VBqp7RSr

`/etc/rpi-issue`
Raspberry Pi reference 2023-02-21
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f2d385517c9631f2ded876deb1115725d0c75995, stage4

`vcgencmd version`
Mar 17 2023 10:50:39
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

`uname -a`
Linux raspberrypi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux

`cat config.txt`
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]

### Logs

`dmesg`
https://paste.togart.de/?017e727aa5665383#3vWJkMQkgha7E3afuYDiK73ZQfC2PD7rwaUTnQLN6oRz

INeDiA, 2023-04-05:
Same exact issue here; could reverting to 6.1.19+ solve this?
Btw, how? 😅

INeDiA, 2023-04-05:
> Could you try using rpi-update? (or wait a couple of days for the next apt kernel update). It's possible this could help: #5378
> 
> which isn't in current apt kernel but will be in the next one

I tried rpi-update but without any result, already on latest.

I read about `apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel` but honestly I don't know how this could help.

INeDiA, 2023-04-05:
> > I tried rpi-update but without any result, already on latest.
> 
> what is `uname -a` reporting?

`Linux raspberrypi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux`
Exactly as OP

popcornmix, 2023-04-05:
That's the apt, not the rpi-update kernel. You should be getting:
```Linux version 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux```

INeDiA, 2023-04-05:
Sorry about that, got lost in my many retries.
Did a `sudo rpi-update`, rebooted, now my firmware is _0b15297bf42d0dcbb463d702c739b281c272f0c1_ and `uname -a` gives
`Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`

And this is still my "screen"
[image]

6by9, 2023-04-05:
You appear to be viewing the system over VNC, so how does that relate to an issue where the HDMI output is black? Do you have an HDMI monitor connected?

INeDiA, 2023-04-05:
> You appear to be viewing the system over VNC, so how does that relate to an issue where the HDMI output is black? Do you have an HDMI monitor connected?

There was a misunderstanding, I don't know about OP but I'm using a GPIO based LCD screen, so output is identical.
I'm using VNC just because the rpi is in a different room.

INeDiA, 2023-04-05:
Should I open a different thread?

It's a [3.5" touchscreen](http://www.lcdwiki.com/3.5inch_RPi_Display), SPI, drivers are available [here](https://github.com/goodtft/LCD-show) (_3.5” RPi Display(MPI3501)_).
The script takes care of basically rewriting [config.txt](https://github.com/raspberrypi/linux/files/11161368/config.txt)

Just in case anyone is wandering: no, the screen is not broken, splash screen and some basic text is shown before the driver kicks in.

`raspinfo`
https://pastebin.com/KUae7V5k

`dmesg`
https://pastebin.com/uxUqQnKE

Thanks for help!

6by9, 2023-04-05:
> Should I open a different thread?

For an SPI display, definitely yes.

ghollingworth, 2023-04-11:
One useful test is to see if this is the same Plymouth interaction we had previously...  Can you ssh into the box and do:

`sudo remove plymouth`

See if this is just a timing thing (i.e. plymouth taking longer than previously and DRM still not being closed down quick enough for xserver to start.

Gordon

Vertux, 2023-04-15:
@ghollingworth I did another test today with the "old" 2023-02-21-raspios-bullseye-arm64 image, which I have used before as basis before I did the update.
Meanwhile there are newer updates and surprise, surprise, now everything works fine. I am not sure what was changed with the current update, but it definitly solved the issue. 
So unfortunately I could not check, if removing `plymouth` would have been solved the issue.

Just for information, here are the data from the current woking system, its all default nothing changed.

`uname -a`
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
`raspinfo`
https://paste.togart.de/?5541b8280e60c05d#CtRtuepyVqjvXAoyWdNbkVJGLH9VZJzQqP3xrTcucttf 

The current update solved the issue for me, so I will close it.

closed 2023-04-15 by Vertux