raspberrypi/linux #5401: IQaudio-codec issue

micahvdm, 2023-03-25:
### Describe the bug

Sample rate gets messed up when using the aux-jack state file. The comment in the driver about the aux Jack pin being disabled at boot to prevent sample rate issues is correct, but it doesn’t seem to be disabling the pin and therefore the sample rate is not setting correctly. The result of this is that setting the card to 48khz sample rate and playing an A2 note produces 119hz instead of 110hz. Playback of 48khz wav files is also messed up. Everything is destined and slower. 

### Steps to reproduce the behaviour

Enable iqaudio-codec
sudo alsactl —file IQaudIO_Codec_AUXIN_record_and_HP_playback.state restore
sudo alsactl store
Reboot

### Device (s)

Raspberry Pi 3 Mod. A+, Raspberry Pi 3 Mod. B, Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-02-21
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 25e2319effa91eb95edd9d9209eb9f8a584d67be, stage2

Jan  5 2023 10:48:31 
Copyright (c) 2012 Broadcom
version 8ba17717fbcedd4c3b6d4bce7e50c7af4155cba9 (clean) (release) (start)

Linux pistomp 5.15.65-rt49-v8+ #2 SMP PREEMPT_RT Sat Mar 25 05:11:57 PDT 2023 aarch64 GNU/Linux

preveen-stack, 2023-05-14:
> The comment in the driver about the aux Jack pin being disabled at boot to prevent sample rate issues is correct, but it doesn’t seem to be disabling the pin 

can you point me to the above comment

micahvdm, 2023-05-14:
/*
	 * Disable AUX Jack Pin by default to prevent PLL being enabled at
	 * startup. This avoids holding the PLL to a fixed SR config for
	 * subsequent streams.
	 *
	 * This pin can still be enabled later, as required by user-space.
	 */

iqaudio-codec.c line: 120