raspberrypi/linux #5402: RPi 4B full upgrade od 32-bit OS suddenly changes the kernel architecture to aarch64

ArkBrj, 2023-03-27:
### Describe the bug

Not really sure if this is the bugbase. Please advise if I need to file this issue somewhere else.
I noticed that on my 32 bit Raspberry Pi OS Lite after doing "sudo apt-get update" followed by "sudo apt-get full-upgrade" and reboot, "uname -m" started reporting aarch64 instead of armv7l.

### Steps to reproduce the behaviour

1. Update Raspberry Pi Imager to the latest (v1.7.4 when I discovered the issue). I use Windows version if this matters.
2. Prepare an SD card with Raspberry Pi OS Lite (32-bit).
3. Boot Raspberry Pi 4B from this SD.
4. Make a note of the architectures reported by "uname -m" (armv7l) and "dpkg --print-architecture" (armhf).
5. Run "sudo apt-get update -y" and "sudo apt-get full-upgrade -y" which is the recommended way to fully upgrade your Raspberry Pi.
6. Reboot.
7. Check the architectures again: "uname -m" now gives aarch64. "dpkg --print-architecture" still reports armhf.

### Device (s)

Raspberry Pi 4 Mod. B

### System

System Information
------------------

Raspberry Pi 4 Model B Rev 1.5
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2023-02-21
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f2d385517c9631f2ded876deb1115725d0c75995, stage2

Linux rpi 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l GNU/Linux
Revision	: c03115
Serial		: 1000000014b32210
Model		: Raspberry Pi 4 Model B Rev 1.5
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Jan  5 2023 10:46:54 
Copyright (c) 2012 Broadcom
version 8ba17717fbcedd4c3b6d4bce7e50c7af4155cba9 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       29813048 1333716  27236104   5% /
devtmpfs         1800660       0   1800660   0% /dev
tmpfs            1965524       0   1965524   0% /dev/shm
tmpfs             786212    1076    785136   1% /run
tmpfs               5120       4      5116   1% /run/lock
/dev/mmcblk0p1    261108   51024    210084  20% /boot
tmpfs             393104       0    393104   0% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
[... 968 lines cut ...]
[   16.354761] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   16.354895] Bluetooth: HCI UART protocol Broadcom registered
[   16.648149] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.648161] Bluetooth: BNEP filters: protocol multicast
[   16.648178] Bluetooth: BNEP socket layer initialized
[   16.662644] NET: Registered PF_ALG protocol family
[   16.680133] cryptd: max_cpu_qlen set to 1000
[   17.114126] vc4-drm gpu: [drm] Cannot find any crtc or sizes
[   25.886727] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   31.834191] cam-dummy-reg: disabling

EEPROM
------

*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Tue 26 Apr 10:24:28 UTC 2022 (1650968668)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

ArkBrj, 2023-03-27:
/var/log/apt/term.log in case if it is useful.

Log started: 2023-03-26  20:34:06
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
(Reading database ... 60%
(Reading database ... 65%
(Reading database ... 70%
(Reading database ... 75%
(Reading database ... 80%
(Reading database ... 85%
(Reading database ... 90%
(Reading database ... 95%
(Reading database ... 100%
(Reading database ... 43727 files and directories currently installed.)
Preparing to unpack .../00-curl_7.74.0-1.3+deb11u7_armhf.deb ...
Unpacking curl (7.74.0-1.3+deb11u7) over (7.74.0-1.3+deb11u5) ...
Preparing to unpack .../01-libcurl4_7.74.0-1.3+deb11u7_armhf.deb ...
Unpacking libcurl4:armhf (7.74.0-1.3+deb11u7) over (7.74.0-1.3+deb11u5) ...
Preparing to unpack .../02-raspberrypi-kernel_1%3a1.20230317-1_armhf.deb ...
Adding 'diversion of /boot/kernel.img to /usr/share/rpikernelhack/kernel.img by rpikernelhack'
Adding 'diversion of /boot/kernel7.img to /usr/share/rpikernelhack/kernel7.img by rpikernelhack'
Adding 'diversion of /boot/kernel7l.img to /usr/share/rpikernelhack/kernel7l.img by rpikernelhack'
Adding 'diversion of /boot/kernel8.img to /usr/share/rpikernelhack/kernel8.img by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-b-plus.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b-plus.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-b-rev1.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b-rev1.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-b.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-cm.dtb to /usr/share/rpikernelhack/bcm2708-rpi-cm.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-zero-w.dtb to /usr/share/rpikernelhack/bcm2708-rpi-zero-w.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-zero.dtb to /usr/share/rpikernelhack/bcm2708-rpi-zero.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2709-rpi-2-b.dtb to /usr/share/rpikernelhack/bcm2709-rpi-2-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2709-rpi-cm2.dtb to /usr/share/rpikernelhack/bcm2709-rpi-cm2.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-2-b.dtb to /usr/share/rpikernelhack/bcm2710-rpi-2-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-3-b-plus.dtb to /usr/share/rpikernelhack/bcm2710-rpi-3-b-plus.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-3-b.dtb to /usr/share/rpikernelhack/bcm2710-rpi-3-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-cm3.dtb to /usr/share/rpikernelhack/bcm2710-rpi-cm3.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-zero-2-w.dtb to /usr/share/rpikernelhack/bcm2710-rpi-zero-2-w.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-zero-2.dtb to /usr/share/rpikernelhack/bcm2710-rpi-zero-2.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2711-rpi-4-b.dtb to /usr/share/rpikernelhack/bcm2711-rpi-4-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2711-rpi-400.dtb to /usr/share/rpikernelhack/bcm2711-rpi-400.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2711-rpi-cm4-io.dtb to /usr/share/rpikernelhack/bcm2711-rpi-cm4-io.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2711-rpi-cm4.dtb to /usr/share/rpikernelhack/bcm2711-rpi-cm4.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2711-rpi-cm4s.dtb to /usr/share/rpikernelhack/bcm2711-rpi-cm4s.dtb by rpikernelhack'
Adding 'diversion of /boot/COPYING.linux to /usr/share/rpikernelhack/COPYING.linux by rpikernelhack'
Adding 'diversion of /boot/overlays/README to /usr/share/rpikernelhack/overlays/README by rpikernelhack'
Adding 'diversion of /boot/overlays/act-led.dtbo to /usr/share/rpikernelhack/overlays/act-led.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adafruit-st7735r.dtbo to /usr/share/rpikernelhack/overlays/adafruit-st7735r.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adafruit18.dtbo to /usr/share/rpikernelhack/overlays/adafruit18.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adau1977-adc.dtbo to /usr/share/rpikernelhack/overlays/adau1977-adc.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adau7002-simple.dtbo to /usr/share/rpikernelhack/overlays/adau7002-simple.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/ads1015.dtbo to /usr/share/rpikernelhack/overlays/ads1015.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/ads1115.dtbo to /usr/share/rpikernelhack/overlays/ads1115.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/ads7846.dtbo to /usr/share/rpikernelhack/overlays/ads7846.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adv7282m.dtbo to /usr/share/rpikernelhack/overlays/adv7282m.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adv728x-m.dtbo to /usr/share/rpikernelhack/overlays/adv728x-m.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/akkordion-iqdacplus.dtbo to /usr/share/rpikernelhack/overlays/akkordion-iqdacplus.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/allo-boss-dac-pcm512x-audio.dtbo to /usr/share/rpikernelhack/overlays/allo-boss-dac-pcm512x-audio.dtbo by rpikernelhack'
[... cut ...]
Setting up libcurl4:armhf (7.74.0-1.3+deb11u7) ...
Setting up libtiff5:armhf (4.2.0-1+deb11u4) ...
Setting up curl (7.74.0-1.3+deb11u7) ...
Setting up libcamera0:armhf (0~git20230302+923f5d70-1) ...
Setting up python3-libcamera (0~git20230302+923f5d70-1) ...
Setting up libcamera-apps-lite (0~git20230301+54a781d-1) ...
Setting up python3-picamera2 (0.3.9-1) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u5) ...
Log ended: 2023-03-26  20:35:43

pelwell, 2023-03-27:
The updated firmware now prefers the 64-bit kernel on a Pi 4 because it should give better performance in most situations. To continue to boot to the 32-bit kernel, add `arm_64bit=0` to config.txt.

ArkBrj, 2023-03-28:
Thanks, this helped.
However, one would argue that this change of kernel architecture is not a good thing from the user's perspective. This brings  the system in somewhat inconsistent state when kernel and dpkg architectures do not match.
I wasted few hours trying to figure out why I was unable to install OctoDash. Apparently OctoDash install script checks the kernel architecture and decides what package architecture to install. When it saw 'aarch64', it selected the arm64 package and dpkg refused to install it. I am not an expert and do not know if OctoDash is doing the right thing, just giving you an example of troubles the users might get.
Anyway, I am happy with the solution. Feel free to close this issue if you are not planning to reconsider the current behavior.

brubbel, 2023-03-30:
For me, it broke hamachi VPN. The binary runs, but it stopped functioning working correctly from VPN point. Took me 4 hours to find this.

hailfinger, 2023-03-31:
64-bit kernel with 32-bit userspace breaks quite a few netlink dependent kernel interfaces. All IPsec based VPNs will no longer work due to a missing compat layer in the kernel. This is not fixable in the kernel without breaking userspace API/ABI compatibility. (Yes, I tried, and enabling the generic compat layer broke other networking applications.)

This is a regression and will lock you out of remote devices if you can only access them via IPsec based VPN. Other VPN types may be affected as well.

taclane, 2023-04-11:
> I wasted few hours trying to figure out why I was unable to install OctoDash. Apparently OctoDash install script checks the kernel architecture and decides what package architecture to install. When it saw 'aarch64', it selected the arm64 package and dpkg refused to install it.

Heh.  Same here!  It wasn't immediately apparent why OctoDash would install fine with a fresh raspbian image, but as soon as the system was `apt upgrade`d, would start running into architecture errors.

https://github.com/UnchartedBull/OctoDash/issues/3559

RaulTrombin, 2023-06-13:
I was having same issue running my libs,

And if i run:
`sudo WANT_64BIT=0 rpi-update`
The system wont boot anymore after reboot.

with normal update, the @pelwell method, arm_64bit=0 > config.txt, works fine.

But it's a bit strange, 
As it have arm7 and x64 versions, why not keep their defaults.

chrisx51-2, 2024-01-03:
> It hosed my install of Paperless-NGX, too.

Same happened to me :-(

Rory-Lambert, 2024-02-02:
> It hosed my install of Paperless-NGX, too.

Has anyone found a workaround for this? Maybe altering the config.txt to arm64=0?

pelwell, 2024-02-02:
See the first response: https://github.com/raspberrypi/linux/issues/5402#issuecomment-1484676216

pelwell, 2024-02-02:
The problems in this area stem from failing to distinguish between kernel architecture and userspace architecture. It is necessary to choose the correct indication for the thing being built.

pelwell, 2024-02-23:
The change occurred in the firmware released with the 6.1.16 kernel: https://github.com/raspberrypi/rpi-firmware/commit/7d6d960523325b4c52804183f4b56e0bb9297102

peaklabs-dev, 2024-10-06:
I think I am running into this problem, how can I check the kernel and userspace each on its own if both are arch64 so 64-bit or only one is 64-bit?

popcornmix, 2024-10-06:
`getconf LONG_BIT` will return 32 for a 32-bit OS and 64 for a 64-bit OS.
`uname -m` tells you if the kernel is 32 or 64-bit.

peaklabs-dev, 2024-10-06:
Thanks, that worked. Can I somehow switch LONG_BIT to 64 if it is on 32? or should I just do a clean install of the 64-bit OS?

popcornmix, 2024-10-06:
Switching OS between 32-bit and 64-bit is a major change (almost all files in the OS change).
It needs a reinstall.

popcornmix, 2024-10-06:
Closing this, as the behaviour here is expected. A 32-bit OS on a pi4 (or pi5) will use a 64-bit kernel by default.

closed 2024-10-06 by popcornmix

peaklabs-dev, 2024-10-06:
@popcornmix So if I want my package to work without having to clean install 64-bit I. have to build a 32-bit ARM image for my docker containers, right?

pelwell, 2024-10-06:
A 64-bit kernel can host both 64-bit and 32-bit processes, including any mix of the two. Static builds just work, but any app that uses shared libraries must be able to locate the correct version.

However, it would help if you say what the problem looks like.

peaklabs-dev, 2024-10-06:
I am working on a user issue for Coolify. When the user install via the install script via this command:
```bash
curl -fsSL https://cdn.coollabs.io/coolify/install.sh | bash
```
then they get the following error: `docker: no matching manifest for linux/arm/v8 in the manifest list entries.` even though all of our [packages](https://github.com/orgs/coollabsio/packages?repo_name=coolify) have a `linux/arm64` version which should also work with `linux/arm/v8`. We use docker for everything. I think the problem stems from the raspberry pi not being fully on the 64-bit version since we do not have docker images built for 32-bit.