raspberrypi/linux #5403: CM4S:  First boot display on drm/card0, subsequent boots on drm/card1, and PCM/Master audio volume out of sync

JinShil, 2023-03-27:
### Describe the bug

I'm using 3 different Raspberry Pi Compute Modules (CM3+, CM4S, and CM4) devices.  All are nearly identical in features with only one connected HDMI display.

# CM3+ (Good)
When booting the CM3+, `ls /sys/class/drm/` reports...
```
card0  card0-HDMI-A-1  card0-Writeback-1  renderD128  version
```
... with the display on `card0` as there is only one card.   As expected `/sys/class/drm/card0-HDMI-A-1/edid` reports the EDID of my display.

When adjusting the `Master` volume slider in `alsamixer`, it also adjusts the `PCM` volume slider reported in `sudo alsamixer`, and vise-versa.  This is important for my use case; the two should be in sync -- and they are.

With subsequent boots, the system remains the same, as expected.

/boot/config.txt
```
# LCD display settings
hdmi_group=2
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0

# Enable the I2C port
dtparam=i2c_arm=on

# RTC
dtoverlay=i2c-rtc,ds3231

# Serial ports
enable_uart=1
core_freq=250
dtoverlay=uart1,txd1_pin=32,rxd1_pin=33

# Audio
dtparam=audio=on
dtoverlay=audremap,pins_40_41

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,noaudio
```

# CM4 (Good)
When booting the CM4, `ls /sys/class/drm/` reports...
```
card0  card1  card1-HDMI-A-1  card1-HDMI-A-2  card1-Writeback-1  renderD128  version
```
... with the display on `card1`.   As expected `/sys/class/drm/card1-HDMI-A-1/edid` reports the EDID of my display.  `/sys/class/drm/card1-HDMI-A-2/edid` exists, but is empty as there is only one display connected.

When adjusting the `Master` volume slider in `alsamixer`, it also adjusts the `PCM` volume slider reported in `sudo alsamixer`, and vise-versa.  The two should be in sync -- and they are.

With subsequent boots, the system remains the same, as expected.

/boot/config.txt
```
# LCD display settings
hdmi_group=2
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0

# Disable pcie
dtparam=pcie=off

# RTC
dtparam=i2c_arm=on
dtoverlay=i2c-rtc,ds3231

# Serial ports
dtoverlay=disable-bt
dtparam=uart0=on
dtoverlay=uart2,txd2_pin=0,rxd2_pin=1

# Audio
dtparam=audio=on
dtoverlay=audremap,pins_12_13

# USB peripherals
dtoverlay=dwc2,dr_mode=host

# Enable the VC4 display driver
dtoverlay=vc4-kms-v3d,noaudio
```

# CM4S (Bad)
When booting the CM4S *the first time*, `ls /sys/class/drm/` reports...
```
card0  card0-HDMI-A-1  card0-Writeback-1  card1  renderD128  version
```
... with the display on `card0`.   As expected `/sys/class/drm/card0-HDMI-A-1/edid` reports the EDID of my display.  

When adjusting the `Master` volume slider in `alsamixer`, it also adjusts the `PCM` volume slider reported in `sudo alsamixer`, and vise-versa.  The two should be in sync -- and they are.

HOWEVER, subsequent boots are different.  The display switches to `card1`.   `ls /sys/class/drm/` reports...
```
card0  card1  card1-HDMI-A-1  card1-Writeback-1  renderD128  version
```
... with the display on `card1`.

The UI displays fine, and everything else on the system works fine except the following:  When adjusting the `Master` volume slider in `alsamixer`, the `PCM` volume slider reported in `sudo alsamixer` **does not** change, and that causes a problem for my use case.

/boot/config.txt
```
# LCD display settings
hdmi_group=2
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0

# Enable the I2C port
dtparam=i2c_arm=on

# RTC
dtoverlay=i2c-rtc,ds3231

# Serial ports
enable_uart=1
core_freq=250
dtoverlay=uart1,txd1_pin=32,rxd1_pin=33

# Audio
dtparam=audio=on
dtoverlay=audremap,pins_40_41

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,noaudio,cma-256
```

### Steps to reproduce the behaviour

1.  Create a new OS using Raspberry Pi imager
2. Boot it on the CM4S
3. Make note that `ls /sys/class/drm` reports the display on `card0`.
4. Note that when adjusting the `Master` volume slider in `alsamixer`, the PCM volume slider in `sudo alsamixer` also adjusts in synchronization.
5. Reboot the system
6. Make note that `ls /sys/class/drm` shows the display now on `card1`
7. Note that when adjusting the `Master` volume slider in `alsamixer`, the PCM volume slider in `sudo alsamixer` **does not** adjust in synchronization.

### Device (s)

Raspberry Pi CM3+, Raspberry Pi CM3+ Lite, Raspberry Pi CM4, Raspberry Pi CM4 Lite, Other

### System

Mar 21 2023 17:18:16
Copyright (c) 2012 Broadcom
version 3cc1c2dfc5460da9e1a0a4f48b48ab508c48bfe5 (clean) (release) (start)

Linux raspberrypi 6.1.21-v8+ #1639

Please note I'm using a pre-release kernel due to  https://github.com/raspberrypi/linux/issues/5390, but the problem also exists in the 5.15 kernels.

JinShil, 2023-03-27:
I'm having trouble now reproducing the issue on the CM4S's first boot where the display appears on `card0`.  Now the display is always appearing on `card1` despite whether it's the first boot or not.

What determines whether the display appears on `card0` or `card1`?

6by9, 2023-03-27:
> What determines whether the display appears on card0 or card1?

Probe order during boot.
v3d has very few dependencies so can generally probe first.
vc4 has quite a few (reset controllers, I2C controllers, etc), so typically ends up with one or more defers before it can fully probe.

JinShil, 2023-03-27:
Assuming the CM4S is mounted in a carrier board originally designed for the CM3+, where should the display end up?  `card0` like that of the CM3+, or `card1` due to its lineage with the CM4?

Is there any difference between `card0` and `card1` on the CM4S?

Why is the `Master` volume in sync with the `PCM` volume when the display is on `card0`, but out of sync when the display is on `card1`?

JinShil, 2023-03-28:
I figured out how to reproduce the issue more reliably.  Removing the i2c and RTC lines from config.txt will put the display on `card0`.  When they are added back, the display appears on `card1`:

/boot/config.txt
```
# display on card`
#dtparam=i2c_arm=on
#dtoverlay=i2c-rtc,ds3231

# card0
dtparam=i2c_arm=on
dtoverlay=i2c-rtc,ds3231
```

Still hoping for answers to the following questions:

1. Is there any difference between card0 and card1 on the CM4S?
2. Why is the Master volume in sync with the PCM volume when the display is on card0, but out of sync when the display is on card1?

JinShil, 2023-03-28:
Now I can no longer reproduce the audio issue even when the display is on `card1`.  This issue has me questioning my sanity.  I'm closing it before it possesses someone else.  If I stop hallucinating, I'll create new issue hopefully with some connection to reality.

closed 2023-03-28 by JinShil

JinShil, 2023-03-28:
I'm able to resolve the issue by adding this line to `/etc/pulse/default.pa`
```
load-module module-alsa-card device_id="0" name="platform-bcm2835_audio" card_name="alsa_card.platform-bcm2835_audio" namereg_fail=false tsched=no fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1"
```
But, I don't know why that is required on the CM4S, but not on other compute modules.