raspberrypi/linux #5410: v4l2loopback-dkms fails on 6.1.19-v8+

aurimas-kudarauskas-epam, 2023-03-28:
### Describe the bug

After running apt full-upgrade installation of v4l2loopback kernel module fails.
Dkms package reports missing header files.

### Steps to reproduce the behaviour

Start with v4l2loopback module installed on Rasbian os with kernel 5.x.x
Make sure that headers for kernel version pressent:
sudo apt-get install v4l2loopback-dkms

run apt update + apt full-upgrade to update into 6.1.xx Kernel version

update fails with failure to rebuild v4l2loopback kernel module

### Device (s)

Raspberry Pi 4 Mod. B

### System

cat /etc/rpi-issue ->
Raspberry Pi reference 2023-02-21
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f2d385517c9631f2ded876deb1115725d0c75995, stage4

vcgencmd version ->
Mar 17 2023 10:50:39
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

uname -a -> 
Linux tvbox 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux

### Logs

sudo apt full-upgrade ->

Loading new v4l2loopback-0.12.5 DKMS files...
It is likely that 6.1.19-v8+ belongs to a chroot's host
Building for 6.1.19+, 6.1.19-v7+, 6.1.19-v7l+ and 6.1.19-v8+
Building initial module for 6.1.19+
Error! Bad return status for module build on kernel: 6.1.19+ (aarch64)
Consult /var/lib/dkms/v4l2loopback/0.12.5/build/make.log for more information.
dpkg: error processing package v4l2loopback-dkms (--configure):
 installed v4l2loopback-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 v4l2loopback-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

--------------------------------------------------------------------------------
cat /var/lib/dkms/v4l2loopback/0.12.5/build/make.log ->

DKMS make.log for v4l2loopback-0.12.5 for kernel 6.1.19+ (aarch64)
Tue 28 Mar 14:26:33 BST 2023
Building v4l2-loopback driver...
make -C /lib/modules/6.1.19+/build M=/var/lib/dkms/v4l2loopback/0.12.5/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.19+'
  CC [M]  /var/lib/dkms/v4l2loopback/0.12.5/build/v4l2loopback.o
In file included from ./include/linux/preempt.h:10,
                 from ./include/linux/spinlock.h:56,
                 from ./include/linux/vmalloc.h:5,
                 from /var/lib/dkms/v4l2loopback/0.12.5/build/v4l2loopback.c:17:
./include/linux/linkage.h:8:10: fatal error: asm/linkage.h: No such file or directory
    8 | #include <asm/linkage.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/v4l2loopback/0.12.5/build/v4l2loopback.o] Error 1
make[1]: *** [Makefile:2012: /var/lib/dkms/v4l2loopback/0.12.5/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.19+'
make: *** [Makefile:43: v4l2loopback.ko] Error 2

### Additional context

Tried:
apt install --reinstall  raspberrypi-kernel-headers
Manually reinstall:
http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20230317-1_arm64.deb ( Got error that system is armhf)

http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20230317-1_armhf.deb (installed without issue)

popcornmix, 2023-03-28:
Does adding `arm_64bit=0` to config.txt (and rebooting) allow this to succeed?

This is a workaround rather than a fix, but would confirm if the issue is purely related to the 64-bit kernel,
rather than other 6.1 kernel changes.

terminet85, 2023-03-28:
> ### Describe the bug
>
> After running apt full-upgrade installation of v4l2loopback kernel module fails. Dkms package reports missing header files.
> ### Steps to reproduce the behaviour
>
> Start with [...]

--force-all

terminet85, 2023-03-28:
> ./include/linux/linkage.h:8:10: fatal error: asm/linkage.h: No such file or directory

./linux-headers-6.1.19-v8+/include/asm-generic/linkage.h
./linux-headers-6.1.19-v7l+/include/asm-generic/linkage.h
./linux-headers-6.1.19+/include/asm-generic/linkage.h

aurimas-kudarauskas-epam, 2023-03-28:
> arm_64bit=0

Setting this, rebooting and reruning apt full-upgrade build v4l2loopback kernel module. Is it safe to use thus hotfix until final fix issued?

popcornmix, 2023-03-28:
>Setting this, rebooting and reruning apt full-upgrade build v4l2loopback kernel module. Is it safe to use thus hotfix until final fix issued?

arm_64bit used to default to 0, and has been changed to default to 1 recently.
So you are just switching back to the old behaviour.

pelwell, 2023-03-29:
They must have been running the 32-bit OS, which will come with the 64-bit kernel as well.

terminet85, 2023-03-30:
> It seems that there is no build dir in /lib/modules/6.1.19-v8+.

https://github.com/raspberrypi/linux/issues/5408