raspberrypi/linux #5418: staging: vc04_services: codec: V4L2_PIX_FMT_RGBA32 -> JPEG encoder not functional

mikegapinski, 2023-04-02:
### Describe the bug

When the input format is set to RGBA32 the driver throws:

[ 362.901009] bcm2835-codec bcm2835-codec: buffer size mismatch sizeimage 0 < min size 786432

for a 1024x768 frame.

RGB32 is not supported so I've converted the frame to RGB24 with NEON SIMD and it works fine with V4L2_PIX_FMT_RGB34

### Steps to reproduce the behaviour

I use the m2m encoder implementation from pikvm, it's widely used and supported. 

I created a sample program to reproduce the issue:

https://github.com/mikegapinski/rpi-m2m-encoder/blob/main/main.cpp

I expect the RGBA32 or RGB32 format to be accepted by the encoder, the conversion to RGB24 should not be required

### Device (s)

Raspberry Pi 4 Mod. B

### System

 5.15.94-v8-gbdf7cbabcc95
 
 Android 13 @ github.com/tesla-android and current version of Raspbian

### Logs

[ 362.901009] bcm2835-codec bcm2835-codec: buffer size mismatch sizeimage 0 < min size 786432

6by9, 2023-04-12:
V4L2_PIX_FMT_RGB32 is a deprecated format (see https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/pixfmt-rgb.html#deprecated-rgb-formats).

It's also not supported by the codec. Supported formats are:
```
pi@raspberrypi:~/rpi-m2m-encoder $ v4l2-ctl --list-formats-out -d 31
ioctl: VIDIOC_ENUM_FMT
	Type: Video Output Multiplanar

	[0]: 'RGBP' (16-bit RGB 5-6-5)
	[1]: 'RGB3' (24-bit RGB 8-8-8)
	[2]: 'BGR3' (24-bit BGR 8-8-8)
	[3]: 'AB24' (32-bit RGBA 8-8-8-8)
	[4]: 'BGR4' (32-bit BGRA/X 8-8-8-8)
	[5]: 'YU12' (Planar YUV 4:2:0)
	[6]: 'YUYV' (YUYV 4:2:2)
	[7]: 'YVYU' (YVYU 4:2:2)
	[8]: 'VYUY' (VYUY 4:2:2)
	[9]: 'UYVY' (UYVY 4:2:2)
```

Switch to V4L2_PIX_FMT_BGR32 (BGR4) and it works. (It is also deprecated and should be swapped for V4L2_PIX_FMT_XBGR32)

The JPEG codec is inherently YUV based, so the list of supported formats is based on the image conversion functions that had been implemented.
It also has no alpha support, so it should be using _XRGB32 and _XRGB32 formats. I'm reluctant to change them now as applications may be relying on them. I wonder if I can map those nicely in try_fmt to avoid regressions.

I was expecting V4L2_PIX_FMT_RGBA32 to work too, but you're right that it doesn't.

https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c#L266 is missing the 5th element in the array, so for image_encode we align to a multiple of 0, giving a stride and overall image size of 0.

mikegapinski, 2023-04-13:
It works! Thank you very much for the fix

closed 2023-04-13 by mikegapinski

mikegapinski, 2023-07-19:
@6by9 I'm back with another RGBA32 thing. I switched from the JPEG encoder to H264, leaving the source intact. The encoder starts up normally but there is something wrong with color mapping:||
[image]

Is this a similar small mapping issue between v4l2 and mmal? I am using the pikvm encoder ported over to AOSP, I think that the config is OK... https://github.com/tesla-android/android-external-tesla-android-virtual-display/blob/main/encode/m2m.c

The same thing happens with MJPEG encoder. 

This means that https://github.com/raspberrypi/linux/pull/5426 fixed the jpeg encoder only

6by9, 2023-07-19:
> This means that #5426 fixed the jpeg encoder only

Er, yes, it was specifically for JPEG  encode and V4L2_PIX_FMT_RGBA32 that was missing an alignment parameter.

> I'd suggest adding a log entry in dmesg for people using those deprecated formats like RGBA32 when the HW encoder starts. This would make it possible to remove support at a later stage. I'd expect only formats available in v4l2-ctl --list-formats-out to work. Using anything else should throw an error/deprecation warning. Not anyone knows that they can use any supported format as long as it has a matching length.

The V4L2 spec requires a VIDIOC_S_FMT or VIDIOC_TRY_FMT with invalid parameters to be amended and returned.
https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/vidioc-g-fmt.html#description
> When the application calls the [VIDIOC_S_FMT](https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/vidioc-g-fmt.html#vidioc-g-fmt) ioctl with a pointer to a struct [v4l2_format](https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/vidioc-g-fmt.html#c.V4L.v4l2_format) structure the driver checks and adjusts the parameters against hardware abilities. Drivers should not return an error code unless the type field is invalid, this is a mechanism to fathom device capabilities and to approach parameters acceptable for both the application and driver.

We do that, dropping back to the first supported format should the image format be invalid.
Adding a logging message on any fixup would result in kernel log spam (GStreamer tends to do lots of probing on startup). Allowing userspace to spam the kernel log is considered a bad thing in the kernel community.

Colour mapping issues for pixel interleaved formats are always a pain, partly in determining the correct layout in memory of a reference frame. I may have a suitable test app to mod to exercise it, but it's low priority.