raspberrypi/linux #5432: Screen Blanking not working with official7" Raspberry Pi screen

asisrodriguezntdhoy, 2023-04-14:
### Describe the bug

Hello world, since the update of Raspbian, in February the screen not sleep. I verify in Raspberry Pi config and is active... but the screen is always working. 
I put the issue in https://forums.raspberrypi.com/viewtopic.php?t=348168 but nothing.
I have 5 Raspberry Pi 3B since years working, four with the official Raspberry Pi screen with this problem and one with a small 3.5" touch screen, with other drive and working perfectly in the screen on/off issue). The other 4, not. After the first screen black... the LEDs moving all the time (so are working) but the touch screen dies in black).
I have done the reset to all the Rasp, updates all the times and now I can see they are working as always.... until i left to touch the screen and then... black all day :-( 
Please, HELP and tanks very much. 

### Steps to reproduce the behaviour

I have 5 Raspberry Pi 3B since years working, four with the official Raspberry Pi screen with this problem and one with a small 3.5" touch screen, with other drive and working perfectly in the screen on/off issue). The other 4, not. After the first screen black... the LEDs moving all the time (so are working) but the touch screen dies in black).

### Device (s)

Raspberry Pi 3 Mod. B

### System

Last Raspbian

### Logs

I don't know this

### Additional context

In the update of 12 April, work for around 15 minutes but, after this time, the finger touch on the screen is not able to switch on the screen.

6by9, 2023-04-19:
Please provide full information about your system.

The system box as you raise the initial issue lists:
> Which OS and version (```cat /etc/rpi-issue```)?
> Which firmware version (```vcgencmd version```)?
> Which kernel version (```uname -a```)?

The output of those 3 commands is the minimum we need to know vaguely what is going on.

Are there any messages in the kernel log (```dmesg```) after the screen blanks? SSH will be active even if your LCD screen is blank.
How are you powering the display and Pi? Micro-USB into the Pi and jumper wires to the display, or something else?

Raspbian was renamed Raspberry Pi OS in May 2020, so either you're running a very old system, or not actually looking at what you're running.

asisrodriguezntdhoy, 2023-04-20:
thankyou. I don't undestand too much aboiut this:
Which OS and version: (attached picture)
Which firmware version: 82f3750a65fadae9a38077e3c2217ad158c8d54
Which kernel version: 6.1.21-V7+ #1642 SMP
[image]

The other things you ask me... I have no idea to do it.
Well, I download just right now the last last last version and installed in one of my Rasperry PI and back to you. But I always update and this has begun to break in the February update... after years or works 24/7. 
Thanks very much. I will back here after the "new" instalation. Thanks

asisrodriguezntdhoy, 2023-04-20:
Hello, I did in two of my Raspberry Pi3:
Donwload a new version, open and use new SD, install... and the same... works for 15 minutes and then, when black screen...atay black..
Please, help to fix this terrible bug. Thanks. Asis

aBUGSworstnightmare-rpi, 2023-04-21:
> Hello, I did in two of my Raspberry Pi3:
> Download a new version, open and use new SD, install... and the same... works for 15 minutes and then, when black screen...stay black.
> Please, help to fix this terrible bug. Thanks. Asis.
Type 'dmesg > output.txt' on your Pi. Post the file (or the entire (!) content here).

asisrodriguezntdhoy, 2023-04-21:
[output.txt](https://github.com/raspberrypi/linux/files/11293911/output.txt)
Here it is. Thanks a lot.

aBUGSworstnightmare-rpi, 2023-04-21:
Is that before the screen blanked or after it you've tried to resume from blanking?

`[   10.871808] vc4_dsi 3f700000.dsi: Failed to create device link (0x180) with 0.panel_disp1
[   10.873376] vc4_dsi 3f700000.dsi: Failed to create device link (0x180) with 0.reg_bridge
..
[   11.689739] edt_ft5x06 10-0038: Unable to fetch data, error: -121`

looks strange - but as said I don't have an official display to test with.

asisrodriguezntdhoy, 2023-04-21:
Before, when go black... I can not do nothing until reset :-(
Thanks

aBUGSworstnightmare-rpi, 2023-04-21:
> Before, when go black... I can not do nothing until reset :-(
> Thanks
please post the result of
`cat /proc/cpuinfo`

asisrodriguezntdhoy, 2023-04-21:
Sorry, I don't nkow "make sure to have SSH enabled and run the command from SSH!"
here it is the pics. Many thanks.
[image]
[image]

aBUGSworstnightmare-rpi, 2023-04-21:
sorry, but (at least) I don't care about any crappy picture!
Did you learn something already?

What if you run below command? Any guess what it will result in
`cat /proc/cpuinfo > pidetails.txt`

I'm not a linux pro myself, but you should learn how to use SSH (or understand what '>' results in when used in the commands)

6by9, 2023-04-21:
````[ 10.871808] vc4_dsi 3f700000.dsi: Failed to create device link (0x180) with 0.panel_disp1``` is normal with the 6.1 kernel - it's harmless noise from the core kernel code.

The occasional  ```edt_ft5x06 10-0038: Unable to fetch data, error: -121``` seems inherent as the touch controller doesn't always respond.

I'll try to dig out a PI3 and check that things are behaving as expected. It sounds like it's the screensaver blanking.
Are you booting to just the console (as in your latest pictures), or using X?

asisrodriguezntdhoy, 2023-04-21:
Thanks, I am a dummie on this... :-(
What is "using X"??? I just reset all the times.
Please, remember this configuration of my RaspBerry Pi 3 + official 7" screen has been working for years... until the update of February.
Thanks so much.

aBUGSworstnightmare-rpi, 2023-04-21:
X means booting in a graphical user interface (Desktop)
CLI means booting into command line interface

6by9, 2023-04-21:
Xserver - https://en.wikipedia.org/wiki/X_Window_System
Console or GUI?

Your initial post you were pointing at the update of 12 April, now you're saying February.
April's apt update bumped to the 6.1 kernel version, whilst Feb was still on 5.15.

asisrodriguezntdhoy, 2023-04-21:
ooopppss sorry for the confusion of dates, The first one date was 28th February, as you can read here: https://forums.raspberrypi.com/viewtopic.php?t=348168

6by9, 2023-04-25:
Fresh image of Bullseye 32bit 2023-02-21 on a Pi3B.
I've used Imager to preset the hostname, password, and enable SSH.
Initial boot has been done with HDMI attached rather than the DSI display.
```sudo apt update``` ```sudo apt dist-upgrade```. Kernel now 6.1.21-v7+
Shutdown, detach HDMI, and attach DSI screen.
Waited for the screensaver to kick in and blank the screen. Tap the screen and it wakes up fine. Repeated several times.

There was an issue in mesa or xserver that resulted in the cursor plane not being cleared off the screen when blanking. That fix is in apt, so would also be picked up by ```sudo apt update``` ```sudo apt dist-upgrade```.

asisrodriguezntdhoy, 2023-04-25:
Thanks very much... but nothing, the same.
I did all this, reset and as past, after 10-15 minutes go black and finish :-(

6by9, 2023-04-26:
What (if anything) extra is logged in the kernel log at that point? (Use SSH to connect in from another system, and run ```dmesg```).

Similarly from an SSH session you should be able to use ```evtest``` to log any input events (```sudo apt install evtest``` if it's not installed). At the moment you haven't determined whether it is the touch interface or the display side that you're having an issue with.

asisrodriguezntdhoy, 2023-04-26:
Thanks very much. I don't understand so much to do this, what you say.
Anyway, the RPi with a small 3,5" touch screen works perfectly. The four RPi with official Rpi display has the problem... since February update. So, I think is a problem of software, the driver for the display.
Or... there is a "hide" screensaver on the RPi that crash the "touch" of the screes???
Thanks very much for the help.

6by9, 2023-04-26:
I've been through with a Pi3 and official 7" display and can not find an issue. Any debugging therefore has to come through you.

SSH - https://www.raspberrypi.com/documentation/computers/remote-access.html#ssh
A way of remotely connecting in to your system.

I'm making an assumption that it is the screen saver as you've given me no other information to work with.
Disable the screen saver - https://raspberry-projects.com/pi/pi-operating-systems/raspbian/screensaver

asisrodriguezntdhoy, 2023-05-06:
Well, first, sorry for so long time to come back here, a hard-work-week.
Second, thanks so much, with the xscreensaver works...
It's not "the solution" but is a good solution.
I hope the driver of the 7" official RPi display will be fixed, but now, I can work with my RPi again.
Thanks very much.