raspberrypi/linux #5438: imx219 sensor 200fps not possible anymore (max. 103fps currently)

Hermann-SW, 2023-04-20:
### Describe the bug

In January "--list-cameras" reports >200fps:
```
pi@pi:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx219 [3280x2464] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
```

After "sudo apt update" and "sudo apt update" only 103fps.
There are only 6 commits to imx219.c since January:
https://github.com/raspberrypi/linux/commits/rpi-6.1.y/drivers/media/i2c/imx219.c
```
pi@raspberrypi4B2:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx219 [3280x2464] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [103.33 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [103.33 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
pi@raspberrypi4B2:~ $ 
```

### Steps to reproduce the behaviour

Watch output of "libcamera-hello --list-cameras"

### Device (s)

Raspberry Pi 4 Mod. B

### System

lates 64bit Raspberry PiOS

6by9, 2023-04-20:
What version of libcamera and libcamera-apps are you running?

Hermann-SW, 2023-04-20:
No special version, just the version resultting from normal system updates and update/upgrade:
```
pi@raspberrypi4B2:~ $ libcamera-hello --version
libcamera-apps build: 54a781dffdd1 01-03-2023 (17:53:39)
libcamera build: v0.0.4+22-923f5d70
pi@raspberrypi4B2:~ $ 
```

I found workaround, install fresh 2023-2-21 64bit PiOS and accept no updates, then 200fps works.

6by9, 2023-04-20:
What does ```v4l2-ctl -d /dev/v4l-subdev0 --list-ctrls-menu``` report in each case?

Hermann-SW, 2023-04-20:
```
pi@raspberrypi4B2:~ $ libcamera-vid --level 4.2 --framerate 300 --buffer-count 10 --width 640 --height 480 --save-pts t.pts -o t.h264 -t 3000 --denoise cdn_off -n > /dev/null 2>&1
pi@raspberrypi4B2:~ $ ptsanalyze d.pts
creating tstamps.csv
200 frames were captured
majority framerate 103fps
  frame delta time[us] distribution
      1 9674
      1 9675
      4 9677
     49 9678
    131 9679
      8 9680
      1 9681
      1 9683
      1 9684
> after skip frame indices (middle column)
0 frame skips (0%)
average framerate 103fps
pi@raspberrypi4B2:~ $ v4l2-ctl -d /dev/v4l-subdev0 --list-ctrls-menu 

User Controls

                       exposure 0x00980911 (int)    : min=4 max=508 step=1 default=508 value=508
                horizontal_flip 0x00980914 (bool)   : default=0 value=1 flags=modify-layout
                  vertical_flip 0x00980915 (bool)   : default=0 value=1 flags=modify-layout

Camera Controls

             camera_orientation 0x009a0922 (menu)   : min=0 max=2 default=2 value=2 flags=read-only
				0: Front
				1: Back
				2: External
         camera_sensor_rotation 0x009a0923 (int)    : min=180 max=180 step=1 default=180 value=180 flags=read-only

Image Source Controls

              vertical_blanking 0x009e0901 (int)    : min=32 max=65055 step=1 default=1283 value=32
            horizontal_blanking 0x009e0902 (int)    : min=2808 max=32112 step=1 default=2808 value=2808
                  analogue_gain 0x009e0903 (int)    : min=0 max=232 step=1 default=0 value=224
                red_pixel_value 0x009e0904 (int)    : min=0 max=1023 step=1 default=1023 value=1023
          green_red_pixel_value 0x009e0905 (int)    : min=0 max=1023 step=1 default=1023 value=1023
               blue_pixel_value 0x009e0906 (int)    : min=0 max=1023 step=1 default=1023 value=1023
         green_blue_pixel_value 0x009e0907 (int)    : min=0 max=1023 step=1 default=1023 value=1023

Image Processing Controls

                 link_frequency 0x009f0901 (intmenu): min=0 max=0 default=0 value=0 flags=read-only
				0: 456000000 (0x1b2e0200)
                     pixel_rate 0x009f0902 (int64)  : min=182400000 max=182400000 step=1 default=182400000 value=182400000 flags=read-only
                   test_pattern 0x009f0903 (menu)   : min=0 max=4 default=0 value=0
				0: Disabled
				1: Color Bars
				2: Solid Color
				3: Grey Color Bars
				4: PN9
                   digital_gain 0x009f0905 (int)    : min=256 max=4095 step=1 default=256 value=256
pi@raspberrypi4B2:~ $ 
```

Hermann-SW, 2023-04-20:
Same output on fresh 2023-2-21 64bit PiOS without updattes:
```
pi@raspberrypi4Bfeb:~ $ libcamera-vid --level 4.2 --framerate 300 --buffer-count 10 --width 640 --height 480 --save-pts t.pts -o t.h264 -t 3000 --denoise cdn_off -n > /dev/null 2>&1
pi@raspberrypi4Bfeb:~ $ ptsanalyze t.pts
creating tstamps.csv
607 frames were captured
majority framerate 205fps
  frame delta time[us] distribution
      1 4863
      1 4868
      1 4872
      1 4873
      4 4875
     53 4876
    397 4877
    136 4878
      6 4879
      2 4883
      1 4886
      1 4889
> after skip frame indices (middle column)
0 frame skips (0%)
average framerate 205fps
pi@raspberrypi4Bfeb:~ $ v4l2-ctl -d /dev/v4l-subdev0 --list-ctrls-menu 

User Controls

                       exposure 0x00980911 (int)    : min=4 max=508 step=1 default=508 value=508
                horizontal_flip 0x00980914 (bool)   : default=0 value=1 flags=modify-layout
                  vertical_flip 0x00980915 (bool)   : default=0 value=1 flags=modify-layout

Camera Controls

             camera_orientation 0x009a0922 (menu)   : min=0 max=2 default=2 value=2 flags=read-only
				0: Front
				1: Back
				2: External
         camera_sensor_rotation 0x009a0923 (int)    : min=180 max=180 step=1 default=180 value=180 flags=read-only

Image Source Controls

              vertical_blanking 0x009e0901 (int)    : min=32 max=65055 step=1 default=1283 value=32
            horizontal_blanking 0x009e0902 (int)    : min=2808 max=32112 step=1 default=2808 value=2808
                  analogue_gain 0x009e0903 (int)    : min=0 max=232 step=1 default=0 value=224
                red_pixel_value 0x009e0904 (int)    : min=0 max=1023 step=1 default=1023 value=1023
          green_red_pixel_value 0x009e0905 (int)    : min=0 max=1023 step=1 default=1023 value=1023
               blue_pixel_value 0x009e0906 (int)    : min=0 max=1023 step=1 default=1023 value=1023
         green_blue_pixel_value 0x009e0907 (int)    : min=0 max=1023 step=1 default=1023 value=1023

Image Processing Controls

                 link_frequency 0x009f0901 (intmenu): min=0 max=0 default=0 value=0 flags=read-only
				0: 456000000 (0x1b2e0200)
                     pixel_rate 0x009f0902 (int64)  : min=364800000 max=364800000 step=1 default=364800000 value=364800000 flags=read-only
                   test_pattern 0x009f0903 (menu)   : min=0 max=4 default=0 value=0
				0: Disabled
				1: Color Bars
				2: Solid Color
				3: Grey Color Bars
				4: PN9
                   digital_gain 0x009f0905 (int)    : min=256 max=4095 step=1 default=256 value=256
pi@raspberrypi4Bfeb:~ $ 
```

6by9, 2023-04-20:
```
                 pixel_rate 0x009f0902 (int64)  : min=182400000 max=182400000 step=1 default=182400000 value=182400000 flags=read-only
```
vs
```
                pixel_rate 0x009f0902 (int64)  : min=364800000 max=364800000 step=1 default=364800000 value=364800000 flags=read-only
```
     
I think it's just patch https://github.com/raspberrypi/linux/commit/1ae775b5d82035f25291184796b74ecc6895728a is missing from 6.1

6by9, 2023-04-20:
See pull https://github.com/raspberrypi/linux/pull/5439.

There were merge conflicts that should be resolved correctly, but I haven't had a chance to test it. If you could, that would be appreciated. ```sudo rpi-update pulls/5439``` should retrieve the build once complete.

closed 2023-04-20 by Hermann-SW