raspberrypi/linux #5445: rpi4b can't recognize rockchip rk3358-evb device's USB otg port

yf13, 2023-04-27:
### Describe the bug

When connecting the USB OTG port of a rockchip rk3358-evb with rpi4b device, I got the following issues from dmesg:

`
[21508.683017] ------------[ cut here ]------------
[21508.683031] WARNING: CPU: 3 PID: 2988 at drivers/firmware/raspberrypi.c:67 rpi_firmware_property_list+0x208/0x264
[21508.683061] Firmware transaction timeout
[21508.683069] Modules linked in: ftdi_sio cp210x usbserial ipv6 af_packet brcmfmac brcmutil cfg80211 rfkill raspberrypi_hwmon bcm2835_isp(C) bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) vc_sm_cma(C) videobuf2_vmalloc nvmem_rmem snd_soc_hdmi_codec rpivid_hevc(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc snd_bcm2835(C) uio_pdrv_genirq uio v3d gpu_sched i2c_brcmstb vc4 snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd cec drm_kms_helper drm drm_panel_orientation_quirks syscopyarea sysfillrect sysimgblt fb_sys_fops backlight squashfs
[21508.683463] CPU: 3 PID: 2988 Comm: kworker/3:1 Tainted: G         C        5.15.86-0-rpi4 #1-Alpine
[21508.683478] Hardware name: BCM2711
[21508.683487] Workqueue: events get_values_poll [raspberrypi_hwmon]
[21508.683521] [<c020ff60>] (unwind_backtrace) from [<c020b314>] (show_stack+0x10/0x14)
[21508.683552] [<c020b314>] (show_stack) from [<c0b60d3c>] (dump_stack_lvl+0x90/0xac)
[21508.683579] [<c0b60d3c>] (dump_stack_lvl) from [<c0220528>] (__warn+0xec/0x148)
[21508.683605] [<c0220528>] (__warn) from [<c0b5d204>] (warn_slowpath_fmt+0x9c/0xc4)
[21508.683627] [<c0b5d204>] (warn_slowpath_fmt) from [<c0995d18>] (rpi_firmware_property_list+0x208/0x264)
[21508.683649] [<c0995d18>] (rpi_firmware_property_list) from [<c0995dd4>] (rpi_firmware_property+0x60/0x108)
[21508.683666] [<c0995dd4>] (rpi_firmware_property) from [<bf227178>] (get_values_poll+0x40/0x110 [raspberrypi_hwmon])
[21508.683707] [<bf227178>] (get_values_poll [raspberrypi_hwmon]) from [<c023d52c>] (process_one_work+0x218/0x52c)
[21508.683735] [<c023d52c>] (process_one_work) from [<c023eaa8>] (worker_thread+0x4c/0x5a4)
[21508.683757] [<c023eaa8>] (worker_thread) from [<c024549c>] (kthread+0x148/0x164)
[21508.683780] [<c024549c>] (kthread) from [<c02000d4>] (ret_from_fork+0x14/0x20)
[21508.683798] Exception stack(0xc39edfb0 to 0xc39edff8)
[21508.683810] dfa0:                                     00000000 00000000 00000000 00000000
[21508.683823] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[21508.683834] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[21508.683843] ---[ end trace 666034909685b435 ]---
`
My kernel version:  

`$ uname -a
Linux pi4 5.15.86-0-rpi4 #1-Alpine SMP Mon Jan 2 15:56:22 UTC 2023 armv7l Linux
`
Regards,

### Steps to reproduce the behaviour

See description

### Device (s)

Raspberry Pi 4 Mod. B

### System

- alpine 3.17.1 armv7
- Linux pi4 5.15.86-0-rpi4 #1-Alpine SMP Mon Jan 2 15:56:22 UTC 2023 armv7l Linux

pelwell, 2023-04-27:
The kernel stack trace tells us nothing useful, only that the firmware/VPU has crashed.

I can't find a datasheet for the rk3358-evb, only the rk3588-evb. Whatever it is, how are you powering it?

yf13, 2023-04-27:
My device rpi4b is powered with type-C port.

Maybe it is a temporary error? After rebooting the rpi4b, I can use that usb otg device now... maybe we can give this lower priority for now and I will close it after watching it for a while

yf13, 2023-04-29:
My EVB is powered by its own adaptor... 
Though it happens occasionally on my rpi4b, I am okay to close this ticket as rebooting rpi4b can resolve it most time.

closed 2023-04-29 by yf13