raspberrypi/linux #5446: XFRM Interface support not enabled by default

SaschaJohn, 2023-04-27:
### Describe the bug

While the current config already allows several XFRM functionalities, it doesn't offer XFRM interface support.
While "ip link" just fails, an additional tool "xfrmi", which comes with strongswan gives a little bit more information.

To ensure a consistent user experience, I propose to enable the necessary module in the default configuration:

.config
CONFIG_XFRM_INTERFACE=m

for further convenience, I also suggest to add
CONFIG_XFRM_STATISTICS=y
CONFIG_XFRM_SUB_POLICY=y

### Steps to reproduce the behaviour

root@raspberrypi:~# ip link add az0 type xfrm dev br0 if_id 42
Error: Unknown device type.

root@raspberrypi:~# /usr/lib/ipsec/xfrmi --name az0 --id 42 --dev br0
received netlink error: Operation not supported (95)
failed to create XFRM interface

### Device (s)

Raspberry Pi 4 Mod. B

### System

Linux raspberrypi 6.1.25-v7l+ #1646 SMP Tue Apr 25 18:21:36 BST 2023 armv7l GNU/Linux

### Logs

root@raspberrypi:~# ip link add az0 type xfrm dev br0 if_id 42
Error: Unknown device type.

root@raspberrypi:~# /usr/lib/ipsec/xfrmi --name az0 --id 42 --dev br0
received netlink error: Operation not supported (95)
failed to create XFRM interface

pelwell, 2023-04-27:
That seems reasonable - see https://github.com/raspberrypi/linux/pull/5447.

If you wait about an hour for the check builds to complete you'll be able to install a test build using `sudo rpi-update pulls/5447`.

SaschaJohn, 2023-04-27:
@pelwell I can confirm, that with the new kernel pulls/5447, it is now possible to create xfrm interfaces.
Thank you for enabling this that fast!

root@raspberrypi:~# ip link add az0 type xfrm dev br0 if_id 42
root@raspberrypi:~# ip addr sh az0
27: az0@br0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/none

closed 2023-04-27 by SaschaJohn

pelwell, 2023-04-27:
Thanks for the feedback - the PR has been merged and the additional XFRM support will be in future releases.