raspberrypi/linux #5454: Raspberry PiOS lite bluetooth audio stutter on kernel  6.1.21-v8+

adrianalin, 2023-04-30:
### Describe the bug

Stuttering bluetooth audio with bluetooth headphones ATH-S200BT.

### Steps to reproduce the behaviour

Just pair headphones and play audio, I use mpd to stream internet radio. Using pulseaudio, but same behavior happens with pipewire.

### Device (s)

Raspberry Pi 4 Mod. B

### System

$ cat /etc/rpi-issue 
Raspberry Pi reference 2023-02-21
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 25e2319effa91eb95edd9d9209eb9f8a584d67be, stage2

$ vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

$ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### Logs

[   16.179011] Bluetooth: Core ver 2.22
[   16.179161] Bluetooth: HCI device and connection manager initialized
[   16.179188] Bluetooth: HCI socket layer initialized
[   16.179203] Bluetooth: L2CAP socket layer initialized
[   16.179227] Bluetooth: SCO socket layer initialized
[   16.195660] Bluetooth: HCI UART driver ver 2.3
[   16.195694] Bluetooth: HCI UART protocol H4 registered
[   16.195806] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   16.196124] Bluetooth: HCI UART protocol Broadcom registered
[   16.468004] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.468024] Bluetooth: BNEP filters: protocol multicast
[   16.468039] Bluetooth: BNEP socket layer initialized
[   16.473073] Bluetooth: MGMT ver 1.22
[   32.080253] Bluetooth: RFCOMM TTY layer initialized
[   32.080288] Bluetooth: RFCOMM socket layer initialized
[   32.080311] Bluetooth: RFCOMM ver 1.11
[  212.522218] Bluetooth: Unexpected continuation frame (len 0)
[  212.605914] Bluetooth: Unexpected continuation frame (len 0)
[  212.614682] Bluetooth: Unexpected continuation frame (len 0)
[  212.722180] Bluetooth: Unexpected continuation frame (len 0)
[  212.730971] Bluetooth: Unexpected continuation frame (len 0)
[  212.817197] Bluetooth: Unexpected continuation frame (len 0)
[  212.825934] Bluetooth: Unexpected continuation frame (len 0)
[  212.910947] Bluetooth: Unexpected continuation frame (len 0)
[  212.919722] Bluetooth: Unexpected continuation frame (len 0)
[  213.007188] Bluetooth: Unexpected continuation frame (len 0)
[  223.541199] Bluetooth: Unexpected continuation frame (len 0)

pelwell, 2023-05-02:
I've never seen that error message before. Looking at the kernel BT stack source code it thinks its receiving something bad, but that could be symptom of some other problem (I'm not blaming the headphones yet).

Which WLAN band and channel are you using? WLAN 2.4GHz and BT share the same spectrum _and_ an antenna, so it's better to use a 5GHz channel if possible. If you have to use the 2.4G band, try changing the channel - I seem to remember that channel 1 is best for BT coexistence, but I may have got that wrong.

pelwell, 2023-05-02:
Also, is this a change in behaviour following a recent software update, or is it the first time you've used the headphone+Pi combination?

adrianalin, 2023-05-03:
I installed Arch Linux on my Raspberry Pi 4:
`
$ uname -a
Linux alarm 6.2.10-1-aarch64-ARCH #1 SMP PREEMPT_DYNAMIC Fri Apr  7 10:32:52 MDT 2023 aarch64 GNU/Linux
`
and bluetooth works ok here with same headphones as mentioned before, streaming internet radio over WiFi and playing audio on the bluetooth headphones, using same WiFi connection (2.4GHz).

I used my headphones with Raspberry Pi 3 before with an older version of raspbian (buster) and also worked ok back than (same WiFi, headphones, streaming internet radio).

avevad, 2023-05-15:
I am having the same issue now, is there a Raspbian fix without installing Arch?

adrianalin, 2023-05-16:
Hmm just bought the official rpi4 power supply and now works ok for me, don't know if has anything to do with that. Also ran pacman -Syu on RPI, not sure what fixed it though

pelwell, 2023-06-26:
I'm closing this as it sounds like your issue has been resolved.

closed 2023-06-26 by pelwell

jennifersmith, 2023-12-10:
I have two Raspberry PI 4Bs with the same Bluetooth audio issues as the original submitter describes. Both are on Raspberry Pi OS Lite: one is on bookworm, one on bullseye. Using official power supply in at least one of these machines and the latest firmware, bluez packages etc.

The Bluetooth device paired to to is a fairly old Logitech UE Mobile Boombox speaker.  When playing audio, I notice the logs full of these messages - maybe a 100 a second or so:

```
2023-08-26T11:06:04.998385+1000 rpi1 kernel: Bluetooth: Unexpected continuation frame (len 0)
2023-08-26T11:06:05.174161+1000 rpi1 kernel: Bluetooth: Unexpected continuation frame (len 0)
2023-08-26T11:06:05.194030+1000 rpi1 kernel: Bluetooth: Unexpected continuation frame (len 0)
2023-08-26T11:06:05.198000+1000 rpi1 kernel: Bluetooth: Unexpected continuation frame (len 0)
...
```
This seems to correspond to the poor audio quality/stutter.  Only happens when playing audio (A2DP - SBC). 

Still investigating the root cause (mainly out of curiosity at this point), but it does seem like a patch to the error handling code around [here](https://github.com/raspberrypi/linux/blob/rpi-6.1.y/net/bluetooth/l2cap_core.c#L8544) to apply a rate limit would stop the torrent of log messages if it's something that is not easy to fix. 

## System 

```
cat /etc/rpi-issue
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2

```

```
vcgencmd version
Oct 17 2023 15:39:16
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)
```

Currently on a custom kernel build for reasons, but it was definitely an issue with the original! 

``` 
uname -a
Linux rpi1 6.1.58-v8+customstuff #4 SMP PREEMPT Wed Nov 22 10:53:16 AEDT 2023 aarch64 GNU/Linux

```

Also in case it helps:

```
apt info bluez-firmware
Package: bluez-firmware
Version: 1.2-9+rpt2

strings /lib/firmware/brcm/BCM4345C0.hcd | grep Raspberry
&BCM43455 37.4MHz Raspberry Pi 3+-0190
```

mschilli87, 2024-02-01:
@jennifersmith: I can reproduce this exact issue when streaming audio via BT to my RP4B running the officiale mo0de audio image. Any update or idea how to workaround this? Streaming audio to the Pi to play it back via speakers attached to it is the main use case for that project.

jackbradach, 2025-06-01:
I just hit this same problem!  Ingredients in my case are RPI5 and a Bose Soundlink Mini II from like a decade ago.