raspberrypi/linux #5460: Bluetooth unreliable on Linux 6.1

agners, 2023-05-09:
### Describe the bug

In the latest Home Assistant OS release 10 we moved from the Raspberry Pi Linux kernel 5.15 to 6.1. Since the release 10 we got quite some reports about Bluetooth outages after a couple of hours runtime, specifically on Raspberry Pi devices. The outages on Raspberry Pi devices have the following kerne logs in common:

```
May 02 12:11:40 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 02 12:11:40 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 02 12:11:40 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 02 12:11:40 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 02 12:11:41 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 02 12:11:41 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Received unexpected HCI Event 0x00
May 02 12:11:41 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 02 12:11:47 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 02 12:11:47 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Received unexpected HCI Event 0x00
May 02 12:11:47 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 02 12:11:53 ha-shelf1-rpi3 kernel: Bluetooth: hci0: command 0x200c tx timeout
May 02 12:11:53 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Opcode 0x200c failed: -110
May 02 12:11:53 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Unable to disable scanning: -110
May 02 12:11:53 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 02 12:11:53 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Received unexpected HCI Event 0x00
May 02 12:11:55 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Opcode 0x200c failed: -110
May 02 12:11:55 ha-shelf1-rpi3 kernel: Bluetooth: hci0: command 0x200c tx timeout
May 02 12:11:55 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Unable to disable scanning: -110
May 02 12:11:55 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Unable to disable scanning: -110
May 02 12:11:55 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
May 02 12:11:55 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Unable to disable scanning: -16
May 02 12:11:55 ha-shelf1-rpi3 kernel: Bluetooth: hci0: stop background scanning failed: -16
May 02 12:12:05 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
May 02 12:12:05 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Unable to disable scanning: -16
May 02 12:12:05 ha-shelf1-rpi3 kernel: Bluetooth: hci0: Unable to disable scanning: -16
```

From this point onward it is no longer possible to use Bluetooth. A reboot recovers the system.

I've been able to reproduce the issue locally on the Raspberry Pi 3 Model B out of three Raspberry Pi devices (Raspberry Pi 3 Model B, 3+ Model B and 4). On the device which shows the problem, I've noticed that `Bluetooth: hci0: Frame reassembly failed (-84)` errors have been quite common even on Linux 5.15, but it seems that the older kernel  was able to recover on its own. On other devices, I've not seen the error even on Linux 5.15, so it seems this problem is device specific (only happens on some). However, from other people reports all device types seem affected (Pi 3/3+/Pi 4/CM4).

We are using the full UART for Bluetooth by default.

Since Home Assistant OS also updated other components (like BlueZ) I built a OS dev version using Linux 5.15.84 (tag `1.20230106`), with all other components the latest version as used in HAOS 10.1.  That build showed the reassembly errors too, but recovered just fine. So with this it is proven that the problem is a Linux 6.1 issue.

This is essentially a upstream report of our Home Assistant OS issue https://github.com/home-assistant/operating-system/issues/2485.

### Steps to reproduce the behaviour

1. Install Home Assistant 10.1
2. Enable Bluetooth, probably also requires some Bluetooth BLE devices (e.g. the popular Xiaomi Mi Temperature and Humidity monitors).
3. Wait for reassembly errors to occur and complete communication breakdown (typically 6-12h)

### Device (s)

Raspberry Pi 3 Mod. B, Raspberry Pi 3 Mod. B+, Raspberry Pi 4 Mod. B, Raspberry Pi CM4, Raspberry Pi CM4 Lite

### System

Home Assistant OS 10.1
Linux 6.1.21 (built from the `1.20230405` tag)

pelwell, 2023-05-10:
Bluetooth modem initialisation has changed recently and is now managed by the kernel, but from that point on it should be the same, so I'm surprised at the nature of the errors you and others are seeing.

I'm pleased to hear that you are using the recommended (and default) configuration of using UART0 (ttyAMA0) to drive the BT modem; it is the best option. The Pi 3 is the most troublesome because it doesn't connect up the flow-control signals (those pins were used for something else), and for that reason the baudrate is limited to 921600. However, that shouldn't have changed with the move to kernel modem initialisation.

Wait... tag 1.20230405 predates the big switchover, which went live three weeks later (2023-04-26), so something else must have changed between 5.15.84 and 6.1.21. In a way that's a relief, but it may make it harder to locate the breaking change.

agners, 2023-05-10:
> I'm pleased to hear that you are using the recommended (and default) configuration of using UART0 (ttyAMA0) to drive the BT modem; it is the best option. The Pi 3 is the most troublesome because it [...]

Just double checked, hci0 is indeed running on ttyAMA0. Also, we are using 460800 as baudrate. It seems we've added a [patch a while ago](https://github.com/home-assistant/operating-system/commit/c371e451449f4619b0c69f7f04414e2e8f9ddb1d) to lower the baudrate, and there was also a reference to the https://github.com/raspberrypi/firmware/issues/1150 issue.

That said, this is my test instance. It could be that other folks which report the problem (especially on RPi4/CM4) actually use a different `config.txt`. I'll have to check.

agners, 2023-05-10:
> Is the WiFi connected and in use?

WiFi is not in use and disconnected.

pelwell, 2023-05-10:
Thanks, that means my current test should be at least somewhat representative.

agners, 2023-05-10:
> Bluetooth modem initialisation has changed recently and is now managed by the kernel, but from that point on it should be the same, so I'm surprised at the nature of the errors you and others are seeing.

I see, this is the change you are referring to: https://github.com/raspberrypi/linux/commit/ec08e93837efa2e75e5cab4d80427fdf7ea23da1

Good to see we can push this out to the kernel, it will make userspace simpler. I'll definitely give this a try sometime in the future.

But I think the problem at hand is unrelated to the error we are seeing, as the HCI communication at this point is already under the kernels control. To me it looks like different parsing/error recovery behavior of the Bluetooth HCI handling logic in the kernel.

pelwell, 2023-05-10:
> But I think the problem at hand is unrelated to the error we are seeing, as the HCI communication at this point is already under the kernels control. To me it looks like different parsing/error recovery behavior of the Bluetooth HCI handling logic in the kernel.

Indeed.

> That said, I wonder why we even have these hci0: Frame reassembly failed. Is the low-level UART communication issues or is this higher level buffer overflows (lack of flow control)?

If I knew that...

agners, 2023-05-11:
So with 115200 I did not get any overruns for 24h+. Today I switched back to 921600, and got two events. And it seems those are buffer overruns:

```
# dmesg | grep hci0
[42686.118707] Bluetooth: hci0: Frame reassembly failed (-84)
[42686.118948] Bluetooth: hci0: Frame reassembly failed (-84)
[53440.602688] Bluetooth: hci0: Frame reassembly failed (-84)
[53440.602927] Bluetooth: hci0: Frame reassembly failed (-84)
# cat /proc/tty/driver/ttyAMA 
serinfo:1.0 driver revision:
0: uart:PL011 rev2 mmio:0x3F201000 irq:114 tx:197234 rx:3403475 oe:2 RTS|CTS|DTR
```

I see that the RPi config uses rather conservative values for the scheduler:
```
CONFIG_PREEMPT_VOLUNTARY=y
CONFIG_HZ=100
```

Maybe higher HZ and/or fully preemptive kernel helps here?

agners, 2023-05-11:
That said, it is still interesting that the 5.15 was able to recover from these overruns. So ideally we'd have a Bluetooth stack which can handle the reassembly errors :sweat_smile:

popcornmix, 2023-05-12:
> I've not yet reviewed/checked downstream changes, but either there is an upstream fix between 6.1.21 and 6.1.28 or there is something broken in the downstream kernel.

rpi-update is now on 6.1.28 so testing that may narrow things down.

agners, 2023-05-22:
I've tested with the kernel tree at https://github.com/raspberrypi/linux/commit/cb9e651718d601c6f2ff1116b4c98c11063302f7. I still use the old initialization mechanism, however, since that is only initialization, it shouldn't matter really.

```
May 22 16:25:26 homeassistant kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 22 16:25:26 homeassistant kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 22 17:00:40 homeassistant kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 22 17:00:40 homeassistant kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 22 17:02:28 homeassistant kernel: Bluetooth: hci0: Frame reassembly failed (-84)
May 22 17:02:30 homeassistant kernel: Bluetooth: hci0: command 0x200c tx timeout
May 22 17:02:30 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -110
May 22 17:02:30 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -110
May 22 17:02:30 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
May 22 17:02:30 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
May 22 17:02:30 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
May 22 17:02:30 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
May 22 17:02:30 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
May 22 17:02:30 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
May 22 17:02:40 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
May 22 17:02:40 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
May 22 17:02:40 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
May 22 17:02:40 homeassistant kernel: Bluetooth: hci0: Opcode 0x200c failed: -16
May 22 17:02:40 homeassistant kernel: Bluetooth: hci0: Unable to disable scanning: -16
May 22 17:02:40 homeassistant kernel: Bluetooth: hci0: stop background scanning failed: -16
```

pelwell, 2023-05-23:
I agree that the initialisation method shouldn't affect the issue, but until I can reproduce and understand it I'm not going to rule it out.

Are you seeing overruns with those errors?

agners, 2023-05-23:
> Are you seeing overruns with those errors?

Yes, the overruns strongly correlate with the `Frame assembly failed` errors.

orsolli, 2025-01-22:
I'm having the same problem on NixOS with kernel 6.6

```
[orsolli@nixos:~]$ sudo cat /proc/tty/driver/ttyAMA
serinfo:1.0 driver revision:
1: uart:PL011 rev2 mmio:0x3F201000 irq:99 tx:79590 rx:132218 oe:3 RTS|CTS|DTR

[orsolli@nixos:~]$ uname -a
Linux nixos 6.6.31 #1-NixOS SMP Tue Jan  1 00:00:00 UTC 1980 aarch64 GNU/Linux

[orsolli@nixos:~]$ sudo dmesg | grep hci0
[   14.555305] Bluetooth: hci0: BCM: chip id 94
[   14.555888] Bluetooth: hci0: BCM: features 0x2e
[   14.559261] Bluetooth: hci0: BCM43430A1
[   14.559329] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
[   14.578710] Bluetooth: hci0: BCM43430A1 'brcm/BCM43430A1.hcd' Patch
[   15.403058] Bluetooth: hci0: BCM: features 0x2e
[   15.406236] Bluetooth: hci0: BCM43438A1 37.4MHz Raspberry Pi 3-0141
[   15.406269] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0508
[49635.767451] Bluetooth: hci0: Frame reassembly failed (-84)
[49637.814581] Bluetooth: hci0: command tx timeout
[61649.960308] Bluetooth: hci0: Frame reassembly failed (-90)
[61649.962551] Bluetooth: hci0: Frame reassembly failed (-84)
[61651.999285] Bluetooth: hci0: Received unexpected HCI Event 0x00
[61651.999291] Bluetooth: hci0: Frame reassembly failed (-90)
[61660.969746] Bluetooth: hci0: command 0x200c tx timeout
[61660.969749] Bluetooth: hci0: Opcode 0x200c failed: -110
[61660.974269] Bluetooth: hci0: Unable to disable scanning: -110

```

fujinice, 2026-10-03:
### Additional field report: Pi 3 Model B Rev 1.2, kernel 6.18.50, HCI capture around a scan-control stall

*Report prepared on 2026-10-04; observations below are from 2026-10-03.*

Adding an observation because the first incident resembles this issue, including the subsequent BCM43430A1 build 0508 report. This is **not a confirmed reproduction of the same underlying bug**, and no kernel-version A/B test has been performed.

There are two distinct observations, which should not be counted as repetitions of one established failure mechanism:

- **13:23, original UART speed setting:** a host scan-disable command preceded the last observed HCI advertisement by about 1.5 ms. Its completion was not found before the next scan-disable command; subsequent attempts returned `Command Disallowed`, while several non-scan HCI commands still succeeded. The initiator and reason for the initial disable are unknown.
- **16:28, after lowering the UART speed setting:** the application timed out while HCI advertising reports were still arriving. Scan-control commands succeeded, but later independent D-Bus probes to BlueZ timed out. The available evidence does not identify the internal BlueZ wait or establish a UART cause for this second incident.

#### Environment

- Raspberry Pi 3 Model B Rev 1.2; 32-bit Raspbian GNU/Linux 13 (trixie), version 13.6.
- Kernel: `6.18.50+rpt-rpi-v7`; package `1:6.18.50-1+rpt1`, build dated 2026-09-11.
- BlueZ: `5.82-1.1+rpt2`.
- Bluetooth: BCM43430A1, chip ID 94, `(001.002.009) build 0508`; firmware identification also reports `BCM43438A1 37.4MHz Raspberry Pi 3-0141`.
- `bluez-firmware`: `1.2-13+rpt2`. Loaded HCD file SHA-256: `c096ad4a5c3f06ed7d69eba246bf89ada9acba64a5b6f51b1e9c12f99bb1e1a7`, matching the BCM43430A1 file in RPi-Distro's `pios/trixie` branch when checked.
- Kernel serdev initialization through `hci_uart_bcm` and PL011 at `3f201000.serial` (ttyAMA1). Device Tree specified `max-speed = 921600` for the first incident. After the planned change described below, the effective Device Tree value was `460800`. The actual baud-rate registers have not been independently measured. No `miniuart-bt` override.
- `CONFIG_PREEMPT_VOLUNTARY=y`, `CONFIG_HZ=100`.
- A Python application continuously scans using Bleak `3.0.2` / dbus-fast `5.0.22` through BlueZ. A separate `btmon -i 0 -w ...` recorder rotates captures onto USB storage and preserves incident-adjacent segments. The recorder does not start discovery or change controller power.
- The desktop/audio stack was also running during the checks. Onboard Wi-Fi (`wlan0`) was connected when inspected after recovery; transient Wi-Fi or interrupt load at the failure instant was not measured.
- Uptime at the first, 13:23 incident was approximately 2.94 days. That one occurrence does not establish a recurrence interval.

#### Scanning workload and reproduction limits

The application is intended to keep reception running continuously. The source reviewed for this report creates `BleakScanner` with a detection callback and adapter, without explicit scan-mode, service-UUID, or BlueZ filter arguments. Its normal receive loop has no scheduled scanner stop/start cycle; it checks adapter health through D-Bus and attempts scanner cleanup/reconnection after errors. The incident investigation verified a two-second timeout on the deployed health-property call, and the logs independently show error-triggered reconnection. The exact failing call was not captured.

The current source description is not a reconstruction of every incident-time setting. **Active/passive mode, the effective discovery filters, and the controller's scan interval/window at onset have not been independently established in the evidence summarized here.** The `Filter duplicates: Disabled` field below belongs to that specific HCI command; it is not a full record of the earlier scan configuration. Other discovery clients and the source of the initial disable have not been identified. There is no minimal reproducer or known deterministic trigger yet.

#### First incident: timeline and sanitized evidence

All times below are **2026-10-03 JST (UTC+09:00)**. The command/event rows are selected and normalized from stored btsnoop captures decoded with btmon; this is not an unmodified full trace.

Before the failure, application observations recorded roughly 4,600–4,900 ambient advertising callbacks per minute, with zero application queue drops. HCI-level advertising reports were also present, roughly 140–170 report events per second immediately before the stall. These are different counting layers, not unique-device counts.

```text
13:23:24–25     Kernel records Frame reassembly failed (-84)
                 and SCO packet for unknown connection handle
                 (second-resolution interval in the investigation notes)
13:23:25.781665  Host -> controller: LE Set Scan Enable (0x200c)
                 Scanning: Disabled; Filter duplicates: Disabled
13:23:25.783212  Last observed HCI advertising report

13:23:27.861518  Host -> controller: LE Set Scan Enable (0x200c), Disabled
13:23:27.862001  Controller -> host: Command Complete for 0x200c
                 ncmd 1; Status: Command Disallowed (0x0c)
13:23:27.862163  Host -> controller: LE Set Scan Enable (0x200c), Disabled
13:23:27.862579  Controller -> host: Command Complete for 0x200c
                 ncmd 1; Status: Command Disallowed (0x0c)
```

[... cut ...] des with a failed BlueZ Peer Ping. GDB can perturb timing; this path has not yet been verified against a real recurrence, and symbol availability may limit the result. Targeted BlueZ internal debug logging was prepared but **not activated** during these observations.

Failure-time UART interrupt latency, an internal BlueZ backtrace, and the precise application D-Bus call that timed out are still missing. These observations do not establish a shared root cause with the first incident.

</details>