raspberrypi/linux #5462: RPi 400: speckling in output during mouse cursor movement (but only for 1-of-the-2 microHDMI ports)

jayaddison, 2023-05-10:
**Is this the right place for my bug report?**
I'm not 100% certain that it is, but it's my best-effort-guess so far.

If the nature of the problem/workaround appears more relevant to the forum audience then please accept my apologies.  I'm slightly reluctant to register for another online account to report a single problem (even one that can often be resolved easily), but can do if that would be preferable.

**Describe the bug**
Visual speckling can occur on an HDMI output when using one -- but only one -- of the two microHDMI ports on the RPi 400.

**Workaround / solution**
The workaround/fix -- when possible! -- is to connect the HDMI cable to the other microHDMI port, the one closer to the USB-C power supply.

**To reproduce**

  * Connect the display output to the microHDMI output closest to the RPi 400 keyboard status LEDs (capslock, numlock, power).
  * Boot into a desktop Linux environment (I'm using Debian Linux with LXDE, so nearish-to-but-admittedly-different-from RPiOS).
  * Move the mouse cursor, particularly through the lower 40-ish or so pixels of the display.

**Expected behaviour**
No unexpected visual output should occur, and the display should continue to refresh as usual.

**Actual behaviour**
Visual speckling in dark blue and green occurs within the lower half of the output while the cursor is in motion.

**System**

* Which model of Raspberry Pi?  RPi 400
* Which OS and version?  Debian Linux (bookworm)
* Which firmware version?  Is repeatable for both Debian's `1.20220830+ds-1` and this repository's [`1.20230405`](https://github.com/raspberrypi/firmware/releases/tag/1.20230405)
* Which kernel version? `6.1.0-8`

**Logs**
Available on request.

**Additional context**

  * The system is running Linux kernel 6.1.0-8 from Debian's `linux-image-arm64` version 6.1.25-1 package.

  * The problem is also replicable using the latest '*.{elf,dat,bin}' files from this repository's version `1.20230405` release.

  * The devicetree 'compatible' entry for the gpu is:

```
$ cat /proc/device-tree/gpu/compatible
brcm,bcm2711-vc5
```

  * The vc4 video driver appears to initialize fine according to dmesg (I don't see any errors).

  * The vc4 driver is in use as a framebuffer device, I think.

```
$ dmesg --notime | grep -i vc4 | tail -n 2
[drm] Initialized vc4 0.0.0 20140616 for gpu on minor 1
vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
```

  * I've attempted to use the 'config_hdmi_boost'[1] setting from the documentation to resolve the problem, since it mentions speckling (in fact, it's where I learned the term 'speckling') specifically, but this has not had any effect.  The documentation there does mention that the option is ignored on RPi4, and it seems possible that that may also apply to RPi400.

  * Reducing the resolution of the desktop does not reduce or eliminate the speckling.

Thank you for any help.

Also reported to Debian at: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1035878

pelwell, 2023-05-10:
[ Moved to our Linux repo, where I think it belongs if anywhere ]

Can you reproduce this on a current RPiOS image? https://www.raspberrypi.com/software/operating-systems/

6by9, 2023-05-10:
That sounds like the issue addressed with https://github.com/raspberrypi/linux/commit/013f247b89b6fa55724cb73a820cadcae745d1ee

You've said you've reproduced it with firmware from both Debian and our repo. Have you tested with kernels from both repos?

jayaddison, 2023-05-10:
Thanks @pelwell, @6by9!

In reply to the RPiOS repro question and also this one:

> You've said you've reproduced it with firmware from both Debian and our repo. Have you tested with kernels from both repos?

Not yet, no - but I will try that soon.

> Information about your monitor resolution might also be useful.

This was replicable at 1920x, 1280x, 720x and 640x resolutions, and with both 60hz and 59.xhz frequencies, with the static effect covering approximately the lower half of the display in each case.

popcornmix, 2023-05-10:
If it's not fixed with an updated RPiOS kernel, then a photo/video of the effect may be helpful.

jayaddison, 2023-05-10:
> > You've said you've reproduced it with firmware from both Debian and our repo. Have you tested with kernels from both repos?
> 
> Not yet, no - but I will try that soon.

Good news: the issue is resolved when using kernel version 6.1.21-v8+ from the 2023-05-03 release.  No more visual noise of this kind occurs when moving the mouse cursor around on any part of the screen.

Thank you all for the swift response!  I'll close the GitHub issue, but let me know if any other details/follow-up would help.

closed 2023-05-10 by jayaddison

mripard, 2023-05-11:
Hi,

So I talked with @danvet again about this today since I couldn't find the previous discussion we had.

There's two issues related to that bug, and the patch you mentioned above:

* The legacy cursor API doesn't wait for vblank completion, and it doesn't work for us at all. A patch has been posted [here](https://lore.kernel.org/dri-devel/20230216111214.3489223-1-daniel.vetter@ffwll.ch/), but hasn't been merged yet. We carried it downstream for a while now, and it's the one (mostly) fixing the bug you've discussed here.

* We've have some similar issues with Kodi even with that patch, that eventually ended up with @popcornmix (re)merging the patch @6by9 just linked to.

This one is a bit more troublesome, since the hardware shouldn't use the KMS state by the time it's freed, and thus that last patch shouldn't make a difference. Yet, it does.

It can hint at multiple things, from the vc4 driver signaling vblank too early to the core losing track of something and getting vblank completion wrong eventually.

Discussing it further, it looks like we could try to detect if and what goes wrong by:

* Adding a driver hook right before calling `complete_all(&commit->flip_done)` in `drm_atomic_helper_setup_commit`
* Making sure it also covers async commits
* Implementing that hook in vc4 to check that the DLIST entry in the state isn't the one the hardware considers active. If it is the same one, we complain loudly.

This should be able to detect cases of freeing the state while it's active.

We can further refine this by adding:

* a boolean (`programmed`?) to `vc4_crtc_state` that reports when the state has been programmed (set in `vc4_crtc_atomic_flush` (probably?))
* another boolean (`checked`?) to `vc4_crtc_state`  that reports that the dlist not being active anymore has been checked (set in the function we implemented above)
* Finally, check in `vc4_crtc_destroy_state` that if `programmed` is set, `checked` is set as well. If not, then we complain too.

That part will allow us to see if we're getting refcounting wrong and freeing the state before we're supposed to.

jayaddison, 2023-06-09:
@mripard possibly a silly question, but: is the fact that this only affects one of the two output ports on the device understandable and/or a clue about the cause?

mripard, 2023-06-09:
There's a couple of causes I can think of:

* It only occurs (on the Pi4 at least) if the overscan margins are enabled by the firmware. It could be that we enable them only for HDMI0 and not for HDMI1, causing the issue to not be visible there.
* It's mostly a timing issue, so the order of the connectors might play a role in how X calls the kernel and thus fall in the "right" timing or not to trigger the corruption

It's all theories, of course :)

jayaddison, 2023-06-09:
Ok, the puzzle continues, but good ideas.  Thank you!

mripard, 2023-06-09:
Is it still a puzzle though? The underlying cause seems identified to me, and we have a couple of solutions to fix it?

jayaddison, 2023-06-09:
I think the problematic effects can be solved (for example, by the upgrade to one of the RPiOS kernels), but that doesn't necessarily mean that the cause was identified (and I took your message to mean that we have some theories but are not certain).

Sometimes I think that real challenge in software engineering is to make sure both are addressed, because otherwise we repeatedly/fractally iterate on fixes without reducing the size of the problem-space, but that's getting into philosophy probably.

mripard, 2023-06-13:
> I think the problematic effects can be solved (for example, by the upgrade to one of the RPiOS kernels)

Wait, which kernel are we talking about here?

> but that doesn't necessarily mean that the cause was identified (and I took your message to mean that we have some theories but are not certain).

The cause of the corruption is pretty much identified, see the commit log of 013f247b89b6fa55724cb73a820cadcae745d1ee

All the code path that trigger that might not be, but we definitely know what's going on at the hardware level.

jayaddison, 2023-06-13:
> > I think the problematic effects can be solved (for example, by the upgrade to one of the RPiOS kernels)
> 
> Wait, which kernel are we talking about here?

Upgrading to kernel 6.1.21-v8+ from the `raspberrypi-kernel_1.20230405-1_arm64.deb` package resolved the speckling problem on my RPi 400.

In more detail: I think that was distributed from https://github.com/raspberrypi/firmware/commit/eb18fe00ed39ea8d09c100609f7557a03362ee19 , which references this repository's commit dbcb82357ef21be47841ba39d117b626d715af31 ... and that _does_ include the fix from 013f247b89b6fa55724cb73a820cadcae745d1ee.
 
> > but that doesn't necessarily mean that the cause was identified (and I took your message to mean that we have some theories but are not certain).
> 
> The cause of the corruption is pretty much identified, see the commit log of [013f247](https://github.com/raspberrypi/linux/commit/013f247b89b6fa55724cb73a820cadcae745d1ee)
> 
> All the code path that trigger that might not be, but we definitely know what's going on at the hardware level.

Ok, thanks.  I should check the firmware settings on the device to confirm whether overscan is configured differently for each of the two HDMI ports.  I haven't intentionally adjusted any settings, but it's possible that I did.

jayaddison, 2025-04-29:
For the record: I eventually found time to apply the fix from 013f247b89b6fa55724cb73a820cadcae745d1ee in isolation against a recent Debian Linux 6.12.22 kernel source; only minimal conflict-resolution was required to rebase it.

I tested two builds of that kernel - one without the patch applied, and then one with it applied -- and confirmed that the speckling behaviour on the affected mini-HDMI port no longer occurred after the patch was applied.

With that testing complete, I've updated the [Debian bugthread](https://bugs.debian.org/1035878) and provided the updated copy of the patch that I used there.