raspberrypi/linux #5464: BSC0 and BSC1 used simultaneously can interact in bad ways due to shared interrupt

6by9, 2023-05-10:
### Describe the bug

From https://forums.raspberrypi.com/viewtopic.php?t=350310

Pi DSI 7" display attached and configured.

BSC0 as i2c-10 is polling the DSI touchscreen at 60Hz.
Run i2cdetect -y 1 so BSC1 is also busy. i2cdetect will sometimes register some phantom addresses as being present.

### Steps to reproduce the behaviour

DSI panel attached.
Repeatedly run ```sudo i2cdetect -y 1``` and note that phantom addresses show up. Using ```i2cdetect -y -r 1``` (read) shows no phantom addresses, whilst ```i2cdetect -y -q 1``` (quick write) fails.

### Device (s)

Raspberry Pi 3 Mod. B, Raspberry Pi 4 Mod. B

### System

32bit Bullseye install.
6.1.21-v7l+
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

### Logs

With bcm2835_i2c.debug=3, normally each i2cdetect poll will log
```
[ 5286.199248] bcm2835_i2c_xfer: msg(1/1) write addr=0x29, len=0 flags= [i2c1]
[ 5286.199421] start_transfer: msg(1/1) write addr=0x29, len=0 flags= [i2c1]
[ 5286.199435] isr: remain=0, status=0x152 : DONE TXD TXE ERR  [i2c1]
[ 5286.199448] i2c-bcm2835 3f804000.i2c: i2c transfer failed: 100
```
On one capture where a phantom address 0x2e is detected, we get:
```
[ 5286.200132] bcm2835_i2c_xfer: msg(1/1) write addr=0x2d, len=0 flags= [i2c1]
[ 5286.200243] bcm2835_i2c_xfer: msg(1/2) write addr=0x38, len=1 flags= [i2c11]
[ 5286.200263] bcm2835_i2c_xfer: msg(2/2) read addr=0x38, len=63 flags= [i2c11]
[ 5286.204159] start_transfer: msg(1/1) write addr=0x2d, len=0 flags= [i2c1]
[ 5286.204194] isr: remain=0, status=0xa0000151 : TA TXD TXE ERR  [i2c1]
[ 5286.204207] isr: remain=0, status=0xa0000051 : TA TXD TXE  [i2c1]
[ 5286.204219] isr: remain=0, status=0xa0000051 : TA TXD TXE  [i2c1]
[ 5286.204230] isr: remain=0, status=0xa0000051 : TA TXD TXE  [i2c1]
[ 5286.204246] i2c-bcm2835 3f804000.i2c: i2c transfer failed: 100
[ 5286.204351] bcm2835_i2c_xfer: msg(1/1) write addr=0x2e, len=0 flags= [i2c1]
[ 5286.204394] start_transfer: msg(1/1) write addr=0x2e, len=0 flags= [i2c1]
[ 5286.204409] isr: remain=0, status=0x10000053 : TA DONE TXD TXE  [i2c1]
```

Having TA active at the same time as DONE would appear at first glance to be the condition that we want to ignore.

A second capture that reported 0x40 isn't quite so clear cut:
```
[ 6152.972887] bcm2835_i2c_xfer: msg(1/1) write addr=0x3f, len=0 flags= [i2c1]
[ 6152.973023] start_transfer: msg(1/1) write addr=0x3f, len=0 flags= [i2c1]
[ 6152.973029] start_transfer: msg(1/2) write addr=0x38, len=1 flags= [i2c11]
[ 6152.973036] isr: remain=0, status=0xa0000151 : TA TXD TXE ERR  [i2c1]
[ 6152.973042] isr: remain=1, status=0x10000055 : TA TXW TXD TXE  [i2c11]
[ 6152.973047] i2c-bcm2835 3f804000.i2c: i2c transfer failed: 100
[ 6152.973051] start_transfer: msg(2/2) read addr=0x38, len=63 flags= [i2c11]
[ 6152.973062] isr: remain=63, status=0x50000051 : TA TXD TXE  [i2c11]
[ 6152.973071] isr: remain=63, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973078] bcm2835_i2c_xfer: msg(1/1) write addr=0x40, len=0 flags= [i2c1]
[ 6152.973079] isr: remain=63, status=0xf0000039 : TA RXR TXD RXD  [i2c11]
[ 6152.973089] isr: remain=51, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973099] isr: remain=51, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973108] isr: remain=51, status=0x40000031 : TA TXD RXD  [i2c11]
[ 6152.973119] isr: remain=51, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973127] isr: remain=51, status=0xd0000039 : TA RXR TXD RXD  [i2c11]
[ 6152.973136] isr: remain=39, status=0x50000051 : TA TXD TXE  [i2c11]
[ 6152.973147] isr: remain=39, status=0x40000031 : TA TXD RXD  [i2c11]
[ 6152.973156] isr: remain=39, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973164] isr: remain=39, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973175] isr: remain=39, status=0xf0000031 : TA TXD RXD  [i2c11]

[ 6152.973175] start_transfer: msg(1/1) write addr=0x40, len=0 flags= [i2c1]
[ 6152.973183] isr: remain=39, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973188] isr: remain=0, status=0x52 : DONE TXD TXE  [i2c1]

[ 6152.973192] isr: remain=39, status=0xd0000039 : TA RXR TXD RXD  [i2c11]
[ 6152.973201] isr: remain=27, status=0x40000031 : TA TXD RXD  [i2c11]
[ 6152.973209] isr: remain=27, status=0x40000031 : TA TXD RXD  [i2c11]
[ 6152.973211] bcm2835_i2c_xfer: msg(1/1) write addr=0x41, len=0 flags= [i2c1]
[ 6152.973219] isr: remain=27, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973227] isr: remain=27, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973235] isr: remain=27, status=0xf0000031 : TA TXD RXD  [i2c11]
[ 6152.973244] isr: remain=27, status=0xd0000039 : TA RXR TXD RXD  [i2c11]
[ 6152.973252] isr: remain=15, status=0x50000051 : TA TXD TXE  [i2c11]
[ 6152.973261] isr: remain=15, status=0x40000031 : TA TXD RXD  [i2c11]
[ 6152.973271] isr: remain=15, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973280] isr: remain=15, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973288] isr: remain=15, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973297] isr: remain=15, status=0xd0000031 : TA TXD RXD  [i2c11]
[ 6152.973305] isr: remain=15, status=0xf0000039 : TA RXR TXD RXD  [i2c11]
[ 6152.973315] isr: remain=3, status=0x50000031 : TA TXD RXD  [i2c11]
[ 6152.973324] isr: remain=3, status=0x32 : DONE TXD RXD  [i2c11]
[ 6152.973334] isr: remain=0, status=0x50 : TXD TXE  [i2c11]
[ 6152.973361] start_transfer: msg(1/1) write addr=0x41, len=0 flags= [i2c1]
[ 6152.973374] isr: remain=0, status=0x152 : DONE TXD TXE ERR  [i2c1]
[ 6152.973385] i2c-bcm2835 3f804000.i2c: i2c transfer failed: 100
[ 6152.973408] bcm2835_i2c_xfer: msg(1/1) write addr=0x42, len=0 flags= [i2c1]
[ 6152.973559] start_transfer: msg(1/1) write addr=0x42, len=0 flags= [i2c1]
[ 6152.973572] isr: remain=0, status=0x152 : DONE TXD TXE ERR  [i2c1]
[ 6152.973583] i2c-bcm2835 3f804000.i2c: i2c transfer failed: 100
```
```status=0x52 : DONE TXD TXE``` exactly matches a device having responded, so I'm not so sure that one can be worked around.

pelwell, 2023-05-10:
The old i2c-bcm2708 driver uses a spinlock to make the master_xfer method atomic. The isr also claims the spinlock, to add further protection on multi-core systems.

You try adding `dtoverlay=i2c-bcm2708` and/or adding spinlocks to i2c-bcm2835.

6by9, 2023-05-10:
The spinlock taken for bcm2708_i2c_master_xfer setup is per device instance, so it's only atomic within each bus. I think the framework already guarantees you that (if not then that's a bug in i2c-bcm2835).

I guess it depends on whether this is the isr and bcm2835_i2c_xfer both accessing state at the same time, or there being the potential for the hardware to return weird state conditions whilst the transfer is in progress (hence causing issues if the isr is called due to the other bus).

I'll try the test and see what happens.

6by9, 2023-05-10:
Adding ```dtoverlay=i2c-bcm2708``` makes no difference - I still get phantom addresses when using SMBus "quick write" mode on i2c-1 whilst the touchscreen is polling away on i2c-10.
That driver has less logging, so it's not as easy to see what's going on.