raspberrypi/linux #5468: 64-bit kernel on 32-bit userland breaks DKMS compilation

classabbyamp, 2023-05-11:
### Describe the bug

when compiling DKMS modules, it tries to look for 64-bit headers and files from the kernel, which do not exist

`echo arm_64bit=0 >> /boot/config.txt` does not fix it on subsequent boots

(as a distro maintainer) how can we disable/revert the 64-on-32 change by default/system-wide? We were already using 64-bit kernels on aarch64-compatible RPIs

### Steps to reproduce the behaviour

1. build the kernel for armv6l with `ARCH=arm bcmrpi_defconfig`
2. this is the make command during dkms module compilation that fails: `make -j $(nproc) -C /lib/modules/6.1.27_1/build prepare0` (logs below)

### Device (s)

Raspberry Pi 1 Mod. B

### System

- OS: Void Linux
- Firmware: `d7f9c2b4ef7e4a8c0b04374a879ce89d7a948453`
- Kernel: `Linux puddle-jumper 6.1.27_1 #1 Wed May 10 17:13:18 UTC 2023 armv6l GNU/Linux`

### Logs

```
make: Entering directory '/usr/src/rpi-kernel-headers-6.1.27_1'
  SYNC    include/config/auto.conf.cmd
crypto/Kconfig:1399: can't open file "arch/arm64/crypto/Kconfig"
make[2]: *** [scripts/kconfig/Makefile:77: syncconfig] Error 1
make[1]: *** [Makefile:704: syncconfig] Error 2
make: *** [Makefile:805: include/config/auto.conf.cmd] Error 2
make: Leaving directory '/usr/src/rpi-kernel-headers-6.1.27_1'
```

### Additional context

related to https://github.com/void-linux/void-packages/pull/43600

tsjk, 2023-05-11:
I'm not sure that this isn't an artefact of us decimating the build directory. The file that raises the complaint should normally be around, and I wouldn't be surprised if stuff were not coded to cope with decimated kernel sources.
But, one could perhaps draw inspiration from Raspberry Pi OS; a version with kernel v6.1 was recently released.
I for one would attempt to not decimate the installed sources as much and see what that yields.

classabbyamp, 2023-05-11:
no, it does not make sense that arm64 files would exist for a kernel built for arm(32). see also #5408 #5410

tsjk, 2023-05-11:
I'm sure you're right. But, bundling the arch dirs has the same effect as deleting the if-clauses: An infinite loop of
```
scripts/Makefile.build:425: warning: overriding recipe for target 'built-in.a'
Makefile:1259: warning: ignoring old recipe for target 'built-in.a'
scripts/Makefile.build:440: warning: overriding recipe for target 'modules.order'
Makefile:1923: warning: ignoring old recipe for target 'modules.order'
```
which seems to result from
```
make -f ./scripts/Makefile.build obj=. prepare
```
being called recursively.
I do understand the possibility of modules failing to build anyway, even if dkms succeeds.
But I mean, I don't understand how this is runtime issue (albeit I'm a novice); dkms and kernel modules are compiled when running the old (v5.15) kernel - before reboot.
I think it's difficult to actually gather intel from those threads. Is it even possible to boot a 64-bit kernel on the Pi 1B?!
And just to confirm: Does dkms work on void after installing 6.1 on the rpi4?

classabbyamp, 2023-09-25:
seems this was due to missing files (namely the root `Kbuild` file, which was causing the infinite loop)

closed 2023-09-25 by classabbyamp