raspberrypi/linux #5470: USB 2.0 driver fails with kernel update from 6.1.27 to  6.1.28-3-rpi-ARCH

baslking, 2023-05-16:
### Describe the bug

USB 2.0 generates kernel trace at boot and then does not correctly detect or use USB connected devices.

expected behavior of 
~lsusb
Bus 001 Device 003: ID 0463:ffff MGE UPS Systems UPS
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Kernel 28 behavior:
~lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### Steps to reproduce the behaviour

Updated kernel no longer detects USB devices boot trace shown below.
Reverting to 6.1.27 kernel resolves the issue 

### Device (s)

Raspberry Pi CM4

### System

NAME="Arch Linux ARM"
PRETTY_NAME="Arch Linux ARM"
ID=archarm
ID_LIKE=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinuxarm.org/"
DOCUMENTATION_URL="https://archlinuxarm.org/wiki"
SUPPORT_URL="https://archlinuxarm.org/forum"
BUG_REPORT_URL="https://github.com/archlinuxarm/PKGBUILDs/issues"

pacman -Qs raspberry
local/firmware-raspberrypi 20230125-1
    Additional firmware for Raspberry Pi
local/raspberrypi-bootloader 20230512-1
    Bootloader files for Raspberry Pi

This kernel works:
Linux pi-nas 6.1.27-1-rpi-ARCH #1 SMP PREEMPT Tue May  2 18:32:20 MDT 2023 aarch64 GNU/Linux
This kernel does not:
Linux pi-nas 6.1.28-3-rpi-ARCH #1 SMP PREEMPT Mon May 15 12:00:19 MDT 2023 aarch64 GNU/Linux

### Logs

journalctl -b snippet:
------------[ cut here ]------------
May 16 11:09:01 pi-nas kernel: bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
May 16 11:09:01 pi-nas kernel: dwc2 fe980000.usb: swiotlb addr 0xffffffffffffffff+8 overflow (mask ffffffff, bus limit ffffffff).
May 16 11:09:01 pi-nas kernel: bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
May 16 11:09:01 pi-nas kernel: WARNING: CPU: 1 PID: 155 at kernel/dma/swiotlb.c:885 swiotlb_map+0x2b4/0x2c4
May 16 11:09:01 pi-nas kernel: bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
May 16 11:09:01 pi-nas kernel: Modules linked in: bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C+) rpivid_hevc(C) rtc_pcf85063 bcm2835_mmal_vchiq(C)
May 16 11:09:01 pi-nas kernel: bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
May 16 11:09:01 pi-nas kernel: v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_bcm2835(C) videobuf2_common snd_pcm vc_sm>
May 16 11:09:01 pi-nas kernel: CPU: 1 PID: 155 Comm: kworker/1:2 Tainted: P C O 6.1.28-3-rpi-ARCH #1
May 16 11:09:01 pi-nas kernel: Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
May 16 11:09:01 pi-nas kernel: Workqueue: usb_hub_wq hub_event
May 16 11:09:01 pi-nas kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
May 16 11:09:01 pi-nas kernel: pc : swiotlb_map+0x2b4/0x2c4
May 16 11:09:01 pi-nas kernel: lr : swiotlb_map+0x2b4/0x2c4
May 16 11:09:01 pi-nas kernel: sp : ffffffc0086bb880
May 16 11:09:01 pi-nas kernel: x29: ffffffc0086bb880 x28: ffffffd7660a53a0 x27: ffffff8101111800
May 16 11:09:01 pi-nas kernel: x26: ffffffd7657e2a38 x25: ffffffffffffffff x24: 0000000000000000
May 16 11:09:01 pi-nas kernel: x23: ffffff810020e410 x22: 0000000000000001 x21: 0000000000000000
May 16 11:09:01 pi-nas kernel: x20: 0000000000000008 x19: 000000010924f380 x18: 0000000000000006
May 16 11:09:01 pi-nas kernel: x17: 6d2820776f6c6672 x16: 65766f20382b6666 x15: 6666666666666666
May 16 11:09:01 pi-nas kernel: x14: 0050a68d1f24b046 x13: ffffffd76572b6a0 x12: ffffffd765f28910
May 16 11:09:01 pi-nas kernel: x11: 0000000000000001 x10: 0000000000001a80 x9 : ffffffd7656eee38
May 16 11:09:01 pi-nas kernel: x8 : ffffffc0086bb5f8 x7 : 0000000000000000 x6 : 00000000000000d0
May 16 11:09:01 pi-nas kernel: x5 : ffffffd765f29000 x4 : ffffffd765f29070 x3 : 0000000000000000
May 16 11:09:01 pi-nas kernel: x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff810322be00
May 16 11:09:01 pi-nas kernel: Call trace:
May 16 11:09:01 pi-nas kernel: swiotlb_map+0x2b4/0x2c4
May 16 11:09:01 pi-nas kernel: dma_map_page_attrs+0xa0/0x200
May 16 11:09:01 pi-nas kernel: usb_hcd_map_urb_for_dma+0x170/0x470
May 16 11:09:01 pi-nas kernel: dwc2_map_urb_for_dma+0x58/0x130 [dwc2]
May 16 11:09:01 pi-nas kernel: usb_hcd_submit_urb+0xb0/0x950
May 16 11:09:01 pi-nas kernel: usb_submit_urb+0x29c/0x5c0
May 16 11:09:01 pi-nas kernel: usb_start_wait_urb+0x78/0x11c
May 16 11:09:01 pi-nas kernel: usb_control_msg+0xc8/0x140
May 16 11:09:01 pi-nas kernel: hub_port_init+0x3dc/0xc80
May 16 11:09:01 pi-nas kernel: hub_event+0x968/0x159c
May 16 11:09:01 pi-nas kernel: process_one_work+0x200/0x474
May 16 11:09:01 pi-nas kernel: worker_thread+0x74/0x43c
May 16 11:09:01 pi-nas kernel: kthread+0xfc/0x110
May 16 11:09:01 pi-nas kernel: ret_from_fork+0x10/0x20
May 16 11:09:01 pi-nas kernel: ---[ end trace 0000000000000000 ]---

graysky2, 2023-05-16:
For context:
* 3.1.28-3 is built from 1229967a3ab861660b6d31ccdc8526bd44712b13
* 3.1.27-1 is build from 83cf6b410113c98ebfd0b7fef70b1a7c9539b0b9

baslking, 2023-05-16:
tail -1 /proc/cpuinfo
Model		: Raspberry Pi Compute Module 4 Rev 1.0

config.txt
# See /boot/overlays/README for all available options

#dtoverlay=vc4-kms-v3d
initramfs initramfs-linux.img followkernel

# USB 2.0
dtoverlay=dwc2,dr_mode=host
#dtoverlay=dwc-otg
#dtoverlay=dwc2
# Uncomment to enable bluetooth
#dtparam=krnbt=on
# Disable BT and WIFI
dtoverlay=disable-wifi
dtoverlay=disable-bt
# Activate RTC
dtoverlay=i2c-rtc,pcf85063a,i2c_csi_dsi
# bcm2708_fb set Framebuffers to zero to save RAM and avoid boot error
max_framebuffers=0
[pi4]
# Run as fast as firmware / board allows
arm_boost=1

No custom .dtb files *as far as I know* 

The fdt:
https://chezking-my.sharepoint.com/:u:/g/personal/brad_kingsrus_net/Eec-3X1BByFKk1CvjDB98swBNFukI5qjlb8YBAXuSlKpxw?e=f5yPTK

pelwell, 2023-05-16:
Your dtb may not be custom, but it also isn't the dtb that goes with that kernel version. `ls -l /proc/device-tree/soc/dma-ranges` should report a length of 32, but yours has 16. Similarly,  `ls -l /proc/device-tree/scb/dma-ranges` should report a length of 48, but yours has 24.

pelwell, 2023-05-16:
Try this version: https://github.com/raspberrypi/rpi-firmware/raw/e5f7c2648572b7acbc4fbc0e654281ea2d2e94bb/bcm2711-rpi-cm4.dtb

baslking, 2023-05-16:
Sorry my dtb is the one downloaded ruunning 6.1.27, since I backed it off.    Would the be the same?

pelwell, 2023-05-16:
No - the set of patches leading up to https://github.com/raspberrypi/linux/commit/1229967a3ab861660b6d31ccdc8526bd44712b13 requires that the drivers and dtb be updated together (something we usually try to avoid, but in general the dtbs should be treated as if they were part of the kernel).

baslking, 2023-05-16:
I assumed so.  I pushed it back to 2.1.28 and yours and mine do match

pelwell, 2023-05-16:
And yet it doesn't work? Please link to the new fdt file for comparison.

pelwell, 2023-05-16:
Ah - it looks like you have an 8GB CM4 (I would keep your location secret!), whereas mine is a 4GB. As a diagnostic tool, does adding `total_mem=4096` to config.txt make it work (at the cost of halving your RAM)?

baslking, 2023-05-16:
Yes that fixes it, but 2x less RAM is a costly fix.

pelwell, 2023-05-16:
I'm seeing something strange, which I'm investigating. In the meantime, does replacing:
```
dtoverlay=dwc2,dr_mode=host
```
with
```
otg_mode=1
```
(which is our default configuration on CM4) help? You can remove the `total_mem=4096` for this test.

baslking, 2023-05-16:
Yes `otg_mode=1` with 8GB RAM worked; the boot log is clean, and the IO errors are gone.

Could that be documented in `/boot/overlays/README` ?

Thanks

pelwell, 2023-05-16:
It's not an overlay thing - it's one of the other config.txt settings, and documented here: https://www.raspberrypi.com/documentation/computers/config_txt.html#otg_mode-raspberry-pi-4-only

pelwell, 2023-05-16:
I hope to find and fix the DMA problem tomorrow.

graysky2, 2023-05-17:
@pelwell - 

> with otg_mode=1 (which is our default configuration on CM4) 

Can you point me to the RPiOS default?  If I can add this to Arch ARM's default for CM4 I will.

pelwell, 2023-05-17:
Yes, that [cm4] section looks very much like (if not the same as) ours.

baslking, 2023-05-17:
My install is over 2 years old, and none of the config.txt.* files have otg_mode in them 
@pelwell Yes I'll be glad to test.  I can temporarily switch back to the overlay that was failing to test the dma-range fix, The otg_mode option is working well and I'll use it going forward.

baslking, 2023-05-18:
**WARNING** : I have backed off to 
`Linux pi-nas 6.1.28-3-rpi-ARCH #1 SMP PREEMPT Mon May 15 12:00:19 MDT 2023 aarch64 GNU/Linux`
My PCI bus ATA  disk array running a zfs pool, started showing errors and kernel logging bus errors.  I have one disk that has misbehaved in the past, but now 3 of 4 disks have errors.   Since backing off to 28-3, I have not seen any new errors.  I see no clear difference in the kernel logs regarding the bus

graysky2, 2023-05-18:
Unrelated to the initial issue, but I recall that, at least at some point in time, the use of ECC memory was highly encouraged for ZFS devices?

baslking, 2023-05-18:
ECC would be nice, and there is some legends around this, the idea being that a scrub operation to check parity with a stuck bit in ram would corrupt the disk, but it's a poor man's NAS based on a CM4 :-)  
I'm running a scrub right now (which is hammering the PCI bus) and I've seen no bus errors 
With `Linux pi-nas 6.1.28-5-rpi-ARCH #1 SMP PREEMPT Wed May 17 12:46:27 MDT 2023 aarch64 GNU/Linux`
I get logs like this:
```
May 17 23:00:29 pi-nas kernel: ata2.00: status: { DRDY }
May 17 23:00:29 pi-nas kernel: ata2.00: cmd ea/00:00:00:00:00/00:00:00:00:00/a0 tag 17
                                        res 40/00:80:f8:0a:81/00:00:6a:00:00/40 Emask 0x10 (ATA bus error)
May 17 23:00:29 pi-nas kernel: ata2.00: failed command: FLUSH CACHE EXT
May 17 23:00:29 pi-nas kernel: ata2: SError: { RecovComm PHYRdyChg 10B8B Dispar }
May 17 23:00:29 pi-nas kernel: ata2.00: irq_stat 0x80400000, PHY RDY changed
May 17 23:00:29 pi-nas kernel: ata2.00: exception Emask 0x10 SAct 0x0 SErr 0x190002 action 0xe frozen
May 17 23:00:27 pi-nas kernel: ata2: EH complete
May 17 23:00:27 pi-nas kernel: ata2.00: retrying FLUSH 0xea Emask 0x10
May 17 23:00:27 pi-nas kernel: ata2.00: configured for UDMA/33
May 17 23:00:26 pi-nas kernel: ata2: SATA link up 1.5 Gbps (SStatus 113 SControl 310)
May 17 23:00:24 pi-nas kernel: ata2: hard resetting link
May 17 23:00:24 pi-nas kernel: ata2.00: status: { DRDY }
May 17 23:00:24 pi-nas kernel: ata2.00: cmd ea/00:00:00:00:00/00:00:00:00:00/a0 tag 23
                                        res 40/00:b0:a8:37:00/00:00:66:00:00/40 Emask 0x10 (ATA bus error)
May 17 23:00:24 pi-nas kernel: ata2.00: failed command: FLUSH CACHE EXT
May 17 23:00:24 pi-nas kernel: ata2: SError: { RecovComm PHYRdyChg 10B8B Dispar }
May 17 23:00:24 pi-nas kernel: ata2.00: irq_stat 0x80400000, PHY RDY changed
May 17 23:00:24 pi-nas kernel: ata2.00: exception Emask 0x10 SAct 0x0 SErr 0x190002 action 0xe frozen
May 17 22:59:25 pi-nas kernel: ata2: EH complete
May 17 22:59:25 pi-nas kernel: zio pool=poolz vdev=/dev/disk/by-id/ata-TOSHIBA_HDWL120_X9DFPDA8T-part1 error=5 type=2 offset=837678829568 size=4096 flags=>
May 17 22:59:25 pi-nas kernel: I/O error, dev sdb, sector 1636093512 op 0x1:(WRITE) flags 0x700 phys_seg 1 prio class 2
May 17 22:59:25 pi-nas kernel: sd 1:0:0:0: [sdb] tag#14 CDB: opcode=0x2a 2a 00 61 84 ce 48 00 00 08 00
May 17 22:59:25 pi-nas kernel: sd 1:0:0:0: [sdb] tag#14 ASC=0x21 ASCQ=0x4 
May 17 22:59:25 pi-nas kernel: sd 1:0:0:0: [sdb] tag#14 Sense Key : 0x5 [current] 
May 17 22:59:25 pi-nas kernel: sd 1:0:0:0: [sdb] tag#14 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=DRIVER_OK cmd_age=3s
May 17 22:59:25 pi-nas kernel: zio pool=poolz vdev=/dev/disk/by-id/ata-TOSHIBA_HDWL120_X9DFPDA8T-part1 error=5 type=2 offset=876179505152 size=4096 flags=>
May 17 22:59:25 pi-nas kernel: I/O error, dev sdb, sector 1711290144 op 0x1:(WRITE) flags 0x700 phys_seg 1 prio class 2
May 17 22:59:25 pi-nas kernel: sd 1:0:0:0: [sdb] tag#13 CDB: opcode=0x2a 2a 00 66 00 37 20 00 00 08 00
May 17 22:59:25 pi-nas kernel: sd 1:0:0:0: [sdb] tag#13 ASC=0x21 ASCQ=0x4 
May 17 22:59:25 pi-nas kernel: sd 1:0:0:0: [sdb] tag#13 Sense Key : 0x5 [current] 
May 17 22:59:25 pi-nas kernel: sd 1:0:0:0: [sdb] tag#13 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=DRIVER_OK cmd_age=3s
May 17 22:59:25 pi-nas kernel: ata2.00: configured for UDMA/33
May 17 22:59:25 pi-nas kernel: ata2: SATA link up 1.5 Gbps (SStatus 113 SControl 310)
May 17 22:59:22 pi-nas kernel: ata2: hard resetting link
May 17 22:59:22 pi-nas kernel: ata2.00: status: { DRDY }
May 17 22:59:22 pi-nas kernel: ata2.00: cmd 61/08:70:48:ce:84/00:00:61:00:00/40 tag 14 ncq dma 4096 out
                                        res 40/00:98:c0:49:00/00:00:66:00:00/40 Emask 0x10 (ATA bus error)
May 17 22:59:22 pi-nas kernel: ata2.00: failed command: WRITE FPDMA QUEUED
May 17 22:59:22 pi-nas kernel: ata2.00: status: { DRDY }
May 17 22:59:22 pi-nas kernel: ata2.00: cmd 61/08:68:20:37:00/00:00:66:00:00/40 tag 13 ncq dma 4096 out
                                        res 40/00:98:c0:49:00/00:00:66:00:00/40 Emask 0x10 (ATA bus error)
May 17 22:59:22 pi-nas kernel: ata2.00: failed command: WRITE FPDMA QUEUED
May 17 22:59:22 pi-nas kernel: ata2: SError: { RecovComm PHYRdyChg 10B8B Dispar }
May 17 22:59:22 pi-nas kernel: ata2.00: irq_stat 0x80400000, PHY RDY changed
May 17 22:59:22 pi-nas kernel: ata2.00: exception Emask 0x10 SAct 0x6000 SErr 0x190002 action 0xe frozen
May 17 22:59:15 pi-nas kernel: ata2: EH complete
May 17 22:59:15 pi-nas kernel: zio pool=poolz vdev=/dev/disk/by-id/ata-TOSHIBA_HDWL120_X9DFPDA8T-part1 error=5 type=2 offset=837678690304 size=4096 flags=>
May 17 22:59:15 pi-nas kernel: I/O error, dev sdb, sector 1636093240 op 0x1:(WRITE) flags 0x700 phys_seg 1 prio class 2
```

akuropka, 2023-06-06:
@graysky2,
> did you forget to diff `/boot/config.txt` against `/boot/config.txt.pacnew` perhaps?

Not directly related to this issue but on my system for config.txt there's never any pacnew file generated.

graysky2, 2023-06-06:
That is because you have not modified `/boot/config.txt`, see: https://github.com/archlinuxarm/PKGBUILDs/blob/master/core/linux-rpi/PKGBUILD#L128-L131

akuropka, 2023-06-06:
I have modified `/boot/config.txt` with Bluetooth disabled (uncommented one line) and one HDMI tweak (added one line). Otherwise my file would have been up to date which was not the case either.