raspberrypi/linux #5472: Linux 6.3 gpio-pwm fails to compile

cbxbiker61, 2023-05-17:
### Describe the bug

drivers/gpio/gpio-pwm.c: In function 'pwm_gpio_probe':
drivers/gpio/gpio-pwm.c:104:22: error: 'struct gpio_chip' has no member named 'of_node'; did you mean 'fwnode'?

I've just been commenting out the line.  Not sure that's the correct solution, but it is the only line of code that attempts to use gpio_chip.of_node.

### Steps to reproduce the behaviour

Attempt to compile 6.3 tree with gpio-pwm enabled.

### Device (s)

Raspberry Pi 4 Mod. B

### System

uname aarch64
firmware is latest
distro is Xilka

pelwell, 2023-05-18:
That should be fixed now, although I've only confirmed that it compiles.

6by9, 2023-05-18:
It looks like it was copy/paste from gpio-bcm-virt and gpio-fsm, both of which used to set gc.of_node.

Looking at ```gpiochip_add_data_with_key``` I think we don't need to do it as it'll be set based on the parent field anyway. Both the other 2 drivers lost setting of_node and aren't setting fwnode.
https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/gpio/gpiolib.c#L608-L611