raspberrypi/linux #5474: Kernel drivers don't probe secondary I2C ports

KenVanHoeylandt, 2023-05-18:
### Describe the bug

Problem: When an I2C driver is enabled (e.g. during bootup or with `modprobe`), it only probes `/sys/class/i2c-dev/i2c-1` .
If you [configured an extra I2C port via GPIO](https://medium.com/cemac/creating-multiple-i2c-ports-on-a-raspberry-pi-e31ce72a3eb2) then this port (e.g. `i2c-**2**`) is not probed.

When this probing failed, I verified that my device was presenting itself properly with tools such as `i2cdetect`.
When I moved the device to `i2c-1`, it worked fine.

This was tested with an Arduino acting as [LTC294x](https://github.com/torvalds/linux/blob/master/drivers/power/supply/ltc2941-battery-gauge.c) device, using the I2C kernel driver, attached to a non-standard I2C port.

### Steps to reproduce the behaviour

- Enable an extra I2C port (add `dtoverlay=i2c-gpio,bus=2,i2c_gpio_sda=22,i2c_gpio_scl=23` to `/boot/config.txt`)
- Add I2C device to GPIO pins
- Verify I2C device works (e.g. `i2cdetect` etc.)
- Reboot Pi
- Run `dmesg` and observe error message: 
```
[   61.465919] LTC2941 1-0064: ltc2941 read_reg(0x0[1]) failed: -EREMOTEIO
[   61.465940] LTC2941 1-0064: error -EREMOTEIO: Could not read status register
[   61.465951] LTC2941: probe of 1-0064 failed with error -121
```

### Device (s)

Raspberry Pi CM4

### System

OS and version:
```
Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 005a8c73b05a2cab394073150208bf4f069e861a, stage4
```
Firmware version:
```
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
```
Kernel version:
```
Linux decktility 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

### Logs

```
[   61.465919] LTC2941 1-0064: ltc2941 read_reg(0x0[1]) failed: -EREMOTEIO
[   61.465940] LTC2941 1-0064: error -EREMOTEIO: Could not read status register
[   61.465951] LTC2941: probe of 1-0064 failed with error -121
```

pelwell, 2023-05-18:
I see your additional bus declaration (bus 2 seems a bit low to me, but it might be OK), and you say that `i2cdetect -y 2` shows your device. What I don't see is how you are telling the kernel that you have a device on that bus and that it should try to load a driver for it. I2C is not a discoverable bus, and you need to provide that information in some form or other.

pelwell, 2023-05-18:
That's not discovery, that's just confirming that there is something there.

Do you think the kernel tries all possible I2C drivers on all possible addresses on all busses? Or even all possible drivers on all addresses where something responds? Well, it doesn't.

> edit2: I forgot to mention how I attached the driver:

Yes, you did, but at least we got there in the end.

> The issue being that the target device has to exist on every system otherwise the overlay can't be fully resolved.

Fortunately the use of "target-path" and aliases can help with that. https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/overlays/i2c-sensor-overlay.dts shows how it can be done. One could almost add the ltc294x into that overlay, but it would mean adding the resistor-sense and prescaler-exponent parameters.

pelwell, 2023-05-18:
I've given the issue a more appropriate (i.e. correct) name.

6by9, 2023-05-18:
Discovery is buses like USB, PCI, and PCIe where there is a defined mechanism to read a unique identifier for the card, and that is then used by the kernel to load the correct driver.

I2C uses DT (or ACPI) to tell the kernel what the connected devices should be, and the probe function then may check if something is connected, and ideally check if it is the right thing (not all devices have a suitable register to ID the device).