raspberrypi/linux #5480: ArchLinuxARM linux-rpi 6.1.28-5 not working on Rasperry Pi 4 Model B

nicheosala, 2023-05-23:
### Describe the bug

I tried installing the package [linux-rpi 6.1.28-5](https://archlinuxarm.org/packages/aarch64/linux-rpi) on my Raspberry Pi 4 Model B.\
Then, I rebooted the system: the boot process fails and I can't connect to the Pi.

### Steps to reproduce the behaviour

```
pacman -S linux-rpi
:: linux-rpi and linux-aarch64 are in conflict (linux). Remove linux-aarch64? [y/N] y
:: linux-rpi and uboot-raspberrypi are in conflict. Remove uboot-raspberrypi? [y/N] y
pacman -S --needed raspberrypi-bootloader raspberrypi-firmware
reboot
```

### Device (s)

Raspberry Pi 4 Mod. B

### System

The OS is Arch Linux ARM rpi aarch64.

vcgencmd not available.

This is the output of `uname -a`:
```
Linux pi 6.2.10-1-aarch64-ARCH #1 SMP PREEMPT_DYNAMIC Fri Apr  7 10:32:52 MDT 2023 aarch64 GNU/Linux
```

### Additional context

The Pi runs in headless mode, I connect to it through SSH. I use an SSD instead of an SD card.

I've noticed a user [reporting](https://archlinuxarm.org/forum/viewtopic.php?f=7&t=16143&start=40#p71191) a very similar issue on the Arch Linux ARM forum.

pelwell, 2023-05-23:
You have my sympathies but, while we're happy to help the Arch devs (hi, @graysky2) when we can, they are going to have to do at least the initial diagnosis.

You might be able to give them some better/quicker feedback on that forum thread.

graysky2, 2023-05-23:
I do not do it myself nor know the setup but I know that there are other users booting linux-rpi off SSD.  Best to search the forums for steps.

closed 2023-05-24 by nicheosala

graysky2, 2023-05-24:
@nicheosala - if you identify a thread is would be nice if you can link it here

nicheosala, 2023-05-25:
> @nicheosala - if you identify a thread or some steps to take to achieve this, is would be nice if you can link it here

Yes, the problem was solved on the Arch linux ARM forum, here: https://archlinuxarm.org/forum/viewtopic.php?f=67&t=16479