raspberrypi/linux #5481: Screen doesn't blank after remote session.

arnofkd, 2023-05-24:
### Describe the bug

On Raspberry Pi 4, Raspberry OS (32bit as well as 64bit), after finishing a remote session, the screen will not blank after the designated time (default 600s).

This is under the condition that no local mouse or keyboard input is detected/made after the session disconnect.

### Steps to reproduce the behaviour

1) Raspberry Pi with Raspberry OS (32 or 64bit) installed.

2) No mouse or keyboard connected to Raspberry

3) Install AnyDesk

4) Initiate remote session from a different device into Raspberry

5) Disconnect the remote session

6) Wait 10 minutes+

7) Screen will not blank and continue staying on indefinitely

### Device (s)

Raspberry Pi 400

### System

Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage4

Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

popcornmix, 2023-05-24:
I would say it's more likely this is an issue with the OS than the kernel (and so better reported [here](https://github.com/RPi-Distro/repo/issues/)).

Assuming it only occurs when using AnyDesk, then it becomes much less likely it's kernel related.

To help identify where it is occurring, it may be possible to use dbus-monitor to check for messages related to DPMS.
It also may be possible to check if kernel is requested to blank using `drm.debug` cmdline option (but I suspect this is less likely).

arnofkd, 2023-05-25:
Thank you for pointing me in the right direction!

closed 2023-05-25 by arnofkd