raspberrypi/linux #5483: Crun permission denied in podman since kernel 4.1.28

Elrondo46, 2023-05-25:
### Describe the bug

Use rlimit in podman engine at rootless results as permission denied. Tested in two rpi4 in Arch Arm

### Steps to reproduce the behaviour

Upgrade to last kernels and you can't start any pods

### Device (s)

Raspberry Pi 4 Mod. B

### System

Kernel 6.1.27 or 6.1.28 all subversions in Arch ARM

Elrondo46, 2023-05-26:
sorry for bad kernel number in title

pelwell, 2023-05-26:
Is it a 32-bit or 64-bit Arch? If 32-bit, does adding `arm_64bit=0` to config.txt make a difference?

pelwell, 2023-05-26:
In that case I don't think you'll get much help here, unless you can identify what is causing the problem. You might have more luck asking the Podman devs.

Elrondo46, 2023-05-26:
Hi,

But this is after the kernel 6.1.27, there is a problem. Not before it…

Envoyé le :vendredi 26 mai 2023 11:13
À : ***@***.***>
Cc : ***@***.***>; ***@***.***>

pelwell, 2023-05-26:
What was the last working version? You don't make it clear whether 6.1.27 works or doesn't.

Elrondo46, 2023-05-26:
In 6.1.27, podman in rootless working without problems. Downgrading successfully. It's since 6.1.28 the problem appears.

pelwell, 2023-05-26:
The podman devs are still much more likely to know what is going wrong (especially when you've provided so little information): https://github.com/containers/podman/issues

Elrondo46, 2023-05-26:
Submitted, I post if I have a solution from them of if they redirect to you

https://github.com/containers/podman/issues/18693

pelwell, 2023-05-26:
Thanks - I hope that have some suggestions.

Elrondo46, 2023-05-26:
They redirect the bug solving to a kernel cause

https://github.com/containers/podman/issues/18696#issuecomment-1564234363

Elrondo46, 2023-05-26:
Please note, users have to recreate pods to be compatible with the last kernel 6.1.28 or 6.1.29. Thats the first time I had this problem...

pelwell, 2023-05-26:
So some upstream kernel change (not one of ours) has caused a parameter saved at "pod" creation to no longer be valid. The Podman team may find a way to handle that more gracefully, but for now you have a workaround.

closed 2023-05-26 by pelwell