raspberrypi/linux #5489: imx477 and imx708: changing long exposure shift doesn't rewrite exposure register

6by9, 2023-06-02:
### Describe the bug

Noted when reviewing imx258, and logging so we don't forget.
https://patchwork.linuxtv.org/project/linux-media/patch/20230530173000.3060865-15-dave.stevenson@raspberrypi.com/

Start the sensor and ```__v4l2_ctrl_handler_setup``` will set all the registers for all controls.
Change ```V4L2_CID_VBLANK``` to a value that requires a different value in ```IMX477_LONG_EXP_SHIFT_REG```
```IMX477_REG_EXPOSURE``` will not get rewritten, therefore the exposure is incorrect.

Similarly there is no guarantee that ```__v4l2_ctrl_handler_setup``` will call s_ctrl for ```V4L2_CID_VBLANK``` (and set ```imx477->long_exp_shift```) before s_ctrl ```V4L2_CID_EXPOSURE```, therefore the initial register setting for exposure may be incorrect.

The shift needs to be computed before the check for ```pm_runtime_get_if_in_use```, and the code after that conditional that applies the settings needs to rewrite ```IMX477_REG_EXPOSURE``` if ```IMX477_LONG_EXP_SHIFT_REG``` changes.

This applies for imx477 and imx708.

@naushir for info.

### Steps to reproduce the behaviour

See above

### Device (s)

Raspberry Pi CM4

### System

6.1 kernel

Which7pyallice11, 2023-07-30:
Er lügt ich die hexe bin Linux ich spiele alle spiele

luigi311, 2024-03-26:
Hello Dave,
Based on our discussion in email im going forward with a V2 of your patch series and since this single patch has issues for now im going to go ahead and drop it from the series and we can submit it again upstream when the issue is fixed. I dont have the technical experience to solve it myself so we can go ahead and just wait til you are able to get to this issue.