raspberrypi/linux #5493: [imx219] 10bit 640x480 video at high fps (200+)

vvarma, 2023-06-08:
### Describe the bug

In commit [9ffaf55e](https://github.com/raspberrypi/linux/commit/9ffaf55e126cf4aad3d46198a2fe7b54cc13fbec), the change switches between analog 2x2 binning and the 2x2 binning based on whether its an 8-bit codec or 10-bit one. But in commit [7a41d0f](https://github.com/raspberrypi/linux/pull/5439/commits/7a41d0f88d60d9520f3c11e93a7875a30f097115) the rate factor is set based on mode.

So in this case of a 10bit 640x480 video capture, the normal 2x2 binning is used but with rate_factor set to 2.

I believe this causes the video at high FPS to be "corrupted"

https://github.com/raspberrypi/linux/assets/2176033/b5199fc5-0315-4e9d-a0f2-a3f404cb8146

as compared to the same at 8bit codec or 10-bit codec with lower FPS

https://github.com/raspberrypi/linux/assets/2176033/ab178929-78b7-4039-b0ed-5c2498c0ea72

https://github.com/raspberrypi/linux/assets/2176033/02098d57-d4c3-405d-ae7e-415c4fe5a7cd

### Steps to reproduce the behaviour

```
libcamera-vid --level 4.2 --framerate 200 --buffer-count 10 --mode 640:480:10:P --save-pts t640x480x10x200.pts -o t640x480x10x200.h264 -t 3000 --denoise cdn_off -n
```
vs
```
libcamera-vid --level 4.2 --framerate 200 --buffer-count 10 --mode 640:480:8:P --save-pts t640x480x8x200.pts -o t640x480x8x200.h264 -t 3000 --denoise cdn_off -n
```
or 
```
libcamera-vid --level 4.2 --framerate 100 --buffer-count 10 --mode 640:480:10:P --save-pts t640x480x10x100.pts -o t640x480x10x100.h264 -t 3000 --denoise cdn_off -n
```

### Device (s)

Raspberry Pi 4 Mod. B

### System

> cat /etc/rpi-issue
```
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage2
```
>  vcgencmd version
```
Mar 17 2023 10:50:39
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
```
> uname -a
```
Linux nvrpi-2 6.1.31-v8+ #4 SMP PREEMPT Tue Jun  6 20:34:04 +08 2023 aarch64 GNU/Linux
```

6by9, 2023-06-09:
Ah the joys of mainline backporting small fixes to stable kernels.

You are right, however having fixed that up there is a further issue that having dropped to normal binning for 10bit readout we max out at 103fps for 10bit modes, and 206fps is only available in 8bit mode.
```
pi@raspberrypi:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx219 [3280x2464] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [103.33 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [83.70 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
```

Looking back over [the original issue,](https://github.com/raspberrypi/libcamera-apps/issues/281#issuecomment-1082894118), 640x480 raw10 does work in special binning mode, it's only 1640x1232 raw 10 that didn't.

Mainline doesn't have https://github.com/raspberrypi/linux/pull/5439/commits/7a41d0f88d60d9520f3c11e93a7875a30f097115, so I'm wondering if that is erroneous, and https://github.com/raspberrypi/linux/commit/9ffaf55e126cf4aad3d46198a2fe7b54cc13fbec is the correct one.

vvarma, 2023-06-11:
@6by9 I have attempted a fix for the above issue in the linked PR. Similar logic is used to ascertain both rate_factor and binning mode. Please let me know of any comments.

6by9, 2023-06-13:
Fixed via #5498

closed 2023-06-13 by 6by9