raspberrypi/linux #5494: Linux rpi 4.19.57-2-ARCH causes kernel panic on Raspberry Pi 2 (ArchLinux ARM) #3087

nightBulb, 2023-06-08:
### Describe the bug

The issue #3087 has re-occurred with `Linux rpi 4.19.57-2-ARCH #1 SMP PREEMPT Sun Jul 14 19:05:33 UTC 2019 armv7l GNU/Linux`

The R Pi 2 refuses to boot with Read/64 errors and Kernel Panic during boot.

The workaround of #3087 i.e. setting `gpu_mem`  setting to 256 from 64 in boot/config.txt
is working.
(But wastes lot of (limited) RAM)

### Steps to reproduce the behaviour

Update Arch Linux ARM, use kernel Linux rpi 4.19.57-2-ARCH and above.

And Reboot.
(Use gpu_mem=64 in boot/config.txt) for boot failure.

### Device (s)

Raspberry Pi 2 Mod. B

### System

* **OS**
Arch Linux ARM

* **uname -a**
Linux rpi 4.19.57-2-ARCH #1 SMP PREEMPT Sun Jul 14 19:05:33 UTC 2019 armv7l GNU/Linux

### Additional context

#3087 Occurred in 2019,
Relevant comment: https://github.com/raspberrypi/linux/issues/3087#issuecomment-520234959,
Something about the kernel config options CONFIG_VMSPLIT_3G=y and CONFIG_HIGHMEM=y. 

Also, https://github.com/raspberrypi/linux/issues/3087#issuecomment-520976492 this comment,
from Arch Linux ARM (maintainer) @graysky2, and conversation ahead might be relevant.

graysky2, 2023-06-08:
You are running an insanely out-of-date kernel.  Are the other packages on your system from that era as well??  Recommend you update ASAP.

nightBulb, 2023-06-08:
@graysky2  Sorry for the bad/incomplete bug report, I was actually facing boot issues, and somehow got my tar based root backups mixed up.

I'm currently updating the booted systemd (R Pi 2) with pacman,
it'll take some time, ( I have fast enough internet, but there are lots of packages).

I'll report back on the outcome ASAP.

I had an up-to-date system before boot troubles.
But, when restoring, I somehow ended up restoring backup from this old version.

p.s. Thanks for the quick response. :+1: :1st_place_medal:

nightBulb, 2023-06-08:
My bad :confused: , somewhere along the various `pacman -Syu` my system got stuck somewhere in Linux 5.x something, which eventually stopped working after reboot.

I accidentally used 'Archlinux-arm-latest' from 2019, instead of actual 'latest' :smiley_cat: 
and late night brain did the rest.

A fresh install seems to be working perfectly fine.

Thanks for all your efforts. Kudos. :+1: :+1: :+1:

closed 2023-06-08 by nightBulb