raspberrypi/linux #5496: Missing /boot/config-[kernel_version], prevents initramfs generation

SylvainGarrigues, 2023-06-09:
### Describe the bug

The initramfs-tools package in Debian bookworm includes the mkinitramfs scripts which checks if the kernel supports the compression method used to compress the initramfs, by `grep`ping the `/boot/config-[new_kernel_version]` file which the `raspberrypi-kernel` package does not install.

Can the package install the configs for the provided kernel* in /boot?

### Steps to reproduce the behaviour

```
update-initramfs -c -v -k 6.1.21-v8+
```

### Device (s)

Raspberry Pi 3 Mod. B

### System

N/A

### Logs

/usr/sbin/mkinitramfs includes a line like `grep /boot/config-VERSION` and this file don't exist.

alsldlflglhljlk, 2023-07-16:
You can download the config file from here: [https://pastebin.com/raw/AVmn0AgF](https://pastebin.com/raw/AVmn0AgF)

But running `sudo touch /boot/config-6.1.21-v8+` should work as well.

SylvainGarrigues, 2023-07-17:
My workaround is actually:
```
modprobe configs && zcat /proc/config.gz > /boot/config-6.1.21-v8+
```

But this doesn't solve the problem that when the Raspberry Pi foundation kernel package will be upgraded, the `/boot/config-[kernel_version]` file won't be found, and the `update-initramfs` script will fail, since the `initramfs-tools` package in Debian bookworm script expects it (I guess kernel packages provided by Debian all provide the config file in the `boot` partition). 

So I guess either the Raspberry Pi repository needs to provided a patched `initramfs-tools` package, or the kernel image package needs to install the kernel config.

alsldlflglhljlk, 2023-07-17:
I haven't seen this error in a while, so can you post the exact error message you get?

SylvainGarrigues, 2023-07-17:
There was no much log except what I already mentioned in the initial report: the post install script of the kernel package fails, I ran `update-initramfs` like the script does, it also failed with a generic error, so I added `set -e -x`
to `/usr/sbin/mkinitramfs` to discover the culprit, which was like: `grep: /boot/config-version: file not found` due to a failing `grep` line in that script (because of the missing file): it does need the config file to know what kind of initrd compression is supported by the kernel, so as to include the relevant modules in the ramfs. 

I guess this happen only if you have configured a ramfs image generation.

alsldlflglhljlk, 2023-07-17:
Run `sudo nano /usr/sbin/mkinitramfs`, and add `sudo modprobe configs && zcat /proc/config.gz > /boot/config-$(uname -r)` to the second line. That should do the trick.

SylvainGarrigues, 2023-07-17:
Sure, but this is a trick / hack just for my machine. I am not looking for help (I fixed this problem weeks ago), I just want to report the issue: anyone who needs an initramfs in bookworm will have this issue, so either the kernel package needs to include the config file, or the initramfs-tools package needs to be patched with what you mentioned. 

Is this the right place to report this issue?

alsldlflglhljlk, 2023-07-17:
[https://github.com/RPi-Distro/pi-gen](https://github.com/RPi-Distro/pi-gen) is probably the right place to report this. Please close this issue.

poizan42, 2023-08-10:
FWIW here is the commands I used to extract the configs for all installed kernel versions:
```shell
wget https://raw.githubusercontent.com/torvalds/linux/master/scripts/extract-ikconfig
for f in /usr/lib/modules/*; do
    v=${f#/usr/lib/modules/}
    sh ./extract-ikconfig "/usr/lib/modules/$v/kernel/kernel/configs.ko.xz" | sudo tee /boot/config-$v > /dev/null
done
```

closed 2023-08-10 by popcornmix