raspberrypi/linux #5497: CM4 after reboot NIC (eth0) disappears / no ethernet connection

JN19aban, 2023-06-11:
### Describe the bug

Hi,

I have a CM4 with EMMC and sometimes after reboot the NIC connection disappears. To get it work again either re-plug the power or doing sudo reboot. 

### Steps to reproduce the behaviour

Just reboot some times.

### Device (s)

Raspberry Pi CM4

### System

25e2319effa91eb95edd9d9209eb9f8a584d67be

version d7f9c2b4ef7e4a8c0b04374a879ce89d7a948453 (clean) (release) (start)

6.1.31-v8+

### Additional context

[image]

pelwell, 2023-06-11:
Please provide the output of the `dmesg` command, ideally two separate runs - once after a clean boot when it works, and once after it has failed to enable the Ethernet. Not photos, please - you can attach text files in a comment.

cleverca22, 2023-06-11:
also run `ifconfig -a`, or `ip addr`

by default, `ifconfig` only shows devices that are up

JN19aban, 2023-06-12:
Hi,

sorry for the screenshot, I was on remote location.

The `dmsesg` with the problem:

`[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.32-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1656 SMP PREEMPT Wed Jun  7 11:37:17 BST 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s78504 r8192 d32088 u118784
[    0.000000] pcpu-alloc: s78504 r8192 d32088 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:FF:E3:D3 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=ac690725-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 861592K/970752K available (11776K kernel code, 2106K rwdata, 3692K rodata, 4160K init, 1077K bss, 43624K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 39776 entries in 156 pages
[    0.000000] ftrace: allocated 156 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000324] Console: colour dummy device 80x25
[    0.000878] printk: console [tty1] enabled
[    0.000942] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000984] pid_max: default: 32768 minimum: 301
[    0.001130] LSM: Security Framework initializing
[    0.001327] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001374] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002670] cgroup: Disabling memory control group subsystem
[    0.004991] cblist_init_generic: Setting adjustable number of callback queues.
[    0.005028] cblist_init_generic: Setting shift to 2 and lim to 1.
[... 741 lines cut ...]
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 10  bytes 1552 (1.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 10  bytes 1552 (1.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0`

The `ip addr` without having this issue:

`1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:ff:e3:d3 brd ff:ff:ff:ff:ff:ff
    inet 192.168.88.185/24 brd 192.168.88.255 scope global dynamic noprefixroute eth0
       valid_lft 310sec preferred_lft 235sec
    inet6 fe80::a6fe:c21d:f2e9:afd5/64 scope link 
       valid_lft forever preferred_lft forever`

Regards

pelwell, 2023-06-12:
1. What kind of display is attached?

2. Apart from the first log being from a system with no Ethernet connectivity and the second being from when it works, were there any other expected differences? I ask because the first log shows it detecting a display, while the second one does not detect a display.

3. How is the CM4/CM4IO combination being powered? One could imagine that with a marginal power supply sometimes it affects one subsystem (Ethernet PHY) and sometimes another (display)?

JN19aban, 2023-06-12:
Hi,

I attach an Eizo monitor for the purpose of the test. The system run headless. Correct on the first log file I have connect the monitor so I can take the log files because of no SSH possible. On the second log I have disconnected the monitor.

Nothing more attached to the system except the monitor and one keyboard.

It is powered on by a PSU 12v 5amp. Also I have tried other with the same result.

This system is only running headless. Connected to it power and Ethernet, nothing more.

I have tried different switches (Cisco SG350, SG300,3750x) Extreme networks (48 SI) and different cables, even I have put everything on online UPS for clean power and the same it does on all occasions. 

Also to say something more. Because this CM4 I have for a long time only worked correct with 5.10.xx kernel. After the upgrade to 5.15.xx and 6.1.xx this symptom is going more and more worse.

With this one I have total 13 CM4 all identical bought on the same day and all running headless. All 12 are still running 5.10.xxx kernel with no issues, nothing, and I am afraid to upgrade due to the current issues. The last CM4 is for testing and have this issue. It is not the power supply on this. Something else maybe.

pelwell, 2023-06-13:
1. What is the approximate frequency of the network failure? I have a CM4 rebooting happily with no problems bringing up eth0.

2. What does `vcgencmd bootloader_version` report?

pelwell, 2023-06-14:
@dp111 Do you have any ideas what could prevent the PWR LED from switching off during a reboot? Is it purely under programmatic control of the GPIO expander?

JN19aban, 2023-06-19:
Then if you have test it for 24 hours of reboots, then I might have a defective CM4 probably. From the rest 12 CM4's I will take one and clone as the "defective" one and I hope that the issue stay on the "defective" one. I will respond back.

Regards.

JN19aban, 2023-07-06:
Hi,

Sorry for the late response, my CM4 are on remote locations.

So from the 13 in total I own, the three I have tested with 6.1.32 -.34 -.36 kernel have the same issue. On reboot sometimes the eth0 is not working. 

All are with default installation method via Raspberry pi imager. The only thing I have enabled is the serial. But without enabling the serial it does the same thing. 

Is there anything else I can check for log files?

Best regards.

JN19aban, 2023-07-22:
Hi,

I have yesterday install the 5.10.y kernel on the three CM4 that have the ethernet issue and works completely fine. 

So there must be an issue with old revisions of CM4 with the PHY and new kernels. I hope that somehow can be resolved this. 

Regards.

efriel94, 2023-09-13:
Hi @JN19aban, how did you build the 5.10 kernel as I have the same issue? Did you clone it and build it directly on the board or cross compile it?

JN19aban, 2023-12-19:
Ok to add something more because still has not been resolved as far I see. 

Except from the 13 CM4's that I have which are Revision 1.0 I have purchased extra 2 more, 1 with emmc and one lite version and both Revision 1.1.

With the emmc it still does this the same way as my 13 HW 1.0 do.

The lite version though is doing it very rarely but it does.

Tested with kernel 6.1.68.

Sadly the issue remains and the CM4 for my application is not trusted to be on remote sites. :(

I do not know what else to do, I hope the developers find what cause this.

@efriel94 I have clone it and built it locally on CM4. But it lacks the PTP support..........

Regards.

ddivins, 2024-04-24:
I think I had the same problem.  Everything was working fine and then I started to lose eth0 on reboot.  In my case this device is using the Official IO Board with a POE Hat.  While powered from POE, I added an external Power supply.  Thats when things started to have issues.  I disabled POE at the switch and reset the CM4, and everything returned and seems to be ok.  I will keep an eye on it, but wanted to see if maybe POE was involved in this case,

anthonio9, 2024-10-10:
Any solutions to the problem? I just tried kernel 6.6.22-v8 and my `eth0` was renamed to `end0`. This is easily fixable by setting `net.ifnames = 0` in cmdline.txt, however even after getting back the `eth0` name the interface does not come up:

```sh
[    1.349191] bcmgenet fd580000.ethernet: GENET 5.0 EPHY: 0x0000
[   25.728869] could not attach to PHY
[   25.773755] bcmgenet fd580000.ethernet eth0: failed to connect to PHY
[   27.480921] could not attach to PHY
[   27.484610] bcmgenet fd580000.ethernet eth0: failed to connect to PHY
```

efriel94, 2024-10-11:
Turned out to be a defective board for me.

MERAJ-SHAIKH-UBIQ, 2025-10-03:
If my custom board had an issue, then why would some CM4s work well with Ethernet while others do not? The behavior should ideally be consistent across all CM4s. I’m starting to think this might be related to the kernel version rather than the hardware.

Here's the log for the reference
[    8.496657] Bluetooth: hci0: BCM: features 0x2f
[    8.498098] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    8.498126] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    9.015378] could not attach to PHY
[    9.015400] bcmgenet fd580000.ethernet eth0: failed to connect to PHY
[    9.165499] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    9.251174] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.251191] Bluetooth: BNEP filters: protocol multicast
[    9.251204] Bluetooth: BNEP socket layer initialized
[    9.255008] Bluetooth: MGMT ver 1.22
[    9.263608] NET: Registered PF_ALG protocol family
[   62.992758] systemd[679]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL se

MERAJ-SHAIKH-UBIQ, 2025-12-05:
Thank you for the clarification — yes, the CM4-IO Wireless Base B is indeed a Waveshare board.

To answer your question about what the board is connected to:

At the time of testing, the board has *nothing* connected except:
- power input
- an Ethernet cable connected directly to my computer, with static IP addresses configured on both sides
- a SIM7600G-H module connected via an M.2 adapter

No additional peripherals, HATs, USB devices, or external wiring.  
The issue still occurs on all CM4 modules I tested.

The eth0 interface continues to cycle UP for ~3–4 minutes and then DOWN for ~1–2 minutes repeatedly, even in this minimal setup.

I will try adding:

    dtparam=eee=off

to `/boot/firmware/config.txt` and report back with the results.

jclark, 2026-09-15:
I encountered this initially on one board. I tried to reproduce it on two other boards and was able to on both. All three are CM4 Rev 1.1 on the official IO board, warm reboots with the Ethernet cable attached:

- rpi-7.0.y `7.0.14-v8+` (Raspberry Pi OS): 34 reboots, failed on boots 10, 27 and 33.
- mainline 7.3-rc2 (Fedora 44, U-Boot + GRUB): first series of 16 boots alternating with stock 7.1.13, failed on the 1st and 2nd 7.3 boots; second series of 54 boots, failed on boots 22, 46, 47, 49 and 53.
- mainline 7.3-rc2 (Ubuntu 26.04): one failure seen, not looped.

Never seen on a cold boot. The kernel signature is always the same, and the MDIO bus statistics after a failed boot are `transfers=2 errors=2`: the BMSR read from `unimac_mdio_reset()` and the PHYSID1 read, both failed, nothing else.

I instrumented `mdio-bcm-unimac` and caught one failing boot:

- Every read of PHY 0 completes with `MDIO_READ_FAIL` set, i.e. no response on the bus. The instrumented driver added re-reads of BMSR/PHYSID1/PHYSID2 at 10 ms, 100 ms, 1 s and 5 s after the first failure and they all fail the same way.
- `MDIO_CFG` and the GENET SYS/EXT registers are identical to a good boot.
- When the driver loaded, `MDIO_CMD` still held `0x1803ffff`: a read of PHY 0 register 3 with `READ_FAIL` set, left by U-Boot about 15 s earlier. So the PHY was already unresponsive at the bootloader stage, and `EXT_RGMII_OOB_CTRL` was `0x00f00000` (RGMII mode never enabled) versus `0x00f10050` on a good boot.

I (and my coding agent) are out of ideas to make progress on this, but I am happy to perform any additional investigation that would be helpful to you.

mkovero, 2026-09-16:
Same symptom here on a CM4 Rev 1.0 (`b03140`) on the official IO board, rpi-7.2.y. Two things that may help.

**On a failing boot the PHY answers at MDIO address 2 instead of 0.** I added a scan to `mdio-bcm-unimac` that reads PHYSID at all 32 addresses when address 0 fails. Every failing warm reboot (normal `reboot` and `sysrq-b` alike) gave:

```
mdio_bus unimac-mdio--19: MDIO-DBG scan: addr  2 answers PHYSID=0x600d84a2
mdio_bus unimac-mdio--19: MDIO-DBG scan: 1 of 32 addresses answered
mdio_bus unimac-mdio--19: MDIO device at address 0 is missing.
```

That fits the stale U-Boot read reported above: the PHY has already moved before Linux starts. A quick patch that redirects address-0 access to the address the scan finds gives a fully working link (including PTP) on such a boot, so the PHY itself is fine.

**Turning off LED3 (`ETH_LEDY`) stopped it here.** In `config.txt`:

```
dtparam=eth_led1=4
```

(LED3 multicolor mode OFF, so the yellow Ethernet LED stays dark.) Warm reboots with the cable attached went from 0 of 12 to 6 of 6, one of them `sysrq-b`.

My guess is that the LED3 pin doubles as an address strap and gets latched while the LED is active, which would also explain why unplugging the cable helps. I can't confirm the pin mapping without the datasheet. The RPi kernel drives LED3 as a link LED; mainline doesn't write that register at all, so if mainline still fails, LED3's reset default may already be active. Someone with the datasheet or the CM4 schematic would know.

`eth_led1` is a Raspberry Pi DT parameter, so it only helps on RPi kernels/DTs.

dp111, 2026-09-16:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

The LEDs pins do double as address pins during reset of the phy.