raspberrypi/linux #5500: Raspberry Pi 3B freezes with connected USB Microphone / Speaker

pipsen, 2023-06-12:
### Describe the bug

Hi everyone,

I'm slowly getting desperate, and I hope you can help me. My setup is a Raspberry Pi 3B with a Jabra Speak 510. I have Rhasspy installed on the Pi, which I want to use for voice control. Everything works fine for a few hours, but at some point, the Pi becomes unresponsive. Ping still works, but that's it. No web interface, no SSH.

I've tested it with both Buster and Bullseye, with both 32-bit and 64-bit versions. I've tried Rhasspy natively and also in a Docker container, but I get the same result. A look at the dmesg log shows the following messages:

Transfer to device x endpoint 0x1 frame xxxx failed - FIQ reported NYET. Data may have been lost.

There are already a few threads about this, all related to USB audio devices:

- https://github.com/raspberrypi/linux/issues/624
- https://forums.raspberrypi.com/viewtopic.php?t=133499
- https://wiki.linuxfoundation.o…ntation/known_limitations
- https://www.osadl.org/Single-View.111+M5c03315dc57.0.html

Various things are recommended regarding USB FIQ. Here's what I've tried and the respective results:

dwc_otg.fiq_fsm_enable=0

Raspberry Pi boots, but the microphone doesn't work at all.

dwc_otg.fiq_fsm_mask=0x5

Raspberry Pi becomes completely unreachable right after boot.

dwc_otg.fiq_fsm_mask=0xf

Everything looks normal, and the dmesg error messages no longer appear, but the freeze still occurs after about 6 hours.

dwc_otg.fiq_fsm_mask=0xd

The test has been running for half an hour now, but I'm already getting lots of "retire_capture_urb: xxxx callbacks suppressed" messages. Doesn't look promising.

Do you have any ideas on what I could do? I'm running out of ideas slowly :(

Thanks for your help!

### Steps to reproduce the behaviour

Install Rhasspy on a Raspberry PI 3B and connect a Jabra Speak 510. Wait for around 6 hours

### Device (s)

Raspberry Pi 3 Mod. B

### System

Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage2

version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

Linux PI 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

### Logs

Either:
Transfer to device x endpoint 0x1 frame xxxx failed - FIQ reported NYET. Data may have been lost.

Or:
retire_capture_urb: xxxx callbacks suppressed

pipsen, 2023-07-10:
My fix was to use the old lagacy OS Linux PI 5.10.103-v7+
=> No problems

P33M, 2023-07-10:
> I'm seeing the same issue using a Plantronics Poly P3200 USB microphone/speaker system. I don't need to wait six hours before the problem happens. I see it after using the speaker once. Also using a Raspberry Pi 3B. For the past month I have been tinkering with OVOS (open voice operating system) and I first noticed the issue today (July 9). I am using 64 bit lite Raspberry Pi OS.
> 
> Transfer to device 4 endpoint 0x3 frame 951 failed - FIQ reported NYET. Data may have been lost.

1) where can I buy one of these microphones, and
2) what do I need to type into a terminal on a fresh install of Raspberry Pi OS to replicate your setup?

Bulwagga, 2023-07-10:
I got the error message to go away by using a slightly larger power supply. 

Microphone (looks like a cheaper model just came out - 5200):
https://www.amazon.com/Plantronics-Calisto-3200-USB-A/dp/B07L9WGRY8/ref=sr_1_1?crid=248ZUZ1EHPHIC&keywords=plantronics+p3200&qid=1689014657&sprefix=plantronics+p3200%2Caps%2C117&sr=8-1

I don't think you need to do anything special beyond installing pulseaudio.
sudo apt install -y portaudio19-dev libpulse-dev pulseaudio-utils pulseaudio
pulseaudio -D

Here are the instructions I was using:
https://openvoiceos.github.io/community-docs/install_raspbian/

The images would be your best bet to install OVOS, the do-it-yourself option _almost_ works. :)  That's what I'm tinkering with.

P33M, 2023-07-27:
> I got the error message to go away by using a slightly larger power supply.

Well there's not much I can do in software for a device that rejects VBus by ignoring host traffic instead of disconnecting cleanly...