raspberrypi/linux #5501: sc16is7xx driver not reading initial modem control signals

mtnbarn, 2023-06-12:
### Describe the bug

The sc16is7xx driver assumes that the modem control signals are in a off state when the driver starts. If a signal is in the on state when the driver is started, it will not be read properly until another line change occurs.

Issue applicable to all PI models

### Steps to reproduce the behaviour

Put CTS into a mark condition before powering up the System, after Linux starts, read line status from the driver, it will not reflect the actual status of the CTS line

### Device (s)

Raspberry Pi 2 Mod. B v1.2, Raspberry Pi 3 Mod. B

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 638c7521ee0c431fafca1e2bd4fd25705b37ea5b, stage2

$ vcgencmd version
Mar 17 2023 10:52:42
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

$ uname -a
Linux raspberrypi 6.1.29-v7+ #2 SMP Sat May 27 10:04:21 PDT 2023 armv7l GNU/Linux

### Additional context

Patch to fix issue

--- drivers/tty/serial/sc16is7xx.c.orig	2023-05-25 11:18:26.493202230 -0700
+++ drivers/tty/serial/sc16is7xx.c	2023-06-12 15:34:26.242438742 -0700
@@ -1212,6 +1212,9 @@ static int sc16is7xx_startup(struct uart
 	      SC16IS7XX_IER_MSI_BIT;
 	sc16is7xx_port_write(port, SC16IS7XX_IER_REG, val);
 
+	/* Initialize the Modem Control signals to current status */
+	one->old_mctrl = sc16is7xx_get_hwmctrl(port);
+	
 	/* Enable modem status polling */
 	spin_lock_irqsave(&port->lock, flags);
 	sc16is7xx_enable_ms(port);

pelwell, 2023-06-13:
This kind of patch should really be sent upstream - it's not Pi-specific - and it looks uncontentious so it should be readily accepted. However, we do have one downstream patch that we haven't upstreamed yet, and I'm feeling generous.

If you give me a `Signed-off-by: My Name <my-email-address>` line I'll attribute the patch to you, otherwise I'll just mention your GitHub handle in the commit message.

mtnbarn, 2023-06-13:
This patch sets the driver modem line state to the actual line state at driver startup.
Signed-off-by schmidt.earl.f@gmail.com
--- drivers/tty/serial/sc16is7xx.c.orig	2023-05-25 11:18:26.493202230 -0700
+++ drivers/tty/serial/sc16is7xx.c	2023-06-12 15:34:26.242438742 -0700
@@ -1212,6 +1212,9 @@ static int sc16is7xx_startup(struct uart
 	      SC16IS7XX_IER_MSI_BIT;
 	sc16is7xx_port_write(port, SC16IS7XX_IER_REG, val);
 
+	/* Initialize the Modem Control signals to current status */
+	one->old_mctrl = sc16is7xx_get_hwmctrl(port);
+	
 	/* Enable modem status polling */
 	spin_lock_irqsave(&port->lock, flags);
 	sc16is7xx_enable_ms(port);

pelwell, 2023-06-13:
Can I change that to:
```

```
?

mtnbarn, 2023-06-13:
Yes please and thank you

closed 2023-06-14 by mtnbarn