raspberrypi/linux #5503: Diagnosing unreliable reboots

pelwell, 2023-06-13:
### Describe the bug

This is a continuation of https://github.com/raspberrypi/firmware/issues/1559, a thread which is so long that the important diagnostic questions have got lost in the middle.

https://github.com/raspberrypi/firmware/issues/1559 was specifically about Pi 4, but the methods for diagnosing the problem are common to all Pis.

I'll be using the word "reboot" throughout this list, but it should apply equally to powering off.

1. When rebooting hangs, wait several minutes to see if it unsticks eventually.
2. Does `sync && sudo reboot -f` hang or reboot?
3. If reboot is very slow but eventually completes,
    1. Add `systemd.log_level=debug systemd.log_target=kmsg log_buf_len=1M enforcing=0` to /boot/cmdline.txt
    2. Put the following in `/usr/lib/systemd/system-shutdown/debug.sh`:
        ```
        #!/bin/sh
        mount -o remount,rw /
        dmesg > /shutdown-log.txt
        mount -o remount,ro /
        reboot
        ```
    3. Make it executable:
        ```
        $ sudo chmod +x /usr/lib/systemd/system-shutdown/debug.sh
        ```
    4. Reboot twice - once to activate the logging, and once to generate /shutdown-log.txt.
    5. Upload /shutdown-log.txt.
    6. Remove the additions to cmdline.txt and delete (or make non-executable) /usr/lib/systemd/system-shutdown/debug.sh
4. If reboot doesn't complete, and if you have a serial/UART cable you can connect to header pins 6, 8, and 10, and a host computer (or another Raspberry Pi) to run a terminal app such as PuTTY or minicom:
    1. Add `enable_uart=1` to /boot/config.txt.
    2. Add `systemd.log_level=debug systemd.log_target=kmsg log_buf_len=1M enforcing=0 ignore_loglevel` to /boot/cmdline.txt, being careful to remove `quiet` if it is present.
    3. Run `sync` to flush outstanding writes, then power cycle.
    4. Log in as usual (it will take a long time to boot, with thousands of lines of output, and if you are logging in over the serial connection you may have to hit Return to get the prompt), then `sudo reboot` and capture the output in your terminal program.

### Steps to reproduce the behaviour

x

### Device (s)

Other

### System

x

beren12, 2023-06-13:
Just to clarify these appears to be a systemd-related bug as I have never had a failed reboot with sysvinit on any of my rpi models. Yes on current releases sysvinit still works unfailingly.

pelwell, 2023-06-13:
You're in danger of confusing apples with oranges. Further sysvinit discussion will be deleted as off-topic.

tianfeng-yang, 2023-06-14:
@pelwell 
My Raspberry Pi 4B has a reboot hang problem, the following is the log information output according to step 4

```
Linux rasp4yzm 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact[  129.491710] systemd[1]: n/a: New incoming connection.
 distribution terms for [  129.499016] systemd[1]: n/a: Connections of user 0: 0 (of 1024 max)
each program are describ[  129.507279] systemd[1]: varlink-12: varlink: setting state idle-server
ed in the
individual fi[  129.516289] systemd[1]: varlink-12: New incoming message: {"method":"io.systemd.UserDatabase.GetUserRecord","parameters":{"userName":"pi","service":"io.systemd.DynamicUser"}}
les in /usr/share/doc/*/[  129.533592] systemd[1]: varlink-12: varlink: changing state idle-server → processing-method
copyright.

Debian GNU[  129.544147] systemd[1]: varlink-12: Sending message: {"error":"io.systemd.UserDatabase.NoRecordFound","parameters":{}}
/Linux comes with ABSOLU[  129.556844] systemd[1]: varlink-12: varlink: changing state processing-method → processed-method
TELY NO WARRANTY, to the[  129.567870] systemd[1]: varlink-12: varlink: changing state processed-method → idle-server
 extent
permitted by ap[  129.578569] systemd[1]: varlink-12: Got POLLHUP from socket.
plicable law.
Last login: Wed J[  129.587294] systemd[1]: varlink-12: varlink: changing state idle-server → pending-disconnect
un 14 08:18:07 CST 2023 [  129.597825] systemd[1]: varlink-12: varlink: changing state pending-disconnect → processing-disconnect
on ttyAMA0
[  129.609048] systemd[1]: varlink-12: varlink: changing state processing-disconnect → disconnected
[  129.619120] systemd[1]: n/a: New incoming connection.
[  129.624252] systemd[1]: n/a: Connections of user 0: 0 (of 1024 max)
[  129.630546] systemd[1]: varlink-12: varlink: setting state idle-server
[  129.637168] systemd[1]: varlink-12: New incoming message: {"method":"io.systemd.UserDatabase.GetUserRecord","parameters":{"uid":1000,"service":"io.systemd.DynamicUser"}}
[  129.652367] systemd[1]: varlink-12: varlink: changing state idle-server → processing-method
[  129.660945] systemd[1]: varlink-12: Sending message: {"error":"io.systemd.UserDatabase.NoRecordFound","parameters":{}}
[  129.671660] systemd[1]: varlink-12: varlink: changing state processing-method → processed-method
[  129.680636] systemd[1]: varlink-12: varlink: changing state processed-method → idle-server
[  129.689288] systemd[1]: varlink-12: Got POLLHUP from socket.
[  129.695056] systemd[1]: varlink-12: varlink: changing state idle-server → pending-disconnect
[  129.703722] systemd[1]: varlink-12: varlink: changing state pending-disconnect → processing-disconnect
[  129.713226] systemd[1]: varlink-12: varlink: changing state processing-disconnect → disconnected
[  129.722331] systemd[1]: Got message type=method_call sender=:1.5 destination=org.freedesktop.systemd1 path=/org/freedesktop/systemd1 interface=org.freedesktop.systemd1.Manager member=StartUnit cookie=15 reply_cookie=0 signature=ss error-name=n/a error-message=n/a
[  129.751276] systemd[1]: Sent message type=method_call sender=n/a destination=org.freedesktop.DBus path=/org/freedesktop/DBus interface=org.freedesktop.DBus member=GetConnectionUnixUser cookie=195 reply_cookie=0 signature=s error-name=n/a error-message=n/a
[  129.773959] systemd[1]: Got message type=method_return sender=org.freedesktop.DBus destination=:1.2 path=n/a interface=n/a member=n/a cookie=23 reply_cookie=195 signature=u error-name=n/a error-message=n/a
[  129.792322] systemd[1]: user@1000.service: Trying to enqueue job user@1000.service/start/replace
[  129.801180] systemd[1]: Added job user@1000.service/start to transaction.
[  129.808039] systemd[1]: Pulling in sysinit.target/start from user@1000.service/start
[  129.815813] systemd[1]: Added job sysinit.target/start to transaction.
[  129.822374] systemd[1]: Pulling in dev-hugepages.mount/start from sysinit.target/start
[  129.830319] systemd[1]: Added job dev-hugepages.mount/start to transaction.
[  129.837298] systemd[1]: Pulling in -.slice/start from dev-hugepages.mount/start
[  129.844626] systemd[1]: Added job -.slice/start to transaction.
[  129.850562] systemd[1]: Pulling in -.mount/start from dev-hugepages.mount/start
[  129.857890] systemd[1]: Added job -.mount/start to transaction.
[  129.863825] systemd[1]: Pulling in -.slice/start from -.mount/start
[  129.870110] systemd[1]: Pulling in proc-sys-fs-binfmt_misc.automount/start from sysinit.target/start
[  129.879261] systemd[1]: Added job proc-sys-fs-binfmt_misc.automount/start to transaction.
[  129.887456] systemd[1]: Pulling in -.mount/start from proc-sys-fs-binfmt_misc.automount/start
[  129.896000] systemd[1]: Pulling in systemd-ask-password-console.path/start from sysinit.target/start
[  129.905164] systemd[1]: Added job systemd-ask-password-console.path/start to transaction.
[  129.913359] systemd[1]: Pulling in -.mount/start from systemd-ask-password-console.path/start
[  129.921903] systemd[1]: Pulling in shutdown.target/stop from systemd-ask-password-console.path/start
[  129.931059] systemd[1]: Added job shutdown.target/stop to transaction.
[  129.937618] systemd[1]: Pulling in emergency.service/stop from systemd-ask-password-console.path/start
[  129.946946] systemd[1]: Added job emergency.service/stop to transaction.
[... cut ...]
[  155.251120] systemd-shutdown[1]: Successfully forked off '(sd-executor)' as PID 988.
[  155.266815] systemd-shutdown[1]: (sd-executor) succeeded.
[  155.273614] systemd-shutdown[1]: Successfully forked off '(sd-sync)' as PID 989.
[  155.281250] systemd-shutdown[1]: Syncing filesystems and block devices.
[  155.288131] systemd-shutdown[1]: Rebooting.
[  155.293791] kvm: exiting hardware virtualization
[  155.366983] reboot: Restarting system
```
After waiting for more than 10 minutes, there has been no response

ragazenta, 2023-10-24:
Hi, after upgrading bootloader to this version and upgrading firmware via `sudo SKIP_KERNEL=1 rpi-update` seems to fix reboot hang problem. Does this version of bootloader really solve the problem? https://github.com/raspberrypi/rpi-eeprom/commit/abce28f9bc8a1f817ff57b709abc2b2d53b78ea4

If I only upgrade the bootloader and not the firmware, reboot still hang sometimes.

```
MESS:00:00:08.034750:0: Read command line from file 'cmdline.txt':
MESS:00:00:08.040611:0: 'console=serial0,115200 console=tty1 root=PARTUUID=11b31d2a-02 rootfstype=ext4 fsck.repair=yes rootwait loglevel=3 quiet logo.nologo'
MESS:00:00:08.372388:0: brfs: File read: 132 bytes
MESS:00:00:20.884023:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:20.889289:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:20.997229:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:21.048287:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:21.067257:0: Loading 'kernel7l.img' to 0x8000 size 0x0
MESS:00:00:21.070300:0: Device tree loaded to 0x2eff3500 (size 0xcaa4)
MESS:00:00:21.079909:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:21.086244:0: uart: Baud rate change done...
MESS:00:00:21.088309:0: uart: Baud rate change done...

<!-- hang -->
```

```
MESS:00:00:08.034018:0: Read command line from file 'cmdline.txt':
MESS:00:00:08.039879:0: 'console=serial0,115200 console=tty1 root=PARTUUID=11b31d2a-02 rootfstype=ext4 fsck.repair=yes rootwait loglevel=3 quiet logo.nologo'
MESS:00:00:08.371587:0: brfs: File read: 132 bytes
MESS:00:00:12.358314:0: brfs: File read: /mfs/sd/kernel7l.img
MESS:00:00:12.361005:0: Loading 'kernel7l.img' to 0x8000 size 0x677610
MESS:00:00:12.367269:0: Device tree loaded to 0x2eff3500 (size 0xcaa4)
MESS:00:00:12.376934:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:12.383269:0: uart: Baud rate change done...
MESS:00:00:12.385330:0: uart: Baud rate change done...
MESS:00:00:12.393865:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined

<!-- hang -->
```

tianfeng-yang, 2023-11-22:
> Hi, after upgrading bootloader to this version and upgrading firmware via `sudo SKIP_KERNEL=1 rpi-update` seems to fix reboot hang problem.

In my case, this was not resolved

TestODTestOD, 2023-12-18:
Hi, I found a reliable solution enabling at boot time the hardware Watchdog of the Broadcom.
Logging with a serial cable the reboot process of the Raspberry Pi OS (Buster), I saw it hangs after all services are normally stopped.
Most of time, reboot is OK after this log, but, randomly (very few), the OS stops at this step just like after a `halt` command.
> [  291.338847] reboot: Restarting system

Enabling the hardware watchdog like described by the link below solves the problem:
[https://diode.io/blog/running-forever-with-the-raspberry-pi-hardware-watchdog](url)

I made hundreds of automatic reboots during multiple days (x288 / day) with and without watchdog enabled this way.
I always used exactly the same hardware and software.

- With the watchdog enabled, I never get any trouble: the raspberry always boots after reboot command.
- Without watchdog (`#dtparam=watchdog=on` in /boot/config.txt ), the systems always hangs after a few (or a few tens) of reboot command.

VoidVolker, 2023-12-18:
@TestODTestOD hi! Sounds great, thx! Will test your solution. Btw, you got a typo in url - it has a wrong target link.

epigramx, 2024-04-04:
Enabling watchdog does not solve it here. Updating to latest full-upgrade doesn't, and upgrading to latest rpi-update doesn't either.

epigramx, 2024-04-28:
My issue is solved and it was unrelated to the kernel or the even the default installation of the OS entirely. I had a booting hanging the networking shutdown.
I encourage people to use `systemd.log_level=debug` at the command line at `/boot/firmware/cmdline.txt` and look for "held back" messages.