raspberrypi/linux #5505: Allo Boss 2 device tree overlay or driver broken

MichaIng, 2023-06-14:
### Describe the bug

We have several reports across multiple forums that the Allo Boss 2 dtb is not working (well) after one of the recent kernel upgrades: https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/overlays/allo-boss2-dac-audio-overlay.dts

- https://audiophilestyle.com/forums/topic/67938-no-sound-output-from-allo-boss-2/
- https://dietpi.com/forum/t/allo-boss-2-does-not-output-sound-after-update-to-dietpi-8-15-8-16/16594
- https://community.roonlabs.com/t/update-to-8-18-2-completley-broke-my-system/244575
- https://forums.raspberrypi.com/viewtopic.php?t=350656

I am not sure who is responsible, Allo, RPi kernel repo, RPi firmware repo, no one? I had a look into it but could not see any obvious error, like false node or such in the dtb. Since it affects quite a lot of users for 2 months without any other solution than downgrading the kernel, I thought it cannot hurt to report it here and hope for the best 🙂.

### Steps to reproduce the behaviour

Use Allo Boss 2 HAT, enable related overlay, play audio, which either plays nothing or highly distorted, depending on audio source and format.

### Device (s)

Raspberry Pi 4 Mod. B

### System

It affects RPi OS, DietPi, moOde, i.e. all distros which use these kernel sources v6.1.
```
Linux AlloPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Mar 17 2023 10:53:00
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start_cd)
```

### Logs

`aplay -l` instead, just to show that the device is detected:
```
**** List of PLAYBACK Hardware Devices ****
card 0: Boss2 [Allo Boss2], device 0: bcm2835-i2s-allo-cs43130 allo-cs43130-0 [bcm2835-i2s-allo-cs43130 allo-cs43130-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

pelwell, 2023-06-14:
I don't have one of these devices, and no other sound cards appear to use the cs43130 codec, so debugging will be slow.

Any thoughts, @babuenir?

MichaIng, 2023-06-14:
Me neither. But if someone has an idea how to debug/which output/logs to check, I will try to get everyone here from the other forums and/or forward/collect infos.

pelwell, 2023-06-14:
Let's assume for now that the main codec driver is OK, since it is an unmodified upstream file. The overlay is extremely simple and hasn't changed since 5.15, so we can ignore that, which leaves the soundcard driver -  `sound/soc/bcm/allo-boss2-dac.c`.
We can start to enable tracing in that driver, but before that we can capture the i2c traffic in the two kernels and compare them.

1. Clear the kernel message log:
```
$ sudo dmesg -C
```
2. Enable i2c tracing:
```
$ sudo sh -c "echo 1 >/sys/kernel/debug/tracing/events/i2c/enable"
```
3. Attempt to play a sound. You can use `speaker-test`, but turn the volume down first. On RPiOS (and perhaps others) you can also use `aplay /usr/share/sounds/alsa/Front_Center.wav`.

4. Capture the tracing output:
```
$ sudo cat /sys/kernel/debug/tracing/trace > i2c_log1.txt
```
5. If you want to run another test without rebooting, first clear the trace so far:
```
$ sudo sh -c "echo >/sys/kernel/debug/tracing/trace"
```
Run that test on both kernels, capturing the log to a different (and ideally suitably named) file each time.

If the playback doesn't appear to work, report what you observe, and the output of `dmesg`.

UvidusNox, 2023-06-17:
Good day, thank you for the help on this issue.

Firsts tests have been run on a vanilla `raspberry pi os 64 bit lite` install combined with enabling the `allo-boss2-dac-audio` overlay / driver. Apt update/upgrade has been ran in addition. 

Kernel version:
```
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

Output of the i2c traces as attached:
[i2c_log_speaker_test_kernel6_1_21.txt](https://github.com/raspberrypi/linux/files/11778491/i2c_log_speaker_test_kernel6_1_21.txt)
[i2c_log_wav_kernel6_1_21.txt](https://github.com/raspberrypi/linux/files/11778492/i2c_log_wav_kernel6_1_21.txt)

After playback there was no output of `dmesg` whatsoever.

Same thing but for the older kernel `Linux AlloPi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux` on diet-pi; gives the following when running `aplay /usr/share/sounds/alsa/Front_Center.wav`:
```
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1349: Channels count non available
```
This is probably also the reason `aplay` isn't captured in that particular log.

[i2c_log_speaker_test_kernel5_15_84.txt](https://github.com/raspberrypi/linux/files/11778486/i2c_log_speaker_test_kernel5_15_84.txt)
[i2c_log_wav_kernel5_15_84.txt](https://github.com/raspberrypi/linux/files/11778488/i2c_log_wav_kernel5_15_84.txt)

In general the output of tracing the speaker test command seems quite comparable.

Thank you for your guidance.

**this is a condensed copy of a post that i put in the dietpi forums moments ago.

MichaIng, 2023-06-17:
The channel count error is btw expected with audio source formats which are not natively supported by the sound driver/card. You can enforce 2 channels with `aplay -c 2 /path/to/some.wav` reap. `speaker-test -c 2`. And when you enable the ALSA auto-conversion/plug plugin, it is converted automatically. This can also be done in `dietpi-config`.

However this is unrelated to the Boss 2 driver issue.

moodeaudio, 2023-06-20:
I could do some tests
Boss 2 kernel 6.1.21 all indications are that playback is working but no sound output.

Btw, their Katana driver also has issue but it's a driver load fail just on a Pi 4B, other Pi's are OK.
`DMESG: allo-katana-codec 1-0030: Failed to read Chip or wrong Chip id: 0`

Here is some ALSA info while playing something via Boss 2
```
pi@moode:~ $ moodeutl --hwparams
access: MMAP_INTERLEAVED
format: S16_LE
subformat: STD
channels: 2
rate: 44100 (44100/1)
period_size: 4410
buffer_size: 22050
```
```
pi@moode:~ $ amixer
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 255
  Mono:
  Front Left: Playback 224 [88%] [-15.50dB] [on]
  Front Right: Playback 224 [88%] [-15.50dB] [on]
Simple mixer control 'PCM De-emphasis Filter',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'PCM Filter Speed',0
  Capabilities: enum
  Items: 'Fast' 'Slow'
  Item0: 'Fast'
Simple mixer control 'PCM High-pass Filter',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'PCM Nonoversample Emulate',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'PCM Phase Compensation',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'Digital',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 255
  Mono:
  Front Left: Playback 224 [88%] [-15.50dB] [on]
  Front Right: Playback 224 [88%] [-15.50dB] [on]
Simple mixer control 'HV_Enable',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
pi@moode:~ $ 
```
```
pi@moode:~ $ dmesg | grep allo-
[    0.087730] platform 3f203000.i2s: Fixed dependency cycle(s) with /soc/i2c@7e804000/allo-cs43130@30
```
```
pi@moode:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: UEFI not found.
[... 358 lines cut ...]
[    8.574740] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video21
[    8.577125] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video22
[    8.582924] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video23
[    8.582993] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[    8.583029] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
[    8.583060] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
[    8.583089] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
[    8.583458] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[    8.790132] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    8.848448] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    8.875601] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[    9.030261] brcmfmac: F1 signature read @0x18000000=0x15264345
[    9.046349] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.048943] usbcore: registered new interface driver brcmfmac
[    9.334627] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60
[   10.718052] hwmon hwmon1: Undervoltage detected!
[   11.454590] 8021q: 802.1Q VLAN Support v1.8
[   12.082548] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   12.492103] 8021q: adding VLAN 0 to HW filter on device eth0
[   12.498434] lan78xx 1-1.1.1:1.0 eth0: Link is Down
[   18.123665] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   19.588209] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   43.830310] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
pi@moode:~ $ 
```

pelwell, 2023-06-20:
I think the cs43130 problem is due to this commit: https://github.com/raspberrypi/linux/commit/e14bd35ef44606c7d55ccb3660cededd421a14a1

It switches from describing a fraction with two separate u16s linked only by name and usage to using a standard u16_fract structure. Unfortunately the author and reviewer failed to notice that the conversion effectively swapped the numerator and denominator.

https://github.com/raspberrypi/linux/pull/5510 is a potential fix - I'm happy that it fixes the numerator/denominator ordering, but there _might_ be other issues as well. Once the autobuilds have completed (in about an hour from now) you will be able to install a test kernel using `sudo rpi-update pulls/5510`.

pelwell, 2023-06-20:
It's good to go, if anyone wants to try:
```
$ sudo rpi-update pulls/5510
```
And if you don't have rpi-update:
```
$ sudo curl -L --output /usr/bin/rpi-update https://raw.githubusercontent.com/raspberrypi/rpi-update/master/rpi-update && sudo chmod +x /usr/bin/rpi-update
```

moodeaudio, 2023-06-21:
One of our users confirmed the patch works.
https://moodeaudio.org/forum/showthread.php?tid=5644&pid=47286#pid47286

pelwell, 2023-06-21:
Excellent. The patch was merged yesterday, so it will be in future kernel releases.

At some point I'll have to notify the upstream devs...

moodeaudio, 2023-06-21:
Super nice!

Boss 2 is beloved by many Pi audio enthusiasts.

pelwell, 2023-06-22:
The patch is now queued up for Linux 6.4: https://kernel.googlesource.com/pub/scm/linux/kernel/git/broonie/sound/+/a9e7c964cea4fb1541cc81a11d1b2fd135f4cf38
Because it has a Fixes: tag it should be back-ported to LTS kernels (i.e. 6.1) as a matter of course, at which point we'll drop our downstream version.

closed 2023-06-22 by pelwell