raspberrypi/linux #5511: Allo Katana driver load fail on Pi-4

moodeaudio, 2023-06-20:
### Describe the bug

The I2S driver for Allo Katana DAC fails to load during boot on Pi-4B. It loads OK on other Pi's.

Note that the Katana DAC has been out of production for several years and the vendor Allo has apparently not been actively maintaining their softwares or providing support in Forums for some time since the Pandemic.

### Steps to reproduce the behaviour

1. Attach Pi-4B to Katana DAC
2. Power up
3. Examine output from aplay -l and DMESG which shows no device present and driver load fail respectively.

### Device (s)

Raspberry Pi 4 Mod. B

### System

# NOTE: Our imgbuild repo https://github.com/moode-player/imgbuild
pi@moode:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-06-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 18aee5cfb8a41421e512aad75835f06fe93c7efb, stage3

pi@moode:~ $ vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

pi@moode:~ $ uname -a
Linux moode 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### Logs

$ Error message
pi@moode:~ $ dmesg | grep allo-
[    0.051861] platform fe203000.i2s: Fixed dependency cycle(s) with /soc/i2c@7e804000/allo-katana-codec@30
[    6.344878] allo-katana-codec 1-0030: Failed to read Chip or wrong Chip id: 0

# Full log
pi@moode:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s78504 r8192 d32088 u118784
[    0.000000] pcpu-alloc: s78504 r8192 d32088 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[... 372 lines cut ...]
[   30.365460] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled

### Additional context

Some history.

This issue was present as far back as 2019 according to an entry from our relnotes for moOde release 6.4.0 2019-11-24. This release was based on Raspbian Buster Lite 10.2 and kernel 4.19.83.

`- FIX: Pi-4B / Allo Katana driver load failure`

Our fix / workaround was to manually load the driver in /etc/rc.local using the code below.

```
# Work around for Katana driver load failure on Pi-4B
AUDIO_DEVICE=$(sqlite3 $SQLDB "select value from cfg_system where param='i2sdevice'")
PI_REVNUM_SEGMENT=$(awk '{if ($1=="Revision") print substr($3,3,3)}' /proc/cpuinfo)

if [ "$AUDIO_DEVICE" = "Allo Katana DAC" ] && [ $PI_REVNUM_SEGMENT = "311" ]; then
	rmmod snd_soc_allo_katana_codec
	rmmod snd_soc_audio_graph_card
	modprobe snd_soc_audio_graph_card
	modprobe snd_soc_allo_katana_codec
    echo `date +'%Y%m%d %H%M%S'` "rc.local ran Katana|Pi-4B driver reload" > /home/pi/katana.log
fi
```

pelwell, 2023-06-20:
Is this just a very late report of a longstanding issue or has something changed (as suggested by the phrase "Our fix / workaround _was_ ...")?

moodeaudio, 2023-06-20:
It was only by chance that I happened to have a 4B connected to a Katana recently and I noticed that with kernel 6.1 our workaround didn't resolve the issue as in the past with 5.y kernels. I was prompted to post this issue after reading the recent Allo Boss 2 issue.

Realistically, I'm prolly one of only a handful that still have a working Katana DAC and within that group prolly the only one that has fired it up with a 4B. Given that it was discontinued so long ago and vendor support is totally gone since the Pandemic it may not be worth investigating.

I'll go ahead and close the issue and if there is interest at some point it can be reopened.

pelwell, 2023-06-20:
I wasn't asking you to justify creating the issue, just trying to understand what has prompted it.

In what way has your workaround stopped working? Does this output correspond to the rmmod/modprobes?
```
[   17.886084] platform soc:sound: deferred probe pending
```

moodeaudio, 2023-06-20:
Thats a good question.

I'll run a test later today.

moodeaudio, 2023-06-20:
Summary on kernel 6.1

1. Driver load fails on Pi-4B with or without workaround, Previously on kernel 5.15 the workaround was successful with 4B.
2. Driver load success on Pi-3B without workaround. 

__#################################__
__# Pi 4B, Kernel 6.1__
__#################################__
```
# Without workaround in rc.local

pi@moode:~ $ aplay -l
aplay: device_list:274: no soundcards found...
pi@moode:~ $ 
pi@moode:~ $ amixer
amixer: Mixer attach default error: No such file or directory
pi@moode:~ $ 
pi@moode:~ $ dmesg | grep allo-
[    0.050930] platform fe203000.i2s: Fixed dependency cycle(s) with /soc/i2c@7e804000/allo-katana-codec@30
[    5.918378] allo-katana-codec 1-0030: Failed to read Chip or wrong Chip id: 0
```
```
# With workaround

pi@moode:~ $ sudo rmmod snd_soc_allo_katana_codec
pi@moode:~ $ sudo rmmod snd_soc_audio_graph_card
pi@moode:~ $ sudo modprobe snd_soc_audio_graph_card
pi@moode:~ $ sudo modprobe snd_soc_allo_katana_codec
pi@moode:~ $ aplay -l
aplay: device_list:274: no soundcards found...
pi@moode:~ $ dmesg | grep allo-
[    0.050930] platform fe203000.i2s: Fixed dependency cycle(s) with /soc/i2c@7e804000/allo-katana-codec@30
[    5.918378] allo-katana-codec 1-0030: Failed to read Chip or wrong Chip id: 0
[  417.572345] allo-katana-codec 1-0030: Failed to read Chip or wrong Chip id: 0
```
```
# Full DMESG

pi@moode:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s78504 r8192 d32088 u118784
[    0.000000] pcpu-alloc: s78504 r8192 d32088 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496944
[... 770 lines cut ...]
[    8.484866] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[    8.503399] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[    8.503504] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[    8.508993] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[    8.509080] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[    8.529606] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[    8.529691] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[    8.542551] bcm2835-codec bcm2835-codec: Device registered as /dev/video18
[    8.543006] bcm2835-codec bcm2835-codec: Loaded V4L2 image_fx
[    8.548991] bcm2835-codec bcm2835-codec: Device registered as /dev/video31
[    8.549076] bcm2835-codec bcm2835-codec: Loaded V4L2 encode_image
[    9.642889] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    9.714910] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    9.750576] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[   10.373196] 88XXau: loading out-of-tree module taints kernel.
[   10.526005] hwmon hwmon1: Undervoltage detected!
[   10.848936] usb 1-1.2: 88XXau 20:0d:b0:47:59:2f hw_info[d7]
[   10.864290] usbcore: registered new interface driver rtl88XXau
[   11.778780] 8021q: 802.1Q VLAN Support v1.8
[   13.314003] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   13.598413] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   13.609671] smsc95xx 1-1.1:1.0 eth0: Link is Down
[   19.747591] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
pi@moode:~ $ 
```

6by9, 2023-06-20:
Does the device detect on address 0x30 if you use ```sudo i2cdetect -y 1``` on the Pi4?

You may need to do
```
sudo rmmod snd_soc_allo_katana_codec
sudo rmmod snd_soc_audio_graph_card
```
first to stop it reporting as ```UU``` due to a kernel driver being loaded on that address, and potentially ```sudo modprobe i2c-dev``` to allow userspace access to the I2C devices.

Also check that the pinmuxing is correct with ```raspi-gpio get 2,3```. It should report SDA1 and SCL1 for the 2 pins.

pelwell, 2023-06-20:
> How can the reg_bits be 8 when the registers start at 0x100?

The regmap seems to set up some page-flipping mechanism, where register 0 is the page selection register.

6by9, 2023-06-20:
Configuring the overlay with no board present gives me
```
pi@bookworm32:~ $ sudo cat /sys/kernel/debug/tracing/trace | grep i2c-1
        modprobe-623     [003] .....    10.749305: i2c_write: i2c-1 #0 a=030 f=0000 l=1 [00]
        modprobe-623     [003] .....    10.749312: i2c_read: i2c-1 #1 a=030 f=0001 l=1
        modprobe-623     [003] .....    10.752260: i2c_result: i2c-1 n=2 ret=-121
```
So not apparently setting a bank register.

The easier way to enable i2c logging is to edit /boot/cmdline.txt and add ```trace_event=i2c``` to the start of the line. I2C logging is then enabled from boot, and ```sudo cat /sys/kernel/debug/tracing/trace | grep i2c-1``` will show the logging info.

moodeaudio, 2023-06-20:
Some additional debug.

__################################__
__# Pi 4B__
__################################__
```
pi@moode:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```
```
pi@moode:~ $ sudo rmmod snd_soc_allo_katana_codec
pi@moode:~ $ sudo rmmod snd_soc_audio_graph_card

pi@moode:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@moode:~ $ 
```
```
pi@moode:~ $ raspi-gpio get 2,3
GPIO 2: level=1 fsel=4 alt=0 func=SDA1 pull=UP
GPIO 3: level=1 fsel=4 alt=0 func=SCL1 pull=UP
```
```
pi@moode:~ $ sudo cat /sys/kernel/debug/tracing/trace | grep i2c-1
        modprobe-328     [000] .....     6.060281: i2c_write: i2c-1 #0 a=030 f=0000 l=1 [00]
        modprobe-328     [000] .....     6.060289: i2c_read: i2c-1 #1 a=030 f=0001 l=1
        modprobe-328     [000] .....     6.061364: i2c_reply: i2c-1 #1 a=030 f=0001 l=1 [b0]
        modprobe-328     [000] .....     6.061369: i2c_result: i2c-1 n=2 ret=2

pi@moode:~ $ sudo rmmod snd_soc_allo_katana_codec
pi@moode:~ $ sudo rmmod snd_soc_audio_graph_card
pi@moode:~ $ sudo modprobe snd_soc_audio_graph_card
pi@moode:~ $ sudo modprobe snd_soc_allo_katana_codec
pi@moode:~ $ sudo cat /sys/kernel/debug/tracing/trace | grep i2c-1
        modprobe-328     [000] .....     6.060281: i2c_write: i2c-1 #0 a=030 f=0000 l=1 [00]
        modprobe-328     [000] .....     6.060289: i2c_read: i2c-1 #1 a=030 f=0001 l=1
        modprobe-328     [000] .....     6.061364: i2c_reply: i2c-1 #1 a=030 f=0001 l=1 [b0]
        modprobe-328     [000] .....     6.061369: i2c_result: i2c-1 n=2 ret=2
        modprobe-1655    [001] .....   202.841822: i2c_write: i2c-1 #0 a=030 f=0000 l=1 [00]
        modprobe-1655    [001] .....   202.841829: i2c_read: i2c-1 #1 a=030 f=0001 l=1
        modprobe-1655    [001] .....   202.842888: i2c_reply: i2c-1 #1 a=030 f=0001 l=1 [b0]
        modprobe-1655    [001] .....   202.842892: i2c_result: i2c-1 n=2 ret=2
```
__################################__
__# Pi 3B__
__################################__
```
pi@moode:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
[... 237 lines cut ...]
       i2cdetect-1584    [000] .....   198.192740: i2c_result: i2c-1 n=1 ret=-121
       i2cdetect-1584    [000] .....   198.192863: i2c_write: i2c-1 #0 a=06d f=0000 l=0 []
       i2cdetect-1584    [000] .....   198.193105: i2c_result: i2c-1 n=1 ret=-121
       i2cdetect-1584    [000] .....   198.193236: i2c_write: i2c-1 #0 a=06e f=0000 l=0 []
       i2cdetect-1584    [000] .....   198.193476: i2c_result: i2c-1 n=1 ret=-121
       i2cdetect-1584    [000] .....   198.193687: i2c_write: i2c-1 #0 a=06f f=0000 l=0 []
       i2cdetect-1584    [000] .....   198.193931: i2c_result: i2c-1 n=1 ret=-121
       i2cdetect-1584    [000] .....   198.194205: i2c_write: i2c-1 #0 a=070 f=0000 l=0 []
       i2cdetect-1584    [000] .....   198.194455: i2c_result: i2c-1 n=1 ret=-121
       i2cdetect-1584    [000] .....   198.194659: i2c_write: i2c-1 #0 a=071 f=0000 l=0 []
       i2cdetect-1584    [000] .....   198.194898: i2c_result: i2c-1 n=1 ret=-121
       i2cdetect-1584    [000] .....   198.195042: i2c_write: i2c-1 #0 a=072 f=0000 l=0 []
       i2cdetect-1584    [000] .....   198.195373: i2c_result: i2c-1 n=1 ret=-121
       i2cdetect-1584    [000] .....   198.195499: i2c_write: i2c-1 #0 a=073 f=0000 l=0 []
       i2cdetect-1584    [000] .....   198.195729: i2c_result: i2c-1 n=1 ret=-121
       i2cdetect-1584    [000] .....   198.195848: i2c_write: i2c-1 #0 a=074 f=0000 l=0 []
       i2cdetect-1584    [000] .....   198.196187: i2c_result: i2c-1 n=1 ret=-121
       i2cdetect-1584    [000] .....   198.196320: i2c_write: i2c-1 #0 a=075 f=0000 l=0 []
       i2cdetect-1584    [000] .....   198.196556: i2c_result: i2c-1 n=1 ret=-121
       i2cdetect-1584    [000] .....   198.196674: i2c_write: i2c-1 #0 a=076 f=0000 l=0 []
       i2cdetect-1584    [000] .....   198.196908: i2c_result: i2c-1 n=1 ret=-121
       i2cdetect-1584    [000] .....   198.197038: i2c_write: i2c-1 #0 a=077 f=0000 l=0 []
       i2cdetect-1584    [000] .....   198.197275: i2c_result: i2c-1 n=1 ret=-121
pi@moode:~ $ 
```

6by9, 2023-06-20:
To read back 0xB0 instead of 0x30 is a bit weird as it's gained the top bit set.

I'd try adding ```i2c_arm_baudrate=50000``` to /boot/config.txt to slow the I2C down (default is 100000) and see if that helps.
If the Katana is trying to use I2C clock stretching then it can have issues.

moodeaudio, 2023-06-20:
That worked. Playback is fine now :-)
I can add this param to the Katana/4B configuration in moOde and we should be all set :-)
```
sudo nano /boot/config.txt
dtparam=i2c_arm_baudrate=50000

Reboot

pi@moode:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Katana [Allo Katana], device 0: bcm2835-i2s-allo-katana-codec allo-katana-codec-0 [bcm2835-i2s-allo-katana-codec allo-katana-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@moode:~ $ 
```

6by9, 2023-06-20:
It'd be nice to know exactly why it differs between Pi3 and Pi4, but that's unlikely to happen without having hardware and hooking up a logic analyser.

The better solution is to update the overlay to automatically drop the I2C baudrate so that it works for all users, not just those using moOde.

moodeaudio, 2023-06-20:
That would be ideal. Whats the process to do that?

pelwell, 2023-06-21:
1. Download this modified overlay: https://drive.google.com/file/d/15NwdKrCx7LEKoRQJTTRYS1gGWBDdt8uh/view?usp=sharing
2. Copy it to /boot/overlays
3. Remove the `dtparam=i2c_arm_baudrate=50000` and reboot.
4. Confirm the audio card is still detected.

If that works I'll push the updated overlay, which is just a minor modification:
```diff
diff --git a/arch/arm/boot/dts/overlays/allo-katana-dac-audio-overlay.dts b/arch/arm/boot/dts/overlays/allo-katana-dac-audio-overlay.dts
index b25fd681f09f3..a83fe48e2bfca 100644
--- a/arch/arm/boot/dts/overlays/allo-katana-dac-audio-overlay.dts
+++ b/arch/arm/boot/dts/overlays/allo-katana-dac-audio-overlay.dts
@@ -30,6 +30,7 @@ __overlay__ {
                        #address-cells = <1>;
                        #size-cells = <0>;
                        status = "okay";
+                       clock-frequency = <50000>;
 
                        allo-katana-codec@30 {
                                #sound-dai-cells = <0>;
```

moodeaudio, 2023-06-21:
That updated overlay worked fine. I tested on both Pi-4B and 3B and in both cases the Katana was recognized and playback was successful :-)

pelwell, 2023-06-21:
Cool. There's so little data involved that the reduced clock rate is unlikely to cause a problem, so I've pushed the updated overlay to rpi-6.1.y. Future releases will include the modified overlay.

moodeaudio, 2023-06-21:
Fantastic!
Many thanks on behalf of Katana enthusiasts!

closed 2023-06-21 by moodeaudio