raspberrypi/linux #5521: Loss of USB ports after few hours running (not responding to keyboard or mouse) on pi4, arm64 bullseye 6.1.21-v8+

mgrouch, 2023-06-26:
### Describe the bug

Loss of USB ports after few hours running (not responding to keyboard or mouse) on pi4, arm64 bullseye   6.1.21-v8+

I'm able to ssh and attaching dmesg output

I haven't had this behavior when kernel was 5.x

### Steps to reproduce the behaviour

Let pi4 run for a day. Happens after several hours. If I leave it overnight - almost every day.
Seems happening with usb3 ports. 
I haven't had this behavior when kernel was 5.x

### Device (s)

Raspberry Pi 4 Mod. B

### System

````
 cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage2

 vcgencmd version
Mar 17 2023 10:51:33
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start_x)

 uname -a
Linux lysmarine 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

````

### Logs

[dmesg.log](https://github.com/raspberrypi/linux/files/11869048/dmesg.log)

### Additional context

Let pi4 run for a day. Happens after several hours. If I leave it overnight - almost every day.
Seems happening with usb3 ports. 
I haven't had this behavior when kernel was 5.x

mgrouch, 2023-06-26:
Happened to me on 2 different pi4s. With different models of USB hubs. 
The software was same in both cases.

pelwell, 2023-06-26:
1. Is the hub powered?
2. Is the ASMT drive powered?
3. If you suspect the kernel version change, have you tried reverting? `sudo rpi-update f5c4fc1` will install the most recent 5.15 kernel.

mgrouch, 2023-06-26:
I had this issue even with no hub in another case (mouse and keyboard plugged into pi4 directly, I didn't keep the log but errors in dmesg looked same)

I'll try reverting kernel as you suggested.

mgrouch, 2023-06-28:
Reverting back to 5.x kernel didn't help. Same issues.
I guess it's USB power supply issue then. Kind of hard to troubleshoot. Takes a day to reproduce and
I do not see dmesg under-voltage messages prior to it.

mgrouch, 2023-06-30:
I think I narrowed it down. The issue happens when 2 USB hubs are chained. Both of the hubs are self powered.