raspberrypi/linux #5522: Bluetooth playing is skipping audio on pi4, arm64 bullseye 6.1.21-v8+

mgrouch, 2023-06-26:
### Describe the bug

Bluetooth playing is skipping audio on pi4, arm64 bullseye 6.1.21-v8+

### Steps to reproduce the behaviour

pair iphone and play a song on pi via bluetooth from iphone

### Device (s)

Raspberry Pi 4 Mod. B

### System

 cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage2

 vcgencmd version
Mar 17 2023 10:51:33
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start_x)

 uname -a
Linux lysmarine 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### Logs

````
 dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000026c00000, size 128 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s78504 r8192 d32088 u118784
[    0.000000] pcpu-alloc: s78504 r8192 d32088 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1280x800M@60 smsc95xx.macaddr=DC:A6:32:D5:46:6D vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=abc95165-02 rootfstype=ext4 fsck.repair=yes rootwait quiet loglevel=1 splash quiet logo.nologo vt.global_cursor_default=1 plymouth.ignore-serial-consoles console=tty3 rd.systemd.show_status=false
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[... 629 lines cut ...]
[  341.382109] Bluetooth: Unexpected continuation frame (len 0)
[  341.723360] Bluetooth: Unexpected continuation frame (len 0)
[  341.790861] Bluetooth: Unexpected continuation frame (len 0)
[  341.859607] Bluetooth: Unexpected continuation frame (len 0)
[  342.063378] Bluetooth: Unexpected continuation frame (len 0)
[  342.198366] Bluetooth: Unexpected continuation frame (len 0)
[  342.265869] Bluetooth: Unexpected continuation frame (len 0)
[  342.497099] Bluetooth: Unexpected continuation frame (len 0)
[  405.169179] Bluetooth: hci0: ACL packet for unknown connection handle 12
[  405.544163] Bluetooth: Unexpected continuation frame (len 0)
[  405.589188] Bluetooth: Unexpected continuation frame (len 0)
[  405.607940] Bluetooth: Unexpected continuation frame (len 0)
[  415.424281] input: Mikhail's iPhone 6s (AVRCP) as /devices/virtual/input/input6
[  438.293442] Bluetooth: hci0: ACL packet for unknown connection handle 12
[  438.542139] Bluetooth: Unexpected continuation frame (len 0)
[  438.588382] Bluetooth: Unexpected continuation frame (len 0)
[  438.607126] Bluetooth: Unexpected continuation frame (len 0)
[  448.240462] input: Mikhail's iPhone 6s (AVRCP) as /devices/virtual/input/input7
[  727.329490] usb 2-2.4: reset SuperSpeed USB device number 4 using xhci_hcd
````

### Additional context

feels like regression in 6.x kernel to me.
I use it on system which gets built by a script and it seems issue started with upgrade to 6.1.x kernel

mgrouch, 2023-06-26:
Well I did downgrade of kernel to 5.x

using `sudo rpi-update f5c4fc1`

and same issue with older kernel.

So kernel is not the cause

mgrouch, 2023-06-26:
Setting WiFi country in

/etc/wpa_supplicant/wpa_supplicant.conf

as

`country=US`

and rebooting

fixed bluetooth skipping with Wi-Fi enabled.

pelwell, 2023-06-26:
WiFi 2.4G and Bluetooth share a frequency band, and the WiFi and Bluetooth halves of the 43455 WiFi/BT chip share an antenna, so there is always going to be a degree of conflict between the two. USB3 radiates into the same frequency band, which makes it even worse.

> Setting WiFi country in /etc/wpa_supplicant/wpa_supplicant.conf as country=US and rebooting fixed bluetooth skipping with Wi-Fi enabled.

i. What was it set to before?
ii. What is the correct setting for your region?
iii. Which channels are chosen with US and with the previous Country Code?

pelwell, 2023-06-26:
FYI my Pi 4 is doing a good job of acting as a Bluetooth audio sink for my Android phone and simultaneously streaming YouTube video (and audio) over 5G WiFi.

mgrouch, 2023-06-26:
> WiFi 2.4G and Bluetooth share a frequency band, and the WiFi and Bluetooth halves of the 43455 WiFi/BT chip share an antenna, so there is always going to be a degree of conflict between the two. [...]

1. WiFi country was unset before.
2. US is correct
3. I didn't choose any channels it was in AP mode on Pi4

pelwell, 2023-06-26:
> it was in AP mode on Pi4

That's the kind of non-standard configuration you should disclose when reporting an issue.

> WiFi country was unset before.

Yeah, that's likely to cause problems.

pelwell, 2023-06-26:
I doubt it - that user said they were on a 2.4G connection, and the new power supply seemed to fix it. Thanks for the nudge though.