raspberrypi/linux #5524: ioctl() errors that started to appear with kernel 6.1.21

vanfanel, 2023-06-28:
### Describe the bug

Using some SDL2 programs on MESA on KMS/DRM or Wayland, I see this on the TTY:
```

ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
```

Which in dmesg means:```

[   91.600471] vc4-drm gpu: swiotlb buffer is full (sz: 393216 bytes), total 32768 (slots), used 1074 (slots)
[   91.603514] vc4-drm gpu: swiotlb buffer is full (sz: 393216 bytes), total 32768 (slots), used 1074 (slots)
[   91.712747] vc4-drm gpu: swiotlb buffer is full (sz: 1638400 bytes), total 32768 (slots), used 104 (slots)
[   91.713764] vc4-drm gpu: swiotlb buffer is full (sz: 1638400 bytes), total 32768 (slots), used 104 (slots)
[   91.737476] vc4-drm gpu: swiotlb buffer is full (sz: 1941504 bytes), total 32768 (slots), used 0 (slots)
[   91.738289] vc4-drm gpu: swiotlb buffer is full (sz: 1941504 bytes), total 32768 (slots), used 0 (slots)
[   91.756947] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 1916 (slots)
[   91.761632] vc4-drm gpu: swiotlb buffer is full (sz: 524288 bytes), total 32768 (slots), used 1916 (slots)
```

This started to happen somewhere between kernel **6.1.25-v8+** and kernel **6.1.34-v8+** 

A good game to reproduce this is SDLPoP:
https://github.com/NagyD/SDLPoP

I am using the `vc4-kms-v3d` video overlay, and direct ALSA audio with the **vc4hdmi** audio device, and latest stable MESA 23.1.3

### Steps to reproduce the behaviour

Run and SDL2 program on KMS/DRM or Wayland (like SDLPoP: https://github.com/NagyD/SDLPoP) and look at console and dmesg.

### Device (s)

Raspberry Pi 4 Mod. B

### System

bullseye
Linux debian 6.1.34-v8+ #1657 SMP PREEMPT Fri Jun 16 12:36:29 BST 2023 aarch64 GNU/Linux

pelwell, 2023-06-28:
How much RAM does the Pi 4 have? If it's 8GB, does putting `total_mem=4096` in config.txt get rid of the warnings?

vanfanel, 2023-06-28:
> How much RAM does the Pi 4 have? If it's 8GB, does putting `total_mem=4096` in config.txt get rid of the warnings?

It has 2 GBs. All my Pi4s are 2 GB.

pelwell, 2023-06-28:
Do you just get a few at startup, or are they more continuous than that? I ended up installing Batocera (based on 6.1.25) a few weels ago for some testing (on an 8GB Pi 4) and SDLPoP runs OK, but the shell gets a few swiotlb full messages on startup:
```
[   24.455905] vc4-drm gpu: swiotlb buffer is full (sz: 1622016 bytes), total 32768 (slots), used 5 (slots)
[   24.456940] vc4-drm gpu: swiotlb buffer is full (sz: 1622016 bytes), total 32768 (slots), used 5 (slots)
[   24.492240] vc4-drm gpu: swiotlb buffer is full (sz: 3235840 bytes), total 32768 (slots), used 455 (slots)
[   24.494049] vc4-drm gpu: swiotlb buffer is full (sz: 3235840 bytes), total 32768 (slots), used 519 (slots)
```
However, there was a significant change in the way DMA addresses are derived in our 6.1.28 kernel releases, and Batocera is currently still stuck on 6.1.25.

vanfanel, 2023-06-28:
> Do you just get a few at startup, or are they more continuous than that? I ended up installing Batocera (based on 6.1.25) a few weels ago for some testing (on an 8GB Pi 4) and SDLPoP runs OK, but [...]

I get them at SDLPop startup only, apparently there are no more after that.

pelwell, 2023-06-28:
Running SDLPop from the RPiOs desktop launcher under 6.1.35 + KMS on both 2GB and 8GB Pis, and I see no errors.

pelwell, 2023-06-28:
What does `dmesg | grep TLB` show?

BTW, my mesa seems to be on 20.3.5.

pelwell, 2023-06-28:
Make that `dmesg | grep -E "(TLB|CMA)"`...

vanfanel, 2023-06-28:
> `dmesg | grep -E "(TLB|CMA)"`..
```

dmesg | grep -E "(TLB|CMA)"
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)

```

That's it.

pelwell, 2023-06-28:
Thanks. Can you tell me the top commit for one of the builds, and upload /sys/firmware/fdt somewhere I can grab it? Or email it to phil@raspberrypi.com if you prefer.

vanfanel, 2023-06-28:
> Thanks. Can you tell me the top commit for one of the builds, and upload /sys/firmware/fdt somewhere I can grab it? Or email it to [phil@raspberrypi.com](mailto:phil@raspberrypi.com) if you prefer.

I don't have a `/sys/firmware/fdt` directory. This is a pretty minimal system. Is there a service that mounts that?

Also, I don't know what software build do you mean: kernel/fw? MESA? SDL2? Other?

vanfanel, 2023-06-28:
Ah well, sorry, ftd seems to be a file.
This is it:
[fdt.zip](https://github.com/raspberrypi/linux/files/11899672/fdt.zip)

Still, I don't understand what top commit do you need.

pelwell, 2023-06-28:
The bug report is about the kernel, this is the kernel repo, I want the kernel top commit. Or is it one of our pre-built kernels?

vanfanel, 2023-06-28:
> The bug report is about the kernel, this is the kernel repo, I want the kernel top commit. Or is it one of our pre-built kernels?

It is the latest kernel available in rpi-update.
So this is one of your prebuilt ones.

pelwell, 2023-06-29:
A clean, fully-updated installation of 32-bit Raspberry Pi OS (but with the 64-bit kernel) shows no problems when I launch a freshly built SDLPoP. Running rpi-update on that gets the same kernel as you, and it is also showing no errors or issues of any kind with SDLPoP, even running at 4K.

vanfanel, 2023-06-29:
> A clean, fully-updated installation of 32-bit Raspberry Pi OS (but with the 64-bit kernel) shows no problems when I launch a freshly built SDLPoP. Running rpi-update on that gets the same kernel as you, and it is also showing no errors or issues of any kind with SDLPoP, even running at 4K.

I have a full 64bit system, both kernel and libs. In fact, it's Raspberry Pi OS 64bit.
I am definitely seeing the errors there.

pelwell, 2023-06-29:
The few errors I was seeing were under Batocera, but that was limited to a 6.1.25 kernel. I want to reproduce them under RPiOS with a current kernel, so I'm now starting again with a 64-bit Lite image.

vanfanel, 2023-06-29:
> The few errors I was seeing were under Batocera, but that was limited to a 6.1.25 kernel. I want to reproduce them under RPiOS with a current kernel, so I'm now starting again with a 64-bit Lite image.

You need to build latest stable libDRM, libSDL2 + SDL2_Mixer and latest stable MESA, all on top Raspberry Pi OS LITE, in order to replicate my setup.

pelwell, 2023-06-29:
Running on 64-bit Lite on 6.1.25-v8+ in console mode -> no errors
rpi-update to 6.1.38-v8+ -> no errors
Building SDL2 -> no success

This is turning into a large effort sink for something that just works out of the box on RPiOS. Please give step-by-step instructions for installing the necessary elements to show the problem on a clean RPiOS Lite.

vanfanel, 2023-06-29:
@pelwell SDL2 building instructions follow.

First of all, change to the ROOT user. We don't want to fiddle with permission nonsense on  a microcomputer.

`su`

PART 1: LIBDRM
=============

Uninstall old libdrm:
`apt-get purge libdrm*`

Install libdrm deps:

`sudo apt-get install xsltproc libpciaccess-dev xutils-dev libtool make cmake automake pkg-config gcc g++ python3-setuptools --no-install-recommends`

Download, decompress and configure/build latest stable libdrm:
```
wget https://dri.freedesktop.org/libdrm/libdrm-2.4.115.tar.xz

tar xvpf libdrm-2.4.115.tar.xz

cd libdrm-2.4.115
mkdir build
cd build

CFLAGS="-march=native -mtune=native -O2" \
CXXFLAGS="-march=native -mtune=native -O2" \
meson setup -Dudev=true -Dvc4=enabled -Dintel=disabled -Dvmwgfx=disabled -Dradeon=disabled -Damdgpu=disabled -Dnouveau=disabled \
-Dfreedreno=disabled -Detnaviv=disabled \
-Dinstall-test-programs=true ..

ninja -j4
ninja install
```

PART 2: MESA
============

Install MESA's deps:

`apt-get install --no-install-recommends flex bison python3-mako python3-setuptools libexpat1-dev libudev-dev gettext ca-certificates xz-utils zlib1g-dev pkg-config libvulkan-dev libvulkan1 vulkan-tools`

Download, config, build and install MESA:
```
https://archive.mesa3d.org//mesa-23.1.3.tar.xz
tar xvpf mesa-23.1.3.tar.xz
cd mesa-23.1.3
mkdir build
cd build

CFLAGS="-march=native -mtune=native -O2" \
CXXFLAGS="-march=native -mtune=native -O2" \
meson setup -Dbuildtype=release -Dglx=disabled -Dplatforms=wayland -Dllvm=disabled -Dvulkan-drivers=broadcom -Dgallium-drivers=v3d,vc4,kmsro \
-Dglvnd=false ..
```

PART 3: SDL2
===========

Clean BCRM closed-source cruft to avoid problems:
`rm -R /opt/vc`

Install SDL2 deps:
`apt-get install libudev-dev libasound2-dev libvorbis-dev libflac-dev libmpg123-dev libmodplug-dev libpng-dev libjpeg-dev libfreetype6-dev --no-install-recommends`

Download, configure and build:
```
wget https://github.com/libsdl-org/SDL/releases/download/release-2.28.0/SDL2-2.28.0.tar.gz
tar xvpf SDL2-2.28.0.tar.gz

cd SDL2-2.28.0
mkdir build
cd build

CFLAGS="-march=native -mtune=native -O2 -DEGL_NO_X11" \
CXXFLAGS="-march=native -mtune=native -O2 -DEGL_NO_X11" \
./configure --enable-video-kmsdrm --disable-video-x11 --disable-dbus --disable-diskaudio --disable-oss --disable-pulseaudio --disable-video-rpi --disable-dummyaudio --disable-video-dummy --enable-video-opengles --enable-video-opengl --enable-video-vulkan --enable-libudev --disable-esd --disable-ime --disable-fcitx --disable-libsamplerate

make -j4
make install
```

PART 4 SDL_Image
===============

Since SDLPop needs this one, we will build and install it too.

Download, config and install:

```
wget https://github.com/libsdl-org/SDL_image/releases/download/release-2.6.3/SDL2_image-2.6.3.tar.gz
tar xvpf SDL2_image-2.6.3.tar.gz
cd SDL2_image-2.6.3

CFLAGS="-march=native -mtune=native -O2" \
CXXFLAGS="-march=native -mtune=native -O2" \
./configure

make -j4
make install
```

pelwell, 2023-06-30:
That's really helpful, but I can't get it to work. There were various differences with my system, so I ended up with this installation script:
```
#!/bin/sh

set -eux

# prep
sudo apt update
sudo apt upgrade -y
sudo rm -rf /opt/vc
# sudo apt purge -y $(apt list | grep libdrm | cut -d/ -f1)
sudo apt install --no-install-recommends -y xsltproc libpciaccess-dev xutils-dev libtool make cmake automake pkg-config gcc g++ python3-setuptools libxml2-dev libffi-dev flex bison python3-mako python3-setuptools libexpat1-dev libudev-dev gettext ca-certificates xz-utils zlib1g-dev pkg-config libvulkan-dev libvulkan1 vulkan-tools
reset

# recent meson
sudo sh -c "echo 'deb http://deb.debian.org/debian bullseye-backports main contrib non-free' >> /etc/apt/sources.list"
sudo apt update
sudo apt install -y meson -t bullseye-backports

# libdrm
wget https://dri.freedesktop.org/libdrm/libdrm-2.4.115.tar.xz
tar xvpf libdrm-2.4.115.tar.xz
cd libdrm-2.4.115
mkdir build
cd build
CFLAGS="-march=native -mtune=native -O2" CXXFLAGS="-march=native -mtune=native -O2" meson setup -Dudev=true -Dvc4=enabled -Dintel=disabled -Dvmwgfx=disabled -Dradeon=disabled -Damdgpu=disabled -Dnouveau=disabled -Dfreedreno=disabled -Detnaviv=disabled -Dinstall-test-programs=true ..
ninja -j4
ninja test
sudo ninja install
cd ../..

# wayland
wget https://gitlab.freedesktop.org/wayland/wayland/-/releases/1.22.0/downloads/wayland-1.22.0.tar.xz
tar xvpf wayland-1.22.0.tar.xz 
cd wayland-1.22.0
mkdir build
cd build
meson setup .. --prefix=/usr --buildtype=release -Ddocumentation=false
ninja
ninja test
sudo ninja install
cd ../..

# wayland-protocols
wget https://gitlab.freedesktop.org/wayland/wayland-protocols/-/releases/1.31/downloads/wayland-protocols-1.31.tar.xz
tar xvpf wayland-protocols-1.31.tar.xz 
cd wayland-protocols-1.31
mkdir build
cd build
meson setup --prefix=/usr --buildtype=release
ninja
ninja test
sudo ninja install
cd ../..

# mesa
wget https://archive.mesa3d.org//mesa-23.1.3.tar.xz
tar xvpf mesa-23.1.3.tar.xz
cd mesa-23.1.3
mkdir build
cd build
CFLAGS="-march=native -mtune=native -O2" CXXFLAGS="-march=native -mtune=native -O2" meson setup -Dbuildtype=release -Dglx=disabled -Dplatforms=wayland -Dllvm=disabled -Dvulkan-drivers=broadcom -Dgallium-drivers=v3d,vc4,kmsro -Dglvnd=false ..
ninja
ninja test
sudo ninja install
cd ../..

# SDL2
wget https://github.com/libsdl-org/SDL/releases/download/release-2.28.0/SDL2-2.28.0.tar.gz
tar xvpf SDL2-2.28.0.tar.gz
cd SDL2-2.28.0
mkdir build
cd build
CFLAGS="-march=native -mtune=native -O2 -DEGL_NO_X11" CXXFLAGS="-march=native -mtune=native -O2 -DEGL_NO_X11" ../configure --enable-video-kmsdrm --disable-video-x11 --disable-dbus --disable-diskaudio --disable-oss --disable-pulseaudio --disable-video-rpi --disable-dummyaudio --disable-video-dummy --enable-video-opengles --enable-video-opengl --enable-video-vulkan --enable-libudev --disable-esd --disable-ime --disable-fcitx --disable-libsamplerate
make -j4
sudo make install
cd ../..

# SDL_Image
wget https://github.com/libsdl-org/SDL_image/releases/download/release-2.6.3/SDL2_image-2.6.3.tar.gz
tar xvpf SDL2_image-2.6.3.tar.gz
cd SDL2_image-2.6.3
CFLAGS="-march=native -mtune=native -O2" CXXFLAGS="-march=native -mtune=native -O2" ./configure
make -j4
sudo make install
cd ..

# SDLPop
wget https://github.com/NagyD/SDLPoP/archive/refs/tags/v1.23.tar.gz -O SDLPoP-1.23.tar.gz
tar xvpf SDLPoP-1.23.tar.gz
cd SDLPoP-1.23/src
make -j4
sudo make install
cd ..
```
After that, prince hangs with the following output:
```
init_digi: SDL_OpenAudio: No available audio device
Failed to load sound 0 '(null)'
Failed to load sound 1 '(null)'
Failed to load sound 2 '(null)'
Failed to load sound 3 '(null)'
Failed to load sound 4 '(null)'
Failed to load sound 5 '(null)'
Failed to load sound 6 '(null)'
Failed to load sound 7 '(null)'
Failed to load sound 8 '(null)'
Failed to load sound 9 '(null)'
Failed to load sound 10 '(null)'
Failed to load sound 11 '(null)'
Failed to load sound 12 '(null)'
Failed to load sound 13 '(null)'
Failed to load sound 14 '(null)'
Failed to load sound 15 '(null)'
Failed to load sound 16 '(null)'
Failed to load sound 17 '(null)'
Failed to load sound 18 '(null)'
Failed to load sound 19 '(null)'
Failed to load sound 20 '(null)'
Failed to load sound 21 '(null)'
Failed to load sound 22 '(null)'
Failed to load sound 23 '(null)'
Failed to load sound 44 '(null)'
Failed to load sound 45 '(null)'
Failed to load sound 46 '(null)'
Failed to load sound 47 '(null)'
Failed to load sound 48 '(null)'
Failed to load sound 49 '(null)'
Failed to load sound 51 '(null)'
```
You'll notice that I commented out the libdrm purge, because that was breaking the desktop environment. Leaving it in and running without the desktop on a Lite OS gives the same failure messages.

vanfanel, 2023-06-30:
@pelwell You don't need / want to install Wayland at all. I use latest stable Wayland and an stable compositor, but that implies installing a LOT of other stuff (I can help with that, of course, but that's a different matter).

SDL2 runs on KMS/DRM with the aforementioned instructions, and it's enough to replicate my setup. Testing on X11 is not desired because thats not where I am seeing the errors and buffer management there is totally different.

If you are using the minimal Raspberry Pi OS Bullseye (or any other Debian-based distro, it's a problem with the kernel), the instructions should work.
But if you try them on a desktop system using legacy X11, they won't.

vanfanel, 2023-06-30:
@pelwell Sorry about my previous post: it seems that even if you are building Wayland, you have tested this on KMS/DRM.

The errors you are seeing (`init_digi: SDL_OpenAudio: No available audio device`) are due to audio being missconfigured.
Please remove PulseAudio if it's installed, be sure that the video overlay in config.txt is set to:
```
dtoverlay=vc4-kms-v3d
```
And also in config.txt comment out the line:
```
dtparam=audio=on
```

Now try speaker-test. If speaker-test works, then SDLPoP (any SDL2 program) will work.

pelwell, 2023-06-30:
Audio seems to be the least of my problems - key events aren't working; it's as if the SDL world is not connected to a keyboard or display.

vanfanel, 2023-06-30:
> Audio seems to be the least of my problems - key events aren't working; it's as if the SDL world is not connected to a keyboard or display.

Create `/etc/udev/rules.d/99-evdev.rules` with this content:
`KERNEL=="event*", NAME="input/%k", MODE="666"`
...and reboot.

Or simply do everything as root (that was the very first step in my instructions :D)

pelwell, 2023-07-01:
To save us both time, would you be able to upload an image somewhere that demonstrates the problem?

vanfanel, 2023-07-02:
Here you go:
**https://www.dropbox.com/s/dhqg1gnxh8g7095/pi_minimal.zip?dl=0**

pelwell, 2023-07-03:
Thank you. Installing that image boots to a console prompt as root. I can then run `prince/prince` and it runs full-screen with no issues - dmesg is clean on all the Pi 4's I've tried it on - 1.1, 1.4, those with a BCM2711B0, BCM2711B1 and BCM2711C0, and a variety of EEPROM versions.

vanfanel, 2023-07-03:
Then what's going on with my Raspberry Pi 4B? What a mistery.
Since I am on a 2GB Pi4, did you try on a 2GB model?

pelwell, 2023-07-03:
I don't have a 2GB model, but I've tried to emulate it with `total_mem=2048`.

What do the following commands report?
````
$ tail -4 /proc/cpuinfo
$ vcgencmd bootloader_version
```
I'll ask around the office and try to match yours as closely as possible.

vanfanel, 2023-07-03:
tail -4 /proc/cpuinfo
```
Hardware        : BCM2835
Revision        : b03112
Serial          : 10000000c4a103d0
Model           : Raspberry Pi 4 Model B Rev 1.2
```

vcgencmd bootloader_version
```
2023/01/11 17:40:52
version 8ba17717fbcedd4c3b6d4bce7e50c7af4155cba9 (release)
timestamp 1673458852
update-time 1683799070
capabilities 0x0000007f
```

vanfanel, 2023-07-04:
@pelwell The error doesn't seem to harm the system stability or anything, but really, isn't this a very strange thing? Are there other differences between Rpi4 models besides RAM? I mean, you tried my image on a 2GB model, right?

Other possible explanation would be the video mode in use, since video modes would use different RAM amounts. What resolution were you using when you tested my image? (Assuming you tested in on a 2GB Pi4)

pelwell, 2023-07-05:
My test board is actually a 4GB unit, but with total_mem=2048 it should behave exactly like a 2GB unit.

The display is 4K DELL, but the frame buffer appears scaled. Can you post the output of:
```
$ base64 /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
```
(assuming your monitor is plugged into HDMI0)?

vanfanel, 2023-07-05:
Sure:
```

AP///////wBaYzG7AQEBASkcAQOANR54LmQ1pVRPnicSUFS/74CzAKlAlQCQQIGAgUCBwAEBAjqA
GHE4LUBYLEUAEyshAAAeAAAA/wBVRzIxODQxMDAzNTYKAAAA/QAYkA+0IgAKICAgICAgAAAA/ABY
RzI0MDEgU0VSSUVTAZICAyPxTpAFBAMHHxQhIhMSAT9AIwl/B4MBAABnAwwAEAAARAI6gBhxOC1A
WCxFABMrIQAAHqSDgGpwOE1ACCD4DBMrIQAAHAEdAHJR0B4gbihVABMrIQAAHowK0Iog4C0QED6W
ABMrIQAAGAAAAAAAAAAAAAAAAAAAAAAAAAAA3w==
```

It's indeed connected to HDMI0.

If you don't have a 2GB unit, I could sent you one for debugging. I can't see any other difference.

vanfanel, 2023-07-05:
@pelwell Oh, well. Then there isn't anything else we can do.
If I come up with some new idea so you can replicate the errors, I will come back and tell you.

Thanks for your debugging efforts, really. I know how valuable time is.

vanfanel, 2023-07-05:
@pelwell Found what causes the errors on my Pi4 but not on yours. Guess what? USB keyboard/mouse receiver. This one in `lsusb`:

`Bus 001 Device 004: ID 046d:c534 Logitech, Inc. Unifying Receiver`

If I unplug it, the errors are gone. If I plug it, errors appear when launching SDL2 games:
```
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
```

pelwell, 2023-07-05:
Oh well done. I'll continue to look in that area as time allows.

vanfanel, 2023-07-06:
@pelwell The modules used by Logitech receivers (I have several and they all use the same) are:
```
hid_logitech_hidpp     45056  0
hid_logitech_dj        28672  0
```

The errors only happen on the Raspberry Pi (let's not forget they are `vc4-drm gpu: swiotlb` errors).
Loading the cores manually via `modprobe` without the receiver being physically connected doesn't cause the errors: the receiver must be connected to trigger the bug.

pelwell, 2023-07-06:
>  (let's not forget they are vc4-drm gpu: swiotlb errors).

So I don't misunderstand you, can you confirm that without the receiver plugged in both the swiotlb and ioctl error messages disappear?

vanfanel, 2023-07-06:
**without the receiver plugged in both the swiotlb and ioctl error messages disappear**

That is true.

Plus: with NO logitech receiver connected, loading the modules manually and then running the game, the errors don't appear.

Errors appear ONLY when receiver is connected and the modules are loaded.

In fact, I can connect the receiver, unload the modules, and the errors DO appear.

So the determining factor is the device being connected.
Connected = errors.
Not connected = no errors, even if the modules are loaded.

vanfanel, 2023-07-06:
And one more strange thing: the devices connected to the logitech receiver DO work after UNLOADING the modules `hid_logitech_hidpp` and `hid_logitech_dj`

But in that case, only half of the errors appear.

So, with the DEVICE CONNECTED, and MODULES LOADED:
```

ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
```
In dmesg:
```

[ 2278.920013] vc4-drm gpu: swiotlb buffer is full (sz: 323584 bytes), total 32768 (slots), used 0 (slots)
[ 2278.920761] vc4-drm gpu: swiotlb buffer is full (sz: 323584 bytes), total 32768 (slots), used 0 (slots)
[ 2278.946357] vc4-drm gpu: swiotlb buffer is full (sz: 438272 bytes), total 32768 (slots), used 0 (slots)
[ 2278.947417] vc4-drm gpu: swiotlb buffer is full (sz: 438272 bytes), total 32768 (slots), used 0 (slots)
```

With DEVICE CONNECTED, and modules UNLOADED:

```
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
```

In dmesg:```

[ 2171.665681] vc4-drm gpu: swiotlb buffer is full (sz: 458752 bytes), total 32768 (slots), used 0 (slots)
[ 2171.666443] vc4-drm gpu: swiotlb buffer is full (sz: 458752 bytes), total 32768 (slots), used 0 (slots)
[ 2171.689858] vc4-drm gpu: swiotlb buffer is full (sz: 1048576 bytes), total 32768 (slots), used 12 (slots)
[ 2171.690812] vc4-drm gpu: swiotlb buffer is full (sz: 1048576 bytes), total 32768 (slots), used 12 (slots)
```

pelwell, 2023-07-07:
I don't believe it - with a Logitech wireless dongle (and mouse) I see exactly the same ioctl errors as you _but_ the kernel message log is clean - no swiotlb warnings. Even running a memory stresser to randomly allocate most of the RAM doesn't change things.

vanfanel, 2023-08-08:
@pelwell Any other experiments I should do with this? Do you need more info? Did you find out something new?

pelwell, 2023-08-08:
I stopped looking at this after trying and failing to recreate the swiotlb errors (which are definitely my field). I haven't investigate the ioctl() error messages. Is there any obvious malfunction?

detiam, 2023-08-11:
> No, everything seems to be working fine otherwise.
> Is there a way to ride out of these errors on the TTY/terminal emulator? I currently pass `loglevel=3` to the kernel, but they are still there.

Hi, I found a related issue on SDL side: https://github.com/libsdl-org/SDL/issues/6799
according to that, I first get vendor ID by
> $ lsusb
> Bus 004 Device 002: ID 17ef:109b Lenovo USB3.0 Hub             
> Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 003 Device 005: ID **046d**:c539 **Logitech, Inc. Lightspeed Receiver**
> Bus 003 Device 004: ID 2109:8883 VIA Labs, Inc. USB Billboard Device   
> Bus 003 Device 003: ID 045e:02fe Microsoft Corp. XBOX ACC
> Bus 003 Device 002: ID 17ef:109a Lenovo USB2.0 Hub             
> Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 001 Device 005: ID 8087:0aa7 Intel Corp. Wireless-AC 3168 Bluetooth
> Bus 001 Device 004: ID 26ce:01a2 ASRock LED Controller
> Bus 001 Device 002: ID 056a:037a Wacom Co., Ltd CTL-472 [One by Wacom (S)]
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

and run SDL program with that env var
`env SDL_HIDAPI_IGNORE_DEVICES=0x046d/0x0000 sdl2-jstest --list`

```shell
$ sdl2-jstest --list
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Invalid argument
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
ioctl (GFEATURE): Broken pipe
No joysticks were found

$ env SDL_HIDAPI_IGNORE_DEVICES=0x046d/0x0000 sdl2-jstest --list
No joysticks were found
```

`ioctl (GFEATURE): *` are gone!

vanfanel, 2023-08-11:
@detiam Ah, I guess it should be fixed on SDL's side then?
I will try to replicate and send a PR to SDL in that case.
Why are you passing `SDL_HIDAPI_IGNORE_DEVICES=0x046d/0x0000` instead of `SDL_HIDAPI_IGNORE_DEVICES=0x046d/0xc539`?

vanfanel, 2023-08-11:
Oh well, lsusb says this here:
```

root@DietPi:~/src/SDL-release-2.28.1/test# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 045e:028e Microsoft Corp. Xbox360 Controller
**Bus 001 Device 006: ID 046d:c534 Logitech, Inc. Unifying Receiver**
Bus 001 Device 004: ID 2341:8036 Arduino SA Leonardo (CDC ACM, HID)
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

And if I do:
`env SDL_HIDAPI_IGNORE_DEVICES=0x046d/0x0000 testjoystick`

I get no errors at all. Great! Let's PR this...

vanfanel, 2023-08-19:
Ok, this was fixed on the SDL2 side, so closing this.

Thanks all involved! :)

closed 2023-08-19 by vanfanel