raspberrypi/linux #5533: Kernel Update to 6.1.37? (see CVE-2023-3269, StackRot)

elhennig, 2023-07-10:
### Describe the bug

There is a kernel security issue with all kernels 6.* called StackRot (CVE-2023-3269).

### Steps to reproduce the behaviour

Yet, there is no update available to a version that fixes the problem (like 6.1.37) on bookworm

### Device (s)

Raspberry Pi 3 Mod. B+

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2018-06-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 0d34fb94398a4bd79711f21d6a0b5d35fe5f4a76, stage2
$ vcgencmd version
Mar 17 2023 10:53:00 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start_cd)
$ uname -a
Linux raspi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

pelwell, 2023-07-10:
6.1.37 is likely to be the next pre-built kernel release, possibly later today. Assuming testing uncovers no new issues then it can be rolled out to the stable rpi-firmware branch (and kernel apt package) soon after.

popcornmix, 2023-07-10:
6.1.37 is available through rpi-update.