raspberrypi/linux #5535: Debian certificates for regulatory.db missing

nbuchwitz, 2023-07-12:
### Describe the bug

Debian distributes the wireless regulatory database in the package `wireless-regdb`. The package contains the upstream db as well as a debian version and manages it through update-alternatives. Per default the Debian database is used:

```
pi@RevPi99986:~ $ ls -l /lib/firmware/regulatory.db*
lrwxrwxrwx 1 root root   31  3. Mai 04:53 /lib/firmware/regulatory.db -> /etc/alternatives/regulatory.db
-rw-r--r-- 1 root root 4388 30. Jun 2022  /lib/firmware/regulatory.db-debian
lrwxrwxrwx 1 root root   35  3. Mai 04:53 /lib/firmware/regulatory.db.p7s -> /etc/alternatives/regulatory.db.p7s
-rw-r--r-- 1 root root 1225 30. Jun 2022  /lib/firmware/regulatory.db.p7s-debian
-rw-r--r-- 1 root root 1182 30. Jun 2022  /lib/firmware/regulatory.db.p7s-upstream
-rw-r--r-- 1 root root 4388 30. Jun 2022  /lib/firmware/regulatory.db-upstream
pi@RevPi99986:~ $ ls -l /etc/alternatives/regulatory.db*
lrwxrwxrwx 1 root root 34 12. Jul 10:08 /etc/alternatives/regulatory.db -> /lib/firmware/regulatory.db-debian
lrwxrwxrwx 1 root root 38 12. Jul 10:08 /etc/alternatives/regulatory.db.p7s -> /lib/firmware/regulatory.db.p7s-debian
```

The database is signed and the signature checked by the kernel (certs need to be present in `net/wireless/certs`). This is fine as long as the official upstream version is used (cert present, signature can be verified). The Debian database though is signed by Ben and Romain and the certificates are added in the Debian build process:

https://salsa.debian.org/kernel-team/linux/-/blob/master/debian/patches/debian/wireless-add-debian-wireless-regdb-certificates.patch

On a Raspberry Pi kernel these signatures are not present and the user will find a error in the kernel log:

```
pi@cm4io:~ $ dmesg |grep cfg80211
[    5.515935] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    5.558783] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    5.564276] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
pi@cm4io:~ $ uname -a
Linux cm4io 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

This error has been present since a while and was here and then mentioned as a red herring in wifi issues, My suggested fix is to include the `debian.hex` from the above mentioned patch in the RPi repo. As this is only used for the regdb I don't see any side effects. Happy to provide a PR for this.

### Steps to reproduce the behaviour

Wifi capable Raspberry Pi with Buster / Bullseye

```pi@cm4io:~ $ dmesg |grep cfg80211
[    5.515935] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    5.558783] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    5.564276] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
``` 

### Device (s)

Raspberry Pi CM4

### System

Linux cm4io 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

pelwell, 2023-07-17:
Fixed via #5535. Thanks.

closed 2023-07-17 by pelwell

ValeZAA, 2025-05-12:
update-alternatives --config regulatory.db 

type 3 reboot

update-alternatives --config regulatory.db

type 1 reboot

now it prints:

```
[    3.681797] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    3.685987] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    3.686051] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'

```