raspberrypi/linux #5547: rv3032 Unknown dtparam `trickle-voltage-millivolt` - ignored

beta-tester, 2023-07-24:
### Describe the bug

i have a rv3032 rtc that is capable to use and charge a super capacitor as backup battery.
but the driver ignores it parameter `trickle-voltage-millivolt`

### Steps to reproduce the behaviour

- add to /boot/config.txt
  ```
  dtparam=i2c_arm=on
  dtoverlay=i2c-rtc,rv3032
  dtparam=trickle-voltage-millivolt=1750
  dtparam=trickle-resistor-ohms=1000
  dtparam=backup-switchover-mode=1
  ```
- reboot
- ```
  $ sudo vcdbg log msg
  001722.889: Loaded overlay 'i2c-rtc'
  001722.904: dtparam: rv3032=true
  001725.360: dtparam: trickle-voltage-millivolt=1750
  001729.757: Unknown dtparam 'trickle-voltage-millivolt' - ignored
  001729.793: dtparam: trickle-resistor-ohms=1000
  001741.455: dtparam: backup-switchover-mode=1
  ```

### Device (s)

Raspberry Pi Zero W / WH

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage2
```

```
$ vcgencmd version
Mar 17 2023 10:52:42 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
```

```
$ uname -a
Linux zero1 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
```

### Logs

```
$ sudo vcdbg log msg
001722.889: Loaded overlay 'i2c-rtc'
001722.904: dtparam: rv3032=true
001725.360: dtparam: trickle-voltage-millivolt=1750
001729.757: Unknown dtparam 'trickle-voltage-millivolt' - ignored
001729.793: dtparam: trickle-resistor-ohms=1000
001741.455: dtparam: backup-switchover-mode=1
```

### Additional context

the parameter `trickle-voltage-millivolt` in the driver rv3032 in the raspberry pi and torwalds repo.

https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/rtc/rtc-rv3032.c#L584
https://github.com/torvalds/linux/blob/master/drivers/rtc/rtc-rv3032.c#L584

in the trowalds repo i can not find an overlay file like in the raspberry pi repo to compare:
https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/overlays/i2c-rtc-common.dtsi#L292
https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/overlays/i2c-rtc-common.dtsi#L338

i have no knowledge about dtoverlays. maybe there is just the entry for the parameter `trickle-voltage-millivolt` missing.

pelwell, 2023-07-24:
The overlay does not include such a parameter, but it is easy to add one. I am going to shorten the name to `trickle-voltage-mv` though.

Please give me a description of what the value does to go in the README.

6by9, 2023-07-24:
You've found the device tree bindings - those define what is, and what is not, permitted to be found in device tree for anyone trying to use the device. With the YAML bindings format, any device tree file can be validated against the binding to ensure that it complies.

Mainline Linux has almost no support for device tree overlays - it expects to find a complete base DT file for the complete system in either arch/arm/boot/dts/ or arch/arm64/boot/dts.
For SBCs where users can plug random bits of hardware in, that is very limiting, and is why we have adopted the use of dtoverlays.

beta-tester, 2023-07-24:
oops, by accident i linked the wrong datasheets.
here the correct ones, i updated my previously post:
https://www.microcrystal.com/en/products/real-time-clock-rtc-modules/rv-3032-c7/
https://www.microcrystal.com/fileadmin/Media/Products/RTC/Datasheet/RV-3032-C7.pdf

pelwell, 2023-07-25:
#5549 is the proposed RTC overlay change. Wait about an hour and then run `sudo rpi-update pulls/5549` to install a kernel with the updated overlay.

beta-tester, 2023-07-25:
thank you very much.
with the new added parameter `trickle-voltage-mv` for setting `trickle-voltage-millivolt`, the super cap gets charged now.

does the `sudo rpi-update pulls/5549` has negative effects with normal apt updates/upgrades when the kernel gets regular updates?

pelwell, 2023-07-25:
"rpi-update" and "apt update" are unaware of each other, so the normal apt update will overwrite the release installed by rpi-update. The only real downside to this is that you may sometimes have force a reinstall because the update mechanism is unaware that the thing it last installed is no longer there.

beta-tester, 2023-07-25:
maybe RV3032 sould be mention for `backup-switchover-mode` as well. it is not `RV3028 only`

```
        trickle-resistor-ohms   Resistor value for trickle charge (DS1339,
                                ABx80x, BQ32000, RV1805, RV3028, RV3032)

        trickle-voltage-mv      Charge pump voltage for trickle charge (RV3032)

        wakeup-source           Specify that the RTC can be used as a wakeup
                                source

        backup-switchover-mode  Backup power supply switch mode. Must be 0 for
                                off or 1 for Vdd < VBackup (RV3028 only)
```

watching the `README` i now understand, why the parameter was shorten. it is just to fit nicely in the existing document format without reformat everything... :D

pelwell, 2023-07-25:
> maybe RV3032 sould be mention for backup-switchover-mode as well

Done, and merged.

pelwell, 2023-07-25:
Yes they will, but I won't make any promises as to when that will be.

closed 2023-07-25 by pelwell