raspberrypi/linux #5550: Waveshare 7.9" DSI not working android pi4

beletruz, 2023-07-25:
### Describe the bug

The latest 6.1.31 doesn't work with the latest raspberry pi os with this display https://www.waveshare.com/wiki/7.9inch_DSI_LCD#Overview by adding dtoverlay=vc4-kms-dsi-waveshare-panel,7_9_inch the display stays off and the led blinks but it does work (it lights up, it displays as it should) except for the touchscreen with android running on the pi (this android https://konstakang.com/devices/rpi4/LineageOS20 )
Using the "dtoverlay=vc4-kms-dsi-waveshare-panel,7_9_inch" and "goodix" for touch this is what it reports about the touchscreen

### Steps to reproduce the behaviour

Flash LieageOS20 -> flash 6.1 kernel -> the display shows the input but the touch doesn't work

### Device (s)

Raspberry Pi 4 Mod. B

### System

OS https://konstakang.com/devices/rpi4/LineageOS20
version d7f9c2b4ef7e4a8c0b04374a879ce89d7a948453 (clean) (release) (start _ x)
Linux localhost 6.1.31-v8-g84ed1b6094c9 #1 SMP PREEMPT Thu Jun 15 UTC 2023 aarch64 Toybox

### Logs

[ 3.486535] Goodix-TS 1-0014: supply AVDD28 not found, using dummy regulator
[ 3.493801] Goodix-TS 1-0014: supply VDDIO not found, using dummy regulator
[ 3.597518] Goodix-TS 1-0014: Error reading 1 bytes from 0x8140: -5
[ 3.629484] Goodix-TS 1-0014: Error reading 1 bytes from 0x8140: -5
[ 3.661185] Goodix-TS 1-0014: I2C communication failure: -5
[ 3.667235] Goodix-TS: probe of 1-0014 failed with error -5
[ 3.674567] i2c i2c-22: Added multiplexed i2c bus 0
[ 3.679906] Goodix-TS 10-0014: supply AVDD28 not found, using dummy regulator
[ 3.687263] Goodix-TS 10-0014: supply VDDIO not found, using dummy regulator
[ 3.695975] Goodix-TS 10-0014: ID 9271, version: 1070
[ 3.718404] input: Goodix Capacitive TouchScreen as /devices/platform/soc/fe205000.i2c/i2c-22/i2c-10/10-0014/input/input0

### Additional context

a link for the conversation I had with Konsta regarding this https://konstakang.com/devices/rpi4/LineageOS20/#comment-6236383678

6by9, 2023-07-25:
So it's only the touch part that has issues?

1-0014 is a Goodix touch screen on i2c-1 - that has been invoked by your adding of "dtoverlay=goodix". If you haven't changed the touch DIP switches on the panel, then this is unnecessary.

10-0014 is a Goodix touch screen on i2c-10, aka i2c_csi_dsi, and is correct for most of the Waveshare panels. It's also probed correctly as ```/devices/platform/soc/fe205000.i2c/i2c-22/i2c-10/10-0014/input/input0```.

You were advised correctly by Konsta in https://konstakang.com/devices/rpi4/LineageOS20/#comment-6234734667
> 'dtoverlay=vc4-kms-dsi-waveshare-panel,7_9_inch' also configures the goodix touchscreen driver for specific resolution so that's the reason to **add just the one line and nothing else.**

Use ```evtest``` to confirm whether the device is registered correctly or not. Konsta had also advised an equivalent in 
> You can check 'dmesg' output booting the device, **'cat /sys/class/input/input*/name' that the touchscreen is listed as input device,** and 'getevent' tool in rooted shell to see if it's sending any input events.

I know nothing of getevent as I suspect that is an Android tool rather than Linux.

I don't have a 7.9" display (only 2.4" and 10.1") so can't test, but your logs imply that the touch driver is loaded correctly (albeit twice due to the ```dtoverlay=goodix``` line).

beletruz, 2023-07-25:
when i use "getevent" these 3 devices get recognized
could not get driver version for /dev/input/mice,
add device 1: /dev/input/event2
Inappropriate ioctl for device
name :
"Goodix Capacitive TouchScreen"
add device 2: /dev/input/event3
name :
" button@15 "
could not get driver version for /dev/input/mousel,
add device 3: /dev/input/event@
Inappropriate ioctt for device
name :
"Razer Razer Mamba"

In which there is also the touch screen and if i touch it it shows "being touched" in the terminal it gives these
/dev/input/event2: 0003 0035 fffff6c1
/dev/input/event2: 0003 0036 fffffecc
and so on. From this I understand that the touchscreen is initialized and the system sees it. How should i proceed next?  
(alsi i think evtest dosen't work on android or at least in my case it was not recognized )

6by9, 2023-07-25:
```
add device 1: /dev/input/event2
Inappropriate ioctl for device
name :
"Goodix Capacitive TouchScreen"
```
Looks good, and if touching it then reports "being touched" then everything at the kernel level is doing exactly what it's meant to.

The Goodix touch driver is a little strange in that it registers as both a touch and keyboard device - it can support a number of touch keys as well as the touch screen, so is both. I've seen other userspace apps doing the wrong thing because they were looking for a plain touch device, not a touch and keyboard device. However that is up into userspace, and not a kernel issue.

beletruz, 2023-07-25:
That being said I should get back to Konsta with this information, right?
If you have any other ideas or theories on the issue I'm all ears (better said eyes :)) )

6by9, 2023-07-25:
If you say getevent is registering touch events, then I see nothing left in the kernel that is wrong.

beletruz, 2023-07-25:
So it's up to Konsta to solve this part on the android ROM side. Many thanks, hopefully we will be victorious in the end 😅

ilarrain, 2023-08-05:
I have the same panel, editing the overlay did the trick. The default values were incorrect for "touchscreen-size-x:0", "touchscreen-size-y:0", "touchscreen-inverted-y", and "touchscreen-swapped-x-y".

beletruz, 2023-08-05:
> I have the same panel, editing the overlay did the trick. The default values were incorrect for "touchscreen-size-x:0", "touchscreen-size-y:0", "touchscreen-inverted-y", and "touchscreen-swapped-x-y". 

You used these on the android build and it worked?

ilarrain, 2023-08-05:
> You used these on the android build and it worked?

No, in stock Raspberry Pi OS, testing latest kernel:
```
$ uname -a
Linux raspberrypi 6.1.42-v8+ #1668 SMP PREEMPT Thu Jul 27 15:34:18 BST 2023 aarch64 GNU/Linux
```

I was having the same issues you discussed in https://konstakang.com/devices/rpi4/LineageOS20/#comment-6236383678

beletruz, 2023-08-05:
> No, in stock Raspberry Pi OS, testing latest kernel:
> ```
> $ uname -a
> Linux raspberrypi 6.1.42-v8+ #1668 SMP PREEMPT Thu Jul 27 15:34:18 BST 2023 aarch64 GNU/Linux
> ```
>
> I was having the same issues [...]

OK I'll try that with a stock kernel because I have already modified the overlays with a file from Konsta

ilarrain, 2023-08-05:
> OK I'll try that with a stock kernel because I have already modified the overlays with a file from Konsta
I saw your discussion. I couldn't download the file he provided, but for testing, I modified the overlay to do the same. Setting sizex=4096,sizey=4096 should do the trick for you too. Or, you can also try the overlay I compiled: https://drive.google.com/file/d/1vxbMmwprV20rWj2UMmCCoejJhksR0Og8/view?usp=drivesdk

beletruz, 2023-08-07:
> > OK I'll try that with a stock kernel because I have already modified the overlays with a file from Konsta
> 
> I saw your discussion. I couldn't download the file he provided, but for testing, I modified the overlay to do the same. Setting sizex=4096,sizey=4096 should do the trick for you too. Or, you can also try the overlay I compiled with the right defaults baked in (it shouldn't need any options as sizex..., other than the preferred rotation): https://drive.google.com/file/d/1vxbMmwprV20rWj2UMmCCoejJhksR0Og8/view?usp=drivesdk

I can't believe it actually worked, thank you so much :) You have no idea how happy you made my evening :D thanks a ton