raspberrypi/linux #5552: CONFIG_USB_FUNCTIONFS is missing

knro, 2023-07-27:
bcm2711 kernel configuration is missing CONFIG_USB_FUNCTIONFS which is required to use CM4 as a USB gadget using functionfs and libcomposite. GadgetFS is enabled but this is an older implementation introduced in 2003 while functionfs is newer was introduced in 2010 and enables MTP support for some devices.

Please enable CONFIG_USB_FUNCTIONFS as a module by default.

pelwell, 2023-07-27:
I think it's telling that the `USB_FUNCTIONFS` support is found in `usb/gadget/legacy/`, suggesting that it is already outdated. Not even Ubuntu (what I consider to be one of the more complete distributions) enables it, even as a module.

USB_CONFIGFS seems to be the preferred gadget interface, and it is enabled in all Pi kernels:
```
CONFIG_USB_CONFIGFS=m
CONFIG_USB_CONFIGFS_SERIAL=y
CONFIG_USB_CONFIGFS_ACM=y
CONFIG_USB_CONFIGFS_OBEX=y
CONFIG_USB_CONFIGFS_NCM=y
CONFIG_USB_CONFIGFS_ECM=y
CONFIG_USB_CONFIGFS_ECM_SUBSET=y
CONFIG_USB_CONFIGFS_RNDIS=y
CONFIG_USB_CONFIGFS_EEM=y
CONFIG_USB_CONFIGFS_MASS_STORAGE=y
CONFIG_USB_CONFIGFS_F_LB_SS=y
CONFIG_USB_CONFIGFS_F_FS=y
CONFIG_USB_CONFIGFS_F_UAC1=y
CONFIG_USB_CONFIGFS_F_UAC2=y
CONFIG_USB_CONFIGFS_F_MIDI=y
CONFIG_USB_CONFIGFS_F_HID=y
CONFIG_USB_CONFIGFS_F_UVC=y
CONFIG_USB_CONFIGFS_F_PRINTER=y
```

knro, 2023-07-27:
Right the legacy bit was a red flag for me too, but since I was looking for a [MTP Responder](https://github.com/viveris/uMTP-Responder) solution, and the only one I found suggested either GadgetFS or FunctionFS. The former proved to be unreliable with CM4, so they reported better compatibility with FunctionFS. Any MTP frameworks that make use of CONFIGFS?

pelwell, 2023-07-27:
Is this one any good? https://www.collabora.com/news-and-blog/blog/2021/04/13/an-easy-mtp-implementation-for-your-next-embedded-linux-project/

It requires CONFIG_USB_CONFIGFS_F_FS:
```
config USB_CONFIGFS_F_FS
	bool "Function filesystem (FunctionFS)"
	depends on USB_CONFIGFS
	select USB_F_FS
	help
	  The Function Filesystem (FunctionFS) lets one create USB
	  composite functions in user space in the same way GadgetFS
	  lets one create USB gadgets in user space.  This allows creation
	  of composite gadgets such that some of the functions are
	  implemented in kernel space (for instance Ethernet, serial or
	  mass storage) and other are implemented in user space.
```
which is already enabled.

knro, 2023-07-28:
Thanks, that was my 2nd option so I'll give it a go! Thank you for your help.

closed 2023-07-28 by knro