raspberrypi/linux #5553: Kernel aarch64 6.1.42 compile issue

Dark-Sky, 2023-07-27:
### Describe the bug

After compiling the latest 6.1.42 kernel today there was no bcm2710 .dtb's built in `arch/arm64/boot/dts/broadcom/`. The bcm2711 .dtb's built ok. The bcm2710 .dtb's built just fine with kernel 6.1.41 yesterday. 

### Steps to reproduce the behaviour

Used `make Image modules dtbs`

### Device (s)

Raspberry Pi 400

### System

Kernel built on manjaro's build server but I did compile the dtb's only here `make dtbs` with the same result on my pi400.

[ray@jellyfin ~]$ vcgencmd version
Apr 25 2023 18:26:03 
Copyright (c) 2012 Broadcom
version d7f9c2b4ef7e4a8c0b04374a879ce89d7a948453 (clean) (release) (start)

[ray@jellyfin ~]$ uname -a
Linux jellyfin 6.1.41-1-MANJARO-ARM-RPI #1 SMP PREEMPT Wed Jul 26 17:12:32 UTC 2023 aarch64 GNU/Linux

### Additional context

I tried to chase down any changes but ran out of time as I have to go to town for a doctor appt. Could the d2a6dc4eaf6d50ba32a9b39b4c6ec713a92072ab.gz downloaded be corrupted.

Dark-Sky, 2023-07-27:
I see the hash has changed to 431319d91b8584c9f28b195ab9a97d7e78905aeb. I'm recompiling the kernel.

pelwell, 2023-07-27:
d2a6dc4 is an upstream commit which will give you an upstream kernel, where the dtbs have different name (bcm2837-*). 431319d is the downstream merge commit that pulls in d2a6dc4 et al., which is what you want.

Dark-Sky, 2023-07-27:
No clue how I wound up with that commit but all good now. Closing this non-issue.

closed 2023-07-27 by Dark-Sky