raspberrypi/linux #5556: RPI4 VNC connection and vsync frame rate (extreme high application lags)

qrp73, 2023-07-29:
### Describe the bug

I'm using Raspberry Pi 4B 4G revision 1.5. I don't have hdmi display, so I'm using VNC connection.

I installed the latest 2023-05-03-raspios-bullseye-arm64.img and I catch the issue with many programs like chromium and other open gl based. The problem here is that these programs have extreme lags when you start it from VNC connection. 

I tested also 2023-05-03-raspios-bullseye-armhf.img it has the same issue, or even worse.

I investigated the problem and found that if you connect through VNC and there is no physical display connected to rpi4, the video driver uses vblank sync at frame rate 1 Hz for some unknown reason, so all programs which behavior is based on display frame rate have extreme lags.

Current workaround is to run these programs from console with vblank_mode=0, for example:
```
vblank_mode=0 chromium-browser
```

it solves the issue and program works without lags. 

But starting program from console with adding vblank_mode=0 is a very inconvenient and crooked way.

Could you please fix that bug by adding user setting for default display frame rate for VNC connection? (when physical display is not connected to hdmi) Or at least please replace hardcoded 1 Hz with hardcoded no wait for vblank or at least use hardcoded adequate frame rate, for example 60 Hz. I think the best way is to just skip vblank sync when app running through VNC connection and there is no physical display to determine display frame rate.

### Steps to reproduce the behaviour

1) Install 2023-05-03-raspios-bullseye-arm64.img or 2023-05-03-raspios-bullseye-armhf.img on sdcard
2) Run RPI4 in headless mode (do not attach display to hdmi) and enable VNC through SSH
3) Connect with VNC client (I'm using tigervnc)
4) Start chromium-browser
5) Notice that it has exterme high lags (freezes for 1-2 seconds) and doesn't allow to see video on yotube.

Actual result: application working with waiting for vblank with refresh rate 1 Hz, as result it leads to extreme high lags in different graphics applications such as chromium-browser.

Expected result: application working with no waiting for vblank at all, or at least using waiting with adequate frame rate, for example 60 fps.

### Device (s)

Raspberry Pi 4 Mod. B

### System

pi@raspi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage4
pi@raspi:~ $ vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
pi@raspi:~ $ uname -a
Linux raspi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### Logs

pi@raspi:~ $ chromium-browser
[144592:144592:0729/070218.521420:ERROR:chrome_browser_cloud_management_controller.cc(162)] Cloud management controller initialization aborted as CBCM is not enabled.
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
[144645:144645:0729/070222.712710:ERROR:sync_control_vsync_provider.cc(139)] Calculated bogus refresh interval=0.987903 s, last_timebase_=66549017266 bogo-microseconds, timebase=66550005169 bogo-microseconds, last_media_stream_counter_=66549, media_stream_counter=66550
[144645:144645:0729/070223.731201:ERROR:sync_control_vsync_provider.cc(139)] Calculated bogus refresh interval=1.01908 s, last_timebase_=66550005169 bogo-microseconds, timebase=66551024252 bogo-microseconds, last_media_stream_counter_=66550, media_stream_counter=66551
[144645:144645:0729/070224.711158:ERROR:sync_control_vsync_provider.cc(139)] Calculated bogus refresh interval=0.979255 s, last_timebase_=66551024252 bogo-microseconds, timebase=66552003507 bogo-microseconds, last_media_stream_counter_=66551, media_stream_counter=66552
[144645:144645:0729/070225.711020:ERROR:sync_control_vsync_provider.cc(139)] Calculated bogus refresh interval=1.00047 s, last_timebase_=66552003507 bogo-microseconds, timebase=66553003974 bogo-microseconds, last_media_stream_counter_=66552, media_stream_counter=66553
[144645:144645:0729/070226.711805:ERROR:sync_control_vsync_provider.cc(139)] Calculated bogus refresh interval=0.978528 s, last_timebase_=66553025959 bogo-microseconds, timebase=66554004487 bogo-microseconds, last_media_stream_counter_=66553, media_stream_counter=66554
[144645:144645:0729/070227.712375:ERROR:sync_control_vsync_provider.cc(139)] Calculated bogus refresh interval=1.00085 s, last_timebase_=66554004487 bogo-microseconds, timebase=66555005336 bogo-microseconds, last_media_stream_counter_=66554, media_stream_counter=66555
[144645:144645:0729/070228.745964:ERROR:sync_control_vsync_provider.cc(139)] Calculated bogus refresh interval=1.03362 s, last_timebase_=66555005336 bogo-microseconds, timebase=66556038960 bogo-microseconds, last_media_stream_counter_=66555, media_stream_counter=66556
[144645:144645:0729/070230.712380:ERROR:sync_control_vsync_provider.cc(139)] Calculated bogus refresh interval=0.965384 s, last_timebase_=66557040109 bogo-microseconds, timebase=66558005493 bogo-microseconds, last_media_stream_counter_=66557, media_stream_counter=66558

### Additional context

Also could someone help me to find the way to set vblank_mode=0 globally in current raspberry pi os?
Thanks

popcornmix, 2023-07-30:
This repo is for issues with the linux kernel. This sounds like a distro issue - can you recreate the issue here:
https://github.com/RPi-Distro/repo/issues/

where the right devs should see it.

qrp73, 2023-07-31:
ok

popcornmix, 2023-07-31:
Thanks - we'll let further discussion occur there.

closed 2023-07-31 by popcornmix