raspberrypi/linux #5561: Jumbo frame support on Pi4 ethernet (Genet)

6by9, 2023-08-02:
### Describe the bug

From #5419 (closed PR) where someone had proposed a total hack to increase the max packet length.

Queried on net-dev
https://www.spinics.net/lists/netdev/msg919558.html
https://www.spinics.net/lists/netdev/msg925029.html
So Florian is going to have a little look into it.

Opening new issue to track anything that comes back from it.

### Steps to reproduce the behaviour

Current max packet size only allows standard ethernet 1500 byte MTU (+ VLAN headers).

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 400, Raspberry Pi CM4, Raspberry Pi CM4 Lite

### System

All kernels to current (6.5)

wtschueller, 2023-08-06:
The freeze occurs also on my CM4. Steps to reproduce:

```sh
 sudo ip link set dev eth0 mtu 2036
 while (true) do dd if=/dev/zero bs=65536 count=1000 |netcat -u 169.254.1.1 65000 -q 0; done

```

Log message:

```
[Aug 4 15:03] ------------[ cut here ]------------
[  +0.000026] NETDEV WATCHDOG: eth0 (bcmgenet): transmit queue 0 timed out
[  +0.000067] WARNING: CPU: 3 PID: 0 at net/sched/sch_generic.c:442 dev_watchdog+0x38c/0x394
[  +0.000004] Modules linked in: algif_hash aes_neon_bs aes_neon_blk crypto_simd cryptd algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc brcmfmac brcmutil sha256_generic cfg80211 rfkill raspberrypi_hwmon bcm2835_unicam v4l2_dv_timings vc4 v3d rpivid_hevc(C) cec gpu_sched bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig drm_kms_helper videobuf2_memops videobuf2_v4l2 dummy_sensor videobuf2_common v4l2_fwnode i2c_mux_pinctrl snd_soc_core i2c_brcmstb i2c_mux snd_compress snd_pcm_dmaengine snd_bcm2835(C) snd_pcm snd_timer vc_sm_cma(C) i2c_bcm2835 snd videodev syscopyarea mc sysfillrect genet sysimgblt fb_sys_fops nvmem_rmem uio_pdrv_genirq uio i2c_dev fuse drm drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[  +0.000180] CPU: 3 PID: 0 Comm: swapper/3 Tainted: G         C        5.10.17-v8+jf+ #26
[  +0.000004] Hardware name: Raspberry Pi Compute Module 4 Rev 1.1 (DT)
[  +0.000004] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
[  +0.000005] pc : dev_watchdog+0x38c/0x394
[  +0.000004] lr : dev_watchdog+0x38c/0x394
[  +0.000002] sp : ffffffc011583ce0
[  +0.000004] x29: ffffffc011583ce0 x28: ffffff8103ef2580
[  +0.000007] x27: 0000000000000004 x26: ffffff8106400480
[  +0.000006] x25: 0000000000000140 x24: 00000000ffffffff
[  +0.000006] x23: ffffff81064003dc x22: 0000000000000003
[  +0.000013] x21: ffffffc011246000 x20: ffffff8106400000
[  +0.000011] x19: 0000000000000000 x18: 00000000fffffffe
[  +0.000010] x17: 00000110372503c0 x16: 000000000000001f
[  +0.000006] x15: 0000000000000020 x14: ffffffffffffffff
[  +0.000006] x13: ffffffc01142ac10 x12: ffffffc01142a863
[  +0.000006] x11: ffffff8100459240 x10: ffffffc0112bd378
[  +0.000006] x9 : ffffffc0100ed8c0 x8 : ffffffc011265378
[  +0.000006] x7 : ffffffc0112bd378 x6 : 0000000000000003
[  +0.000008] x5 : 0000000000000000 x4 : 0000000000000000
[  +0.000009] x3 : ffffffc01124a628 x2 : 0000000000000000
[  +0.000007] x1 : 0000000000000000 x0 : ffffff8100250000
[  +0.000006] Call trace:
[  +0.000005]  dev_watchdog+0x38c/0x394
[  +0.000006]  call_timer_fn+0x38/0x200
[  +0.000006]  __run_timers.part.0+0x230/0x300
[  +0.000005]  run_timer_softirq+0x54/0x90
[  +0.000007]  __do_softirq+0x1a8/0x544
[  +0.000006]  irq_exit+0xf0/0x104
[  +0.000007]  __handle_domain_irq+0xc0/0x13c
[  +0.000004]  gic_handle_irq+0x5c/0xf0
[  +0.000003]  el1_irq+0xc4/0x180
[  +0.000007]  arch_cpu_idle+0x18/0x30
[  +0.000005]  default_idle_call+0x40/0x1d4
[  +0.000011]  do_idle+0x260/0x2a4
[  +0.000004]  cpu_startup_entry+0x34/0x7c
[  +0.000005]  secondary_start_kernel+0x138/0x184
[  +0.000004] ---[ end trace f6b66bf422679008 ]---
```
Info

```
System Information
------------------

Raspberry Pi Compute Module 4 Rev 1.1
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2023-02-21
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 25e2319effa91eb95edd9d9209eb9f8a584d67be, stage2

Linux raspberrypi 5.10.17-v8+jf+ #26 SMP PREEMPT Thu Aug 3 13:02:25 CEST 2023 aarch64 GNU/Linux
Revision: d03141
Serial: 10000000f5502b46
Model: Raspberry Pi Compute Module 4 Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Jan  5 2023 10:46:54
Copyright (c) 2012 Broadcom
version 8ba17717fbcedd4c3b6d4bce7e50c7af4155cba9 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0
```

Could increasing TBUF_TXCHK_PKTRDY_THLD (name taken from https://github.com/raspberrypi/linux/pull/5419#issuecomment-1632736483) be of any help? What is its address?

wtschueller, 2023-08-06:
I can confirm, that incoming jumbo packets are not fragmented into multiple buffers. This observation does not agree with my old records. I believe that I made mistakes interpreting the log messages in the past.

Using a similar patch like https://github.com/raspberrypi/linux/pull/5419#issuecomment-1633165459  and code like this
```sh
for ((SIZE=3700; SIZE<3900; ((SIZE=SIZE+1)) )) do ping 169.254.1.1 -c 1 -v -s ${SIZE}; done
```
I see the following log:

```
...
[  +0,005467] bcmgenet_desc_rx:p_ind=3299 c_ind=3298 read_ptr=226 len_stat=0x0f3c7f80 len= 3900 reg_len_stat=0x0f3c7f80 reg_len= 3900
[  +0,005369] bcmgenet_desc_rx:p_ind=3300 c_ind=3299 read_ptr=227 len_stat=0x0f3d7f80 len= 3901 reg_len_stat=0x0f3d7f80 reg_len= 3901
[  +0,004767] bcmgenet_desc_rx:p_ind=3301 c_ind=3300 read_ptr=228 len_stat=0x0f3e7f80 len= 3902 reg_len_stat=0x0f3e7f80 reg_len= 3902
[  +0,004952] bcmgenet_desc_rx:p_ind=3302 c_ind=3301 read_ptr=229 len_stat=0x0f3f7f80 len= 3903 reg_len_stat=0x0f3f7f80 reg_len= 3903
[  +0,006352] bcmgenet_desc_rx:p_ind=3303 c_ind=3302 read_ptr=230 len_stat=0x0f407f80 len= 3904 reg_len_stat=0x0f407f80 reg_len= 3904
[  +0,005886] bcmgenet_desc_rx:p_ind=3304 c_ind=3303 read_ptr=231 len_stat=0x0f403f80 len= 3904 reg_len_stat=0x0f817f80 reg_len= 3969 # I expect 3905 instead
[  +1,686197] bcmgenet_desc_rx:p_ind=3305 c_ind=3304 read_ptr=232 len_stat=0x0f403f80 len= 3904 reg_len_stat=0x0f827f80 reg_len= 3970 # I expect 3906 instead
...

```
Packet length from register is 64 bytes larger then I would expect in case of length exceeding threshold.

wtschueller, 2023-08-07:
Well, I guess TBUF_PKT_RDY_THLD is at 0x10. This test triggers the  time out/freeze at mtu = 2036:

```sh
for ((MTU=1500; MTU<=3840; ((MTU=MTU+4)))) do sudo ifconfig eth0 mtu ${MTU} ; ip -d link list eth0 ; dd if=/dev/zero bs=65536 count=1024 |netcat -u 169.254.1.1 65000 -q 0; done
```
The following patch pushes that limit by 0x70*16 = 1792 to 3828. So I can safely use mtu = 3824 now.
```diff
Date: Mon, 7 Aug 2023 15:15:18 +0200
Subject: net: bcm: genet: Set the TBUF_PKT_RDY_THLD register

Set the TBUF_PKT_RDY_THLD register to the max value that it can
take to allow for larger frames. Allows for mtu up to 3824.

diff --git a/drivers/net/ethernet/broadcom/genet/bcmgenet.c b/drivers/net/ethernet/broadcom/genet/bcmgenet.c
index 9749d6678..4a6038b7c 100644
--- a/drivers/net/ethernet/broadcom/genet/bcmgenet.c
+++ b/drivers/net/ethernet/broadcom/genet/bcmgenet.c
@@ -2502,6 +2502,7 @@ static void init_umac(struct bcmgenet_priv *priv)
 	 */
 	reg = 0xf0;
 	bcmgenet_rbuf_writel(priv, reg, RBUF_PKT_RDY_THLD);
+	bcmgenet_writel(reg, priv->base + priv->hw_params->tbuf_offset + TBUF_PKT_RDY_THLD);
 
 	/* enable rx checksumming */
 	reg = bcmgenet_rbuf_readl(priv, RBUF_CHK_CTRL);
diff --git a/drivers/net/ethernet/broadcom/genet/bcmgenet.h b/drivers/net/ethernet/broadcom/genet/bcmgenet.h
index 52068d8bf..d470364ca 100644
--- a/drivers/net/ethernet/broadcom/genet/bcmgenet.h
+++ b/drivers/net/ethernet/broadcom/genet/bcmgenet.h
@@ -283,6 +283,8 @@ struct bcmgenet_mib_counters {
 #define  TBUF_EEE_EN			(1 << 0)
 #define  TBUF_PM_EN			(1 << 1)
 
+#define TBUF_PKT_RDY_THLD		0x10
+
 #define TBUF_CTRL_V1			0x80
 #define TBUF_BP_MC_V1			0xA0
```

cdalvaro, 2023-12-14:
This feature would be great

6by9, 2023-12-15:
I've just pinged the thread on net-dev to see if Florian ever found any docs. I'm afraid it's a fairly low priority all round.

mattjudge, 2025-01-13:
Thanks so much for your effort on this, it's good to see the issue is still alive. If it's useful, another example use-case datapoint is that for GigE vision cameras.

herisson-88, 2026-05-20:
I deleted my comment because I realized that it is not a kernel regression but the patch of my kernel distrib probably lost...

herisson-88, 2026-09-02:
It would be so nice to not have to patch for jumbo :-)

6by9, 2026-09-02:
> It would be so nice to not have to patch for jumbo :-)

As I asked you earlier in this thread, what patch are you applying? Or provide a link to the distribution / kernel that you are running which supports it. Something that can be analysed.

At present the best I have managed to achieve is 3840 byte packets, not the full 9000 that most network adapters support. Whilst an improvement, it's not overly useful.
I will ask Broadcom again to see if there is more information,

herisson-88, 2026-09-02:
But Audiolinux (I am not the maintainer of AL) use one of this from this thread ;-) And for our usage even a small rise of MTU is usefull. Could you give me your last version patch for 3840 ?

6by9, 2026-09-02:
My patch for 3840 bytes are the top two patches in #5534, and they still apply on 6.18
https://github.com/raspberrypi/linux/pull/5534/changes/681f4aacd153b4ac4234229ae0a070dfc5e089b1
https://github.com/raspberrypi/linux/pull/5534/changes/0035c464c24b4229565c4b5b02f18be773d58a92
They actually allow the request for a 9000 byte MTU, but the hardware won't work with that.

Reasoning for 3840 bytes is covered in https://www.spinics.net/lists/netdev/msg919558.html
> I've had a look through and can increase the buffer sizes
appropriately, but it seems that we get the notification of the packet
whenever the threshold defined in GENET_0_RBUF_PKT_RDY_THLD is hit.
The status block appears to be written at this point, so we end up
dropping the packet because the driver believes it is fragmented (SOP
but not EOP set [1]), and presumably any checksum offload is from that
point in the frame too.
Setting RBUF_PKT_RDY_THLD to 0xF0 (units of 16bytes) allows for 3840
byte buffers plus the 64byte status block, and that all seems to work

and https://github.com/raspberrypi/linux/pull/5419#issuecomment-1631067165
> There is a register "GENET_0_RBUF_PKT_RDY_THLD - RX Buffer Packet Ready Threshold Register" at offset 0x08 which is the threshold for PKT_RDY being asserted, in units of 16bytes. Default is 0x80, which * 16bytes = 2048.
However it's only an 8bit value, so max 255 * 16 = 4080bytes, however recommended to be divisible by the burst size (256B).
Increase it to 0xf0 for 3840 bytes, and I can receive up to 3796 byte ping payloads. dma_length_status is 0x0f407f80.

The ideal may be to support fragmented receive, but I'm not sure we have docs on exactly how that works, nor for how that would actually populate the buffers in memory.
I'll try to find a few minutes to create a clean PR enabling 3840bytes, and advertising that to userspace.

Seeing as Audio-linux appears to be a paid for distribution, I can't find any information about their packaging or kernel patching.
You are totally entitled to request source code for their kernel under the [GPLv2 licence](https://www.gnu.org/licenses/old-licenses/gpl-2.0.en.html) clause 3, particularly as you are a paying user of it.

herisson-88, 2026-09-02:
To answer your question about the AudioLinux kernel: the maintainer (Piero) kindly shared the patch he currently ships. It is the old-style approach: RX_BUF_LENGTH raised to 10240 in bcmgenet.c, plus ETH_DATA_LEN / ETH_FRAME_LEN redefined to 9000 / 9014 in include/uapi/linux/if_ether.h and VLAN_ETH_DATA_LEN / VLAN_ETH_FRAME_LEN to 9000 / 9018 in include/linux/if_vlan.h. RBUF_PKT_RDY_THLD is untouched. Full diff below.

Which, if I follow your explanation, accounts exactly for what AudioLinux users observe in the field: the patch advertises 9000, but the highest MTU that actually works is 2032 — and 2032 = 2048 (default threshold) - 2 (alignment) - 14 (ethernet header). The whole user base settled on that number empirically without knowing why; your threshold mechanism explains it to the byte. Consider it a field confirmation of your analysis.

Piero says he will try your two commits on 6.18.x in the next days. I can test on my CM4 as well and report back here. A clean PR advertising the real ~3824 limit would definitely have users.

<details>
<summary>The AudioLinux patch as shared by the maintainer (timestamps stripped)</summary>

```diff
--- a/drivers/net/ethernet/broadcom/genet/bcmgenet.c
+++ b/drivers/net/ethernet/broadcom/genet/bcmgenet.c
@@ -50,7 +50,7 @@
 #define GENET_Q0_TX_BD_CNT    \
     (TOTAL_DESC - priv->hw_params->tx_queues * priv->hw_params->tx_bds_per_q)
 
-#define RX_BUF_LENGTH        2048
+#define RX_BUF_LENGTH        10240
 #define SKB_ALIGNMENT        32
 
 /* Tx/Rx DMA register offset, skip 256 descriptors */
 
--- a/include/linux/if_vlan.h
+++ b/include/linux/if_vlan.h
@@ -22,8 +22,8 @@
 /*
  * According to 802.3ac, the packet can be 4 bytes longer. --Klika Jan
  */
-#define VLAN_ETH_DATA_LEN    1500    /* Max. octets in payload    */
-#define VLAN_ETH_FRAME_LEN    1518    /* Max. octets in frame sans FCS */
+#define VLAN_ETH_DATA_LEN    9000    /* Max. octets in payload    */
+#define VLAN_ETH_FRAME_LEN    9018    /* Max. octets in frame sans FCS */
 
 #define VLAN_MAX_DEPTH    8        /* Max. number of nested VLAN tags parsed */
 
--- a/include/uapi/linux/if_ether.h
+++ b/include/uapi/linux/if_ether.h
@@ -33,8 +33,8 @@
 #define ETH_TLEN    2        /* Octets in ethernet type field */
 #define ETH_HLEN    14        /* Total octets in header.    */
 #define ETH_ZLEN    60        /* Min. octets in frame sans FCS */
-#define ETH_DATA_LEN    1500        /* Max. octets in payload    */
-#define ETH_FRAME_LEN    1514        /* Max. octets in frame sans FCS */
+#define ETH_DATA_LEN    9000        /* Max. octets in payload    */
+#define ETH_FRAME_LEN    9014        /* Max. octets in frame sans FCS */
 #define ETH_FCS_LEN    4        /* Octets in the FCS        */
 
 #define ETH_MIN_MTU    68        /* Min IPv4 MTU per RFC791    */
```

</details>

herisson-88, 2026-09-03:
My test with your patch give me exactly the same limit as yours : thanks a lot !!!

herisson-88, 2026-09-04:
Hope this patch will finish on the main : it would be usefull for many distributions

dsnyder0pc, 2026-09-04:
The patch works perfectly for my AudioLinux + Diretta project (https://diretta.diy):

[image]

herisson-88, 2026-09-09:
@6by9 a hope of merging this on main 🤗 ?
Thanks

6by9, 2026-09-09:
> [@6by9](https://github.com/6by9) a hope of merging this on main 🤗 ? Thanks

This is an issue rather a pull request, so there isn't any "this" to merge.

See #7614 for a pull request based on my current testing.

nbuchwitz, 2026-09-10:
Without offloading there seems to be a way to bump this even further to MTU 9000. See https://github.com/raspberrypi/linux/pull/7617