raspberrypi/linux #5562: SPI Clock Phase on CM4 spidev1.X (hardware bug?)

thecalle, 2023-08-04:
### Describe the bug

Trying to configure the SPI1 port as mode=3 we get a IOCTL error

### Steps to reproduce the behaviour

Config line in /boot/config.txt:

dtoverlay=spi1-3cs,cs0_spidev=disabled

Result:

pi@raspberrypi:~ $ ls -l /dev/*spi*
crw-rw---- 1 root spi 153, 1 Aug  4 14:14 **/dev/spidev1.1**
crw-rw---- 1 root spi 153, 0 Aug  4 14:14 **/dev/spidev1.2**

Developing a custom carrier board for the Raspberry CM4 supporting an Ethernet Switch. The chip of the switch has a SPI communication interface with the micro-processor. It supports only **CPHA=1** and **CPOL=1 SPI mode**.

We will configure it and read its status using that SPI port. Since we are also using 4 UARTs, the only CM4 available SPI port is the **SPI1**, but we found it has a critical linux driver/hardware bug.
Trying to configure the SPI1 port as **mode=3 we get a IOCTL error**. We found it is a problem of the linux driver as described in the link below:
https://raspberrypi.stackexchange.com/questions/142495/spi-clock-phase-on-cm4-spidev1-0

### Device (s)

Raspberry Pi CM4

### System

OS Version: Debian GNU Linux 11
Firmware: Apr 25 2023 18:26:03
Copyright (c) 2012 Broadcom
version d7f9c2b4ef7e4a8c0b04374a879ce89d7a948453 (clean) (release) (start)

Kenel: Linux raspberrypi 6.1.42-v8+ #1 SMP PREEMPT ARM64

### Additional context

Code to access SPI developed in Python with SPI libraries. Error is generated from OS.

thecalle, 2023-08-04:
Can we fix that driver supporting also mode=3? It seems the driver has such limitation due to a hardware bug, is it true? Does this bug also affect the port when working at low frequency rate (2 MHz)?

pelwell, 2023-08-04:
Unfortunately the driver is correct - the hardware SPI1  (and SPI2) interfaces don't support modes 2 & 3 (CPHA=1) - you can try to enable it but the wrong data is sent. Commit  e9dd4edcc98593bbcdffc0c4f37545b8fd0ad3ea explains it. I don't think it's speed related at all, just a simple mistake.

thecalle, 2024-09-04:
Hope in new hardware releases.

closed 2024-09-04 by thecalle

pelwell, 2024-09-04:
The 40-pin header on Pi 5 is driven by the RP1 chip, which has completely different SPI hardware that does support the SPI_CPHA mode.